FreshTomato-Project/freshtomato-arm #35: [BB #201] User ACL in Samba File Sharing

xixix-ft, 2025-01-29:
Ref. https://bitbucket.org/pedro311/freshtomato-arm/issues/201/user-acl-in-samba-file-sharing

Sysres created an issue 2022-02-07

Presently in Samba File Sharing we have only 2 options: “No authentication” or a single user/password for “Authentication required”. Both options aren’t very good. It is either full guest access for everybody or single account for just one user you have to share with everybody.

Would it be possible to add simple user ACL to Samba File Sharing similar to what we have for FTP Server? I am sure presently it could be done in command prompt through SSH but I am talking about having such ACL in GUI. Personally I have 2 USB drives connected to my home router and would love to be able to control access to them separately as well and have an ability to create several user accounts with different read/write access rights. I am not talking about having down to the folder level ACL found in professional NAS drives but something simple limited to the root of each drive.

HommeOursPorc, 2025-08-13:
I did not try any of this, just sharing for reference/documenting propose.

Here's some ideas I found on the ddwrt forum:
https://forum.dd-wrt.com/forum/viewtopic.php?p=242751&sid=7c00f432f58b3ebd81ec64c8ed9a04d7

And also found this on LI forum which could help here:
https://www.linksysinfo.org/index.php?threads/samba-server-on-tomato-for-15-users.77021/post-332449

This could be useful too:
http://tomatousb.wikidot.com/tut:adding-your-own-users

And this: 
http://tomatousb.wikidot.com/forum/t-437201

xixix-ft, 2025-08-13:
I ended up resorting to optware Samba 4.18.0 (Mar 2023) and quit standard FT's 3.6.25 (Feb 2015)... too many security issues
And I can't really think of our dev willing to upgrade FT to 4.18+