FreshTomato-Project/freshtomato-arm #48: [BB #309] Administration > Scripts - Firewall scripts saved in GUI are executed twice

xixix-ft, 2025-01-29:
Ref. https://bitbucket.org/pedro311/freshtomato-arm/issues/309/administration-scripts-firewall-scripts

HommeOursPorc created an issue 2023-12-24

Anything saved in the firewall scripts in the GUI (Administration > Scripts) is runned twice on both MIPS and ARM in 2023.5.

Please see https://www.linksysinfo.org/index.php?threads/bug-firewall-scripts-from-gui-get-executed-twice.78379/

File /tmp/script_fire.sh is automatically created and also nvram get script_fire gets populated then anything is executed twice…..

‌
Comments (11)

    HommeOursPorc reporter
        edited description
        changed title to [Administration > Scripts - Firewall scripts saved in GUI are executed twice](https://bitbucket.org/pedro311/freshtomato-arm/issues?title=Administration+%3E+Scripts+-+Firewall+scripts+saved+in+GUI+are+executed+twice)
        2023-12-24
    pedro repo owner
        changed status to [new](https://bitbucket.org/pedro311/freshtomato-arm/issues?status=new)
        2023-12-29
    M_ars

    Hi, i have one/two ideas to solve this. Next Year 🍻 😎 - i will need to test it
        2023-12-29
    HommeOursPorc reporter

    If I use an iptables rule instead of a ebtables one, the iptables rule is only added once. The firewall script is executed twice, but the iptables rule is only added one time. I didn't check the firewall code, but I suppose either there's a validation process before adding the rule or the rules are flushed and generated from 0 when firewall_start() is called. If I manually start and stop the firewall, iptables rule is added once (good!), ebtables rules is not flushed and added over the existing one so we end up with 2 identical ebtables rules! (Bad) An easy solution would be to add a ebtables -F in firewall_stop() ??? That might cause problems with @M_ars WET patch for the IPv6 bug.
        2024-01-01
    HommeOursPorc reporter

    Sorry it's start_firewall and stop_firewall and NOT firewall_start
        2024-01-01
    M_ars

    Yes, “ebtables -F” would cause problems for WET Mode to catch IPv6. There more i think about my ideas the less i like them 😅
        2024-01-02
    pedro repo owner

    @M_ars : can we remove this WET patch for other modes?
        2024-11-19
    M_ars

        can we remove this WET patch for other modes? 

    there is already a check in place for WET mode, so all other modes are not affected.

        If I manually start and stop the firewall, iptables rule is added once (good!), ebtables rules is not flushed and added over the existing one so we end up with 2 identical ebtables rules! 

    Yes, the iptables rule logic is completely independent of ebtables. We could think about an extra script tab ?

    For example:
        Init Script
        Shutdown Script
        Firewall Script
        EBtables Script (<-- Add / New)
        Wan up (main)
        MultiWAN up

    What do you think?

        An easy solution would be to add a ebtables -F in firewall_stop() ??? That might cause problems with @M_ars WET patch for the IPv6 bug. 

    Yes, that would be a problem for the WET patch and IPv6
        2024-11-21
    HommeOursPorc reporter

    Can the WET patch be moved in services.c or in the firewall code instead of network.c ?? Maybe flush all ebtables rules, then add patch and then add user rules from the firewall script tab? Like this the rules will not stacks up everytime the firewall service is restarted. Would this work?

    ‌

    Will this create a dead zone where IPv6 would be allowed between the moment the firewall is being restarted? I think there will be a gap that will leave the router with no IPv6 blockage between the moment the firewall is going down and before the firewall is up again. Maybe I’m wrong though.
        2024-11-21
    M_ars

    The ebtable rules will be applied only one time before we bring up the wireless radios (so no chance for any IPv6 traffic to crash the wl driver). The iptable rules will be called at least two times in our FreshTomato starting process and the wireless radios are already up and running
        2024-11-23
    HommeOursPorc reporter

    This makes sense. My idea isn’t good. Thanks @M_ars 🙂
        2024-11-23

HommeOursPorc, 2025-02-09:
Sorry for being such a mess. Just to be clear, the firewall scripts are executed every time the firewall start/restart which is completely normal. Iptables rules seems to be flushed, but ebtables rules are NOT flushed and accumulate every time the firewall is started/restarted.

From my last post in the thread on the forum:
If I use an iptables rule instead of a ebtables one, the iptables rule is only added once. The firewall script is executed twice, but the iptables rule is only added one time. I didn't check the firewall code, but I suppose either there's a validation process before adding the rule or the rules are flushed and generated from scratch when the firewall start.

If I manually start and stop the firewall, iptables rule is added once (good!), ebtables rules is not flushed and added over the existing one so we end up with 2 identical ebtables rules! (Bad).

I hope this clarifies things 🙂

I think @MarsTomato have a good idea --> add a ebtables tab in the GUI in Administration > Scripts.

HommeOursPorc, 2025-10-15:
An other idea, if @MarsTomato rules are always the first 3 rules, remove all other rules. Like: 

If is NOT wet --> flush everything

if IS wet --> flush rules no 4,5,6,etc, only keeping Mars rules that should be rules no 1,2,3

forlotto, 2025-10-15:
I've always wondered why WET was chosen as an acronym instead of WEB or WETB.  
WET stands for Wireless Ethernet Bridge for those following along. To eliminate any source of confusion.
So what they are saying is when you are running as a wireless ethernet bridge they are trying to determine what happens with ebtables to make things function correctly in this mode.

HommeOursPorc, 2025-10-15:
> An other idea, if [@MarsTomato](https://github.com/MarsTomato) rules are always the first 3 rules, remove all other rules. Like:
> 
> If is NOT wet --> flush everything
> 
> if IS wet --> flush rules no 4,5,6,etc, only keeping Mars rules that should be rules no 1,2,3

Forget this idea. It won't work if the user insert the rule in front of the other rules.... Wow... Now I feel very stupid. Brb, gonna hide under a rock.

HommeOursPorc, 2025-10-15:
> So what they are saying is when you are running as a wireless ethernet bridge they are trying to determine what happens with ebtables to make things function correctly in this mode.

There's a bug in the wireless driver that does a kernel panic when any IPv6 goes through a Wireless Ethernet Bridge. Since it is a bridge, we need to use ebtables rules (Layer 2). Blocking IPv6 prevent the kernel panic. 

The problem is that ebatbles rules are not flushed when the firewall is restarted, so they duplicate. It is not the case with iptables rules which are flushed before being re-populated on a firewall restart.

forlotto, 2025-10-15:
Not my thoughts just a quick search. It appears the way to go is atomic loading loading everything in one shot to prevent downtime using ebtables restore and a file that holds the rules and it must be loaded after the firewall is up requiring a sleep timer  or wait time basically. 

Instructions for Fresh Tomato
1. Navigate to the Scripting Interface
Log in to your Fresh Tomato router's web interface (typically at http://192.168.1.1).
Go to Administration > Scripts.
Select the Firewall tab. 
2. Add the hardened ebtables commands
In the Firewall script text box, add the following commands. This script uses the ebtables-restore command with a Here Document (a form of input redirection) to perform an atomic flush and reload of the firewall rules. This avoids the brief period of vulnerability that can occur with separate ebtables -F and ebtables -A commands. 
bash
# Wait for the firewall to be fully functional before applying rules
# This is a good practice to ensure all necessary router services are up
# before altering the ebtables.
sleep 5

# Use ebtables-restore with a Here Document for atomic reload
/sbin/ebtables-restore <<EOF
*filter
:INPUT ACCEPT
:FORWARD ACCEPT
:OUTPUT ACCEPT
-F
-A FORWARD -p IPv6 -j DROP
COMMIT
EOF
Use code with caution.

3. Save the script and reboot
Click the Save button below the text box to apply the changes.
Reboot your router for the changes to take effect reliably. The script will be executed automatically during the router's boot process when the firewall service starts. 
How this method is hardened for Fresh Tomato
GUI-Managed: The rules are stored in a persistent memory location (NVRAM) managed by the router's firmware, so they will survive reboots and are restored properly. You don't need to manually create files on the router's flash memory, which can be prone to errors or wear.
Atomic Operation: The use of ebtables-restore with a Here Document ensures that the existing ebtables rules are flushed and the new ones are loaded in a single, atomic transaction. This eliminates the race condition and the brief period where the firewall could be left open.
Synchronous Execution: By placing the code in the "Firewall" script, you ensure that the ebtables rules are loaded at the correct time in the boot sequence, specifically during the firewall's initialization phase.
Graceful Start: Including a brief sleep command is a good practice for Fresh Tomato. It provides a small delay, giving the router's core services time to initialize fully before applying the firewall rules, which increases the overall reliability of the boot process.