FreshTomato-Project/freshtomato-arm #53: [BB #337] 2024.3 - DDNS - External IP Checker failing with Unknown response (-1)

xixix-ft, 2025-01-29:
Ref. https://bitbucket.org/pedro311/freshtomato-arm/issues/337/20243-ddns-external-ip-checker-failing

Vladislav Staroselskiy created an issue 2024-09-09

Hardware: ASUS (T-Mobile) AC1900
Version: 2024.3
Firmware: freshtomato-RT-AC68U-K26ARM-2024.3-AIO-64K.trx
Configuration: No WAN IP addresses, but Management IP assigned to the br1 interface, with access to Internet

Issue: am getting “Unknown response (-1)” in a log when I flip DDNS from the “Custom IP Address” mode to the “External IP Checker”. The “Custom IP Address” works and updates Cloudfare with the defined IP just fine - so, my assumption that the issue has nothing to do with zone/token, but rather with the IP checker.

This is what I am getting in the log:

```
root@ASUS4:/tmp/home/root# tail /var/log/messages
Sep  9 13:52:36 ASUS4 user.info preinit[1]: ddns is stopped
Sep  9 13:52:36 ASUS4 user.info preinit[1]: ddns is started
Sep  9 13:52:37 ASUS4 daemon.err mdu[20595]: Unknown response (-1).
Sep  9 13:53:29 ASUS4 user.info preinit[1]: ddns is stopped
Sep  9 13:53:29 ASUS4 user.info preinit[1]: ddns is started
root@ASUS4:/tmp/home/root# 

```
I suspect it has something to do with curl attempting to use wrong interface to reach out the external checker.

I am successfully getting the IP address when running curl from shell without the interface parameter:

```
root@ASUS4:/tmp/home/root# curl -s --connect-timeout 5 http://ipinfo.io/ip
184.170.X.Xroot@ASUS4:/tmp/home/root#
```

This is my routing table (note br1 being in the default route):

```
root@ASUS4:/tmp/home/root# netstat -rn
Kernel IP routing table
Destination     Gateway         Genmask         Flags   MSS Window  irtt Iface
10.0.10.4       0.0.0.0         255.255.255.252 U         0 0          0 br2
10.0.10.8       0.0.0.0         255.255.255.252 U         0 0          0 br3
192.168.1.0     0.0.0.0         255.255.255.0   U         0 0          0 br0
10.0.9.0        0.0.0.0         255.255.255.0   U         0 0          0 br1
127.0.0.0       0.0.0.0         255.0.0.0       U         0 0          0 lo
0.0.0.0         10.0.9.1        0.0.0.0         UG        0 0          0 br1
root@ASUS4:/tmp/home/root# 
```

[This comment](https://www.linksysinfo.org/index.php?threads/get-ip-address-from-external-ip-checker.73331/#post-318582) gave me a clue to try the explicit WAN interface, which is failing (and I suspect it to be a cause)

```
root@ASUS4:/tmp/home/root# nvram get wan_ifname
vlan1
root@ASUS4:/tmp/home/root# nvram get wan2_ifname
root@ASUS4:/tmp/home/root# nvram get wan3_ifname
root@ASUS4:/tmp/home/root# nvram get wan4_ifname
root@ASUS4:/tmp/home/root# curl -s --interface vlan1 --connect-timeout 5 http://ipinfo.io/ip
root@ASUS4:/tmp/home/root#

```
As you see above, I am not getting any response when I use wrong interface.

Please, let me know if I can help with further troubleshooting. Thanks!

[image]

    screenshot2.png

[image]

    screenshot1.png
[... 942 lines cut ...]

    Jan 09 12:29:53 <= D {"result":[{"id":"<redacted>","zone_id":"<redacted>","zone_name":"<redacted>","name":"<redacted>","type":"A","content":"184.<redacted>","proxiable":true,"proxied":false,"ttl":1,"settings":{},"meta":{"auto_added":false,"managed_by_apps":false,"managed_by_argo_tunnel":false},"comment":null,"tags":[],"created_on":"2024-01-26T11:58:17.151364Z","modified_on":"2024-07-31T21:16:23.511601Z"}],"success":true,"errors":[],"messages":[],"result_info":{"page":1,"per_page":10
    Jan 09 12:29:53 <= D 0,"count":1,"total_count":1,"total_pages":1}}

    ‌
        2025-01-09
    Vladislav Staroselskiy reporter

    /tmp/mdu0-cloudflare.txt

    Note the last line here - it reported “Malformed encoding found in chunked-encoding" without giving the second chunk of the array that I mentioned above:

    Jan 09 12:29:52 == I TLSv1.3 (OUT), TLS handshake, Client hello (1)
    Jan 09 12:29:52 == I TLSv1.3 (IN), TLS handshake, Server hello (2)
    Jan 09 12:29:52 == I TLSv1.3 (IN), TLS handshake, Encrypted Extensions (8)
    Jan 09 12:29:52 == I TLSv1.3 (IN), TLS handshake, Certificate (11)
    Jan 09 12:29:53 == I TLSv1.3 (IN), TLS handshake, CERT verify (15)
    Jan 09 12:29:53 == I TLSv1.3 (IN), TLS handshake, Finished (20)
    Jan 09 12:29:53 == I TLSv1.3 (OUT), TLS change cipher, Change cipher spec (1)
    Jan 09 12:29:53 == I TLSv1.3 (OUT), TLS handshake, Finished (20)
    Jan 09 12:29:53 => H GET /client/v4/zones/<redacted_zone_id>/dns_records?type=A&name=<reducted_name>.<reducted_domain>&order=name&direction=asc HTTP/1.1
    Jan 09 12:29:53 => H Host: api.cloudflare.com
    Jan 09 12:29:53 => H Accept: */*

TEXT CANNOT BE LONGER THAN 65535 CHARS - TRUNCATED

Vlad-Star, 2025-01-30:
Looks like I won't be able to paste logs larger than 65k directly to the body

pedro0311, 2025-01-30:
65k is a lot ;)

pedro0311, 2025-01-30:
Before we go any further, could you please check the update via curl in the router shell (remember to add the interface) to see if it works?

Vlad-Star, 2025-01-30:
I did some testing... my global API key works, but API tokens created for the specific domain (with DNS->Edit permission) currently returning auth error (even though they used to work fine a month ago)

`{"success":false,"errors":[{"code":10001,"message":"Unable to authenticate request"}]}`

I will open a ticket with Cloudflare shortly to fix that.

Meanwhile, I think I found what the problem with DNS update at Cloudflare is. They won't take "Authorization: Bearer" header for the updates anymore - it fails. They now require proper **X-Auth-Email** and **X-Auth-Key** headers. 

My successful curl query for the DNS update looks like that:

```
curl https://api.cloudflare.com/client/v4/zones/$ZONE_ID/dns_records/$DNS_RECORD_ID \
    -X PUT \
    -H "X-Auth-Email: $CLOUDFLARE_EMAIL" \
    -H "X-Auth-Key: $CLOUDFLARE_API_KEY" \
    -H 'Content-Type: application/json' \
    -d '{
      "content": "'$IP'",
      "name": "'$FQDN'",
      "proxied": false,
      "type": "A"
    }'

```
Re: https://developers.cloudflare.com/api/resources/dns/subresources/records/methods/update/

Vlad-Star, 2025-01-30:
Looks like they don't allow to open support cases for free accounts :(

Vlad-Star, 2025-01-30:
(never mind, I was wrong, re-testing)

Vlad-Star, 2025-01-30:
also works with **--interface br1** parameter

pedro0311, 2025-01-30:
OK, try this one: https://bitbucket.org/pedro311/freshtomato-arm/downloads/freshtomato-RT-AC68U-K26ARM-2024.5.171-beta-mdu-log-AIO-64K.trx

pedro0311, 2025-01-30:
And this: https://bitbucket.org/pedro311/freshtomato-arm/downloads/freshtomato-RT-AC68U-K26ARM-2024.5.173-beta-mdu-cb-AIO-64K.trx

pedro0311, 2025-01-30:
Oh, and what about the first request to CF API, this one with (GET): 
`/client/v4/zones/_ZONE_/dns_records?type=A&name=_HOST_&order=name&direction=asc`
?

Vlad-Star, 2025-01-30:
> Oh, and what about the first request to CF API, this one with (GET): `/client/v4/zones/_ZONE_/dns_records?type=A&name=_HOST_&order=name&direction=asc` ?

`curl -X GET "https://api.cloudflare.com/client/v4/zones/$ZONE_ID/dns_records" -H "Authorization: Bearer $CLOUDFLARE_API_KEY" -H "Content-Type:application/json"`

returns success  with the list of records

Vlad-Star, 2025-01-30:
v107
```
root@ASUS4:/tmp/home/root# awk -f redaction.awk dict.txt /var/log/messages | egrep 'mdu|ddns|curl|wget'
Jan 30 15:29:12 ASUS4 user.debug preinit[1]: *** stop_ddns: IN
Jan 30 15:29:12 ASUS4 user.info preinit[1]: ddns is stopped
Jan 30 15:29:12 ASUS4 user.debug preinit[1]: *** start_ddns: IN
Jan 30 15:29:12 ASUS4 user.info preinit[1]: ddns is started
Jan 30 15:29:12 ASUS4 user.debug ddns-update[3945]: *** ddns_update_main: args:
Jan 30 15:29:12 ASUS4 user.debug ddns-update[3945]: *** update: IN num=[0] dirty=[0] force=[0]
Jan 30 15:29:12 ASUS4 user.debug ddns-update[3945]: *** update: cru d ddns0
Jan 30 15:29:12 ASUS4 user.debug ddns-update[3945]: *** update: cru d ddnsf0
Jan 30 15:29:12 ASUS4 user.debug ddns-update[3945]: *** update: !f_exist(/var/lib/mdu/ddnsx0.msg) - creating ...
Jan 30 15:29:12 ASUS4 user.debug ddns-update[3945]: *** mdu <<<<<<<
Jan 30 15:29:12 ASUS4 daemon.debug mdu[3954]: *** main: IN v2.1 [libcurl]
Jan 30 15:29:12 ASUS4 daemon.debug mdu[3954]: *** main: checking for no WAN mode - true, using custom interface: br1
Jan 30 15:29:12 ASUS4 daemon.debug mdu[3954]: *** check_cookie: IN
Jan 30 15:29:12 ASUS4 daemon.debug mdu[3954]: *** read_tmaddr: IN cachename: /var/lib/mdu/ddnsx0.cache
Jan 30 15:29:12 ASUS4 daemon.debug mdu[3954]: *** check_cookie: no cookie
Jan 30 15:29:12 ASUS4 daemon.debug mdu[3954]: *** main: proceeding DDNS server update [service: cloudflare ] ...
Jan 30 15:29:12 ASUS4 daemon.debug mdu[3954]: *** update_cloudflare: IN
Jan 30 15:29:12 ASUS4 daemon.debug mdu[3954]: *** _http_req: IN host=[api.cloudflare.com] query=[/client/v4/zones/<redacted_zone>/dns_records?type=A&name=<redacted_name>.<redacted_domain>&order=name&direction=asc] ssl=[1] header=[User-Agent: Mozilla/5.0 (X11; Linux x86_64; rv:10.0) Gecko/20100101 Firefox/109.0^M Authorization: Bearer <redacted_bearer>^M Content-Type: application/json^M Cache-Control: no-cache] auth=[0] data=[(null)] req=[GET] ifname=[br1]
Jan 30 15:29:12 ASUS4 daemon.debug mdu[3954]: *** _http_req: resolving IP of server api.cloudflare.com ...
Jan 30 15:29:12 ASUS4 daemon.debug mdu[3954]: *** curl_resolve_ip: IN
Jan 30 15:29:12 ASUS4 daemon.debug mdu[3954]: *** curl_setup: IN
Jan 30 15:29:12 ASUS4 daemon.debug mdu[3954]: *** curl_setup: OUT
Jan 30 15:29:12 ASUS4 daemon.debug mdu[3954]: *** curl_resolve_ip: after curl_setup()
Jan 30 15:29:12 ASUS4 daemon.debug mdu[3954]: *** curl_resolve_ip: after CURLOPT_URL
Jan 30 15:29:12 ASUS4 daemon.debug mdu[3954]: *** curl_resolve_ip: after CURLOPT_HTTPHEADER
Jan 30 15:29:12 ASUS4 daemon.debug mdu[3954]: *** curl_resolve_ip: after CURLOPT_INTERFACE
Jan 30 15:29:12 ASUS4 daemon.debug mdu[3954]: *** curl_resolve_ip: before curl_easy_perform trys=[4]
Jan 30 15:29:13 ASUS4 daemon.debug mdu[3954]: *** curl_resolve_ip: after curl_easy_perform r=[56] trys=[4]
Jan 30 15:29:13 ASUS4 daemon.debug mdu[3954]: *** curl_cleanup: IN
Jan 30 15:29:13 ASUS4 daemon.debug mdu[3954]: *** curl_cleanup: after curl_dfile
Jan 30 15:29:13 ASUS4 daemon.debug mdu[3954]: *** curl_cleanup: after curl_easy_cleanup()
Jan 30 15:29:13 ASUS4 daemon.debug mdu[3954]: *** curl_cleanup: OUT
Jan 30 15:29:13 ASUS4 daemon.debug mdu[3954]: *** curl_resolve_ip: OUT IP=[104.19.192.29]
Jan 30 15:29:13 ASUS4 daemon.debug mdu[3954]: *** _http_req: IP=[104.19.192.29] of host=[api.cloudflare.com]
Jan 30 15:29:13 ASUS4 daemon.debug mdu[3954]: *** _http_req: after curl_resolve_ip
Jan 30 15:29:13 ASUS4 daemon.debug mdu[3954]: *** _http_req: after curl_wbuf
Jan 30 15:29:13 ASUS4 daemon.debug mdu[3954]: *** curl_setup: IN
Jan 30 15:29:13 ASUS4 daemon.debug mdu[3954]: *** curl_setup: OUT
Jan 30 15:29:13 ASUS4 daemon.debug mdu[3954]: *** _http_req: after curl_setup()
Jan 30 15:29:13 ASUS4 daemon.debug mdu[3954]: *** _http_req: after CURLOPT_URL
Jan 30 15:29:13 ASUS4 daemon.debug mdu[3954]: *** _http_req: after CURLOPT_HTTPHEADER
Jan 30 15:29:14 ASUS4 daemon.debug mdu[3954]: *** _http_req: after CURLOPT_INTERFACE
Jan 30 15:29:14 ASUS4 daemon.debug mdu[3954]: *** _http_req: after CURLOPT_HTTPAUTH
Jan 30 15:29:14 ASUS4 daemon.debug mdu[3954]: *** _http_req: after CURLOPT_WRITEDATA
Jan 30 15:29:14 ASUS4 daemon.debug mdu[3954]: *** _http_req: after CURLOPT_UPLOAD
Jan 30 15:29:14 ASUS4 daemon.debug mdu[3954]: *** _http_req: after route_adddel()
Jan 30 15:29:14 ASUS4 daemon.debug mdu[3954]: *** _http_req: before curl_easy_perform trys=[4]
Jan 30 15:29:15 ASUS4 daemon.debug mdu[3954]: *** _http_req: after curl_easy_perform r=[56] trys=[4]
Jan 30 15:29:15 ASUS4 daemon.debug mdu[3954]: *** _http_req: before curl_slist_free_all()
Jan 30 15:29:15 ASUS4 daemon.debug mdu[3954]: *** _http_req: connected, response=[200]
Jan 30 15:29:15 ASUS4 daemon.debug mdu[3954]: *** curl_cleanup: IN
Jan 30 15:29:15 ASUS4 daemon.debug mdu[3954]: *** curl_cleanup: after curl_dfile
Jan 30 15:29:15 ASUS4 daemon.debug mdu[3954]: *** curl_cleanup: after curl_easy_cleanup()
Jan 30 15:29:15 ASUS4 daemon.debug mdu[3954]: *** curl_cleanup: OUT
Jan 30 15:29:15 ASUS4 daemon.debug mdu[3954]: *** update_cloudflare: after http_req() query=[/client/v4/zones/<redacted_zone>/dns_records?type=A&name=<redacted_name>.<redacted_domain>&order=name&direction=asc] header=[User-Agent: Mozilla/5.0 (X11; Linux x86_64; rv:10.0) Gecko/20100101 Firefox/109.0^M Authorization: Bearer <redacted_bearer>^M Content-Type: application/json^M Cache-Control: no-cache]
Jan 30 15:29:15 ASUS4 daemon.debug mdu[3954]: *** update_cloudflare: after http_req(1)
[... cut ...]
Jan 30 15:29:15 <= D {"result":[{"id":"<redacted_id>","zone_id":"<redacted_zone>","zone_name":"<redacted_domain>","name":"<redacted_name>.<redacted_domain>","type":"A","content":"<redacted_ip>","proxiable":true,"proxied":false,"ttl":1,"settings":{},"meta":{"auto_added":false,"managed_by_apps":false,"managed_by_argo_tunnel":false},"comment":null,"tags":[],"
Jan 30 15:29:15 <= D created_on":"2024-01-26T11:58:17.151364Z","modified_on":"2025-01-30T17:23:49.058729Z"}],"success":true,"errors":[],"messages":[],"result_info":{"page":1,"per_page":100,"count":1,"total_count":1,"total_pages":1}}
Jan 30 15:29:15 == I Malformed encoding found in chunked-encoding

root@ASUS4:/tmp/home/root#
```

Vlad-Star, 2025-01-30:
v173
```
root@ASUS4:/tmp/home/root# awk -f redaction.awk dict.txt /var/log/messages | egrep 'mdu|ddns|curl|wget'
Jan 30 15:44:47 ASUS4 user.debug preinit[1]: *** stop_ddns: IN
Jan 30 15:44:47 ASUS4 user.info preinit[1]: ddns is stopped
Jan 30 15:44:47 ASUS4 user.debug preinit[1]: *** start_ddns: IN
Jan 30 15:44:47 ASUS4 user.info preinit[1]: ddns is started
Jan 30 15:44:47 ASUS4 user.debug ddns-update[2040]: *** ddns_update_main: args:
Jan 30 15:44:47 ASUS4 user.debug ddns-update[2040]: *** update: IN num=[0] dirty=[0] force=[0]
Jan 30 15:44:47 ASUS4 user.debug ddns-update[2040]: *** update: cru d ddns0
Jan 30 15:44:47 ASUS4 user.debug ddns-update[2040]: *** update: cru d ddnsf0
Jan 30 15:44:47 ASUS4 user.debug ddns-update[2040]: *** update: !f_exist(/var/lib/mdu/ddnsx0.msg) - creating ...
Jan 30 15:44:47 ASUS4 user.debug ddns-update[2040]: *** mdu <<<<<<<
Jan 30 15:44:47 ASUS4 daemon.debug mdu[2049]: *** main: IN v2.1 [libcurl]
Jan 30 15:44:47 ASUS4 daemon.debug mdu[2049]: *** main: checking for no WAN mode - true, using custom interface: br1
Jan 30 15:44:47 ASUS4 daemon.debug mdu[2049]: *** check_cookie: IN
Jan 30 15:44:47 ASUS4 daemon.debug mdu[2049]: *** read_tmaddr: IN cachename: /var/lib/mdu/ddnsx0.cache
Jan 30 15:44:47 ASUS4 daemon.debug mdu[2049]: *** check_cookie: no cookie
Jan 30 15:44:47 ASUS4 daemon.debug mdu[2049]: *** main: proceeding DDNS server update [service: cloudflare] ...
Jan 30 15:44:47 ASUS4 daemon.debug mdu[2049]: *** update_cloudflare: IN
Jan 30 15:44:47 ASUS4 daemon.debug mdu[2049]: *** _http_req: IN host=[api.cloudflare.com] query=[/client/v4/zones/<redacted_zone>/dns_records?type=A&name=<redacted_name>.<redacted_domain>&order=name&direction=asc] ssl=[1] header=[User-Agent: Mozilla/5.0 (X11; Linux x86_64; rv:10.0) Gecko/20100101 Firefox/109.0^M Authorization: Bearer <redacted_bearer>^M Content-Type: application/json^M Cache-Control: no-cache] auth=[0] data=[(null)] req=[GET] ifname=[br1]
Jan 30 15:44:47 ASUS4 daemon.debug mdu[2049]: *** curl_setup: IN
Jan 30 15:44:47 ASUS4 daemon.debug mdu[2049]: *** curl_setup: after curl_global_sslset() result=[0]
Jan 30 15:44:47 ASUS4 daemon.debug mdu[2049]: *** curl_setup: after CURLOPT_FOLLOWLOCATION cc=[0]
Jan 30 15:44:47 ASUS4 daemon.debug mdu[2049]: *** curl_setup: after CURLOPT_MAXREDIRS cc=[0]
Jan 30 15:44:47 ASUS4 daemon.debug mdu[2049]: *** curl_setup: after CURLOPT_CONNECTTIMEOUT cc=[0]
Jan 30 15:44:47 ASUS4 daemon.debug mdu[2049]: *** curl_setup: after CURLOPT_TIMEOUT cc=[0]
Jan 30 15:44:47 ASUS4 daemon.debug mdu[2049]: *** curl_setup: after CURLOPT_VERBOSE cc=[0]
Jan 30 15:44:47 ASUS4 daemon.debug mdu[2049]: *** curl_setup: after CURLOPT_ERRORBUFFER cc=[0]
Jan 30 15:44:47 ASUS4 daemon.debug mdu[2049]: *** curl_setup: after CURLOPT_DEBUGFUNCTION cc=[0]
Jan 30 15:44:47 ASUS4 daemon.debug mdu[2049]: *** curl_setup: after CURLOPT_DEBUGDATA cc=[0]
Jan 30 15:44:47 ASUS4 daemon.debug mdu[2049]: *** curl_setup: after CURLOPT_DEFAULT_PROTOCOL cc=[0]
Jan 30 15:44:47 ASUS4 daemon.debug mdu[2049]: *** curl_setup: after CURLOPT_URL cc=[0]
Jan 30 15:44:47 ASUS4 daemon.debug mdu[2049]: *** curl_setup: after CURLOPT_INTERFACE cc=[0]
Jan 30 15:44:47 ASUS4 daemon.debug mdu[2049]: *** curl_setup: OUT
Jan 30 15:44:47 ASUS4 daemon.debug mdu[2049]: *** _http_req: after curl_setup()
Jan 30 15:44:47 ASUS4 daemon.debug mdu[2049]: *** _http_req: after CURLOPT_HTTP_VERSION cc=[0]
Jan 30 15:44:47 ASUS4 daemon.debug mdu[2049]: *** _http_req: before header=[User-Agent: Mozilla/5.0 (X11; Linux x86_64; rv:10.0) Gecko/20100101 Firefox/109.0^M Authorization: Bearer <redacted_bearer>^M Content-Type: application/json^M Cache-Control: no-cache]
Jan 30 15:44:48 ASUS4 daemon.debug mdu[2049]: *** _http_req: after headers
Jan 30 15:44:48 ASUS4 daemon.debug mdu[2049]: *** _http_req: after CURLOPT_HTTPHEADER cc=[0]
Jan 30 15:44:48 ASUS4 daemon.debug mdu[2049]: *** _http_req: after CURLAUTH_NONE cc=[0]
Jan 30 15:44:48 ASUS4 daemon.debug mdu[2049]: *** _http_req: after CURLOPT_UPLOAD cc=[0]
Jan 30 15:44:48 ASUS4 daemon.debug mdu[2049]: *** _http_req: after CURLOPT_READDATA cc=[0]
Jan 30 15:44:48 ASUS4 daemon.debug mdu[2049]: *** _http_req: after CURLOPT_INFILESIZE cc=[0]
Jan 30 15:44:48 ASUS4 daemon.debug mdu[2049]: *** _http_req: resolving IP of server api.cloudflare.com ...
Jan 30 15:44:48 ASUS4 daemon.debug mdu[2049]: *** _http_req: before curl_easy_perform trys=[4]
Jan 30 15:44:49 ASUS4 daemon.debug mdu[2049]: *** _http_req: after curl_easy_perform r=[56] trys=[4]
Jan 30 15:44:49 ASUS4 daemon.debug mdu[2049]: *** _http_req: return=[56] IP=[104.19.192.175] of host=[api.cloudflare.com]
Jan 30 15:44:50 ASUS4 daemon.debug mdu[2049]: *** _http_req: after CURLOPT_WRITEFUNCTION cc=[0]
Jan 30 15:44:50 ASUS4 daemon.debug mdu[2049]: *** _http_req: after CURLOPT_WRITEDATA cc=[0]
Jan 30 15:44:50 ASUS4 daemon.debug mdu[2049]: *** _http_req: connecting ...
Jan 30 15:44:50 ASUS4 daemon.debug mdu[2049]: *** _http_req: before curl_easy_perform trys=[4]
Jan 30 15:44:50 ASUS4 daemon.debug mdu[2049]: *** _http_req: after curl_easy_perform r=[56] trys=[4]
Jan 30 15:44:50 ASUS4 daemon.debug mdu[2049]: *** _http_req: connected, response=[200] bytes=[563] received=[{"result":[{"id":"<redacted_id>","zone_id":"<redacted_zone>","zone_name":"<redacted_domain>","name":"<redacted_name>.<redacted_domain>","type":"A","content":"<redacted_ip>","proxiable":true,"proxied":false,"ttl":1,"settings":{},"meta":{"auto_added":false,"managed_by_apps":false,"managed_by_argo_tunnel":false},"comment":null," [... cut ...]
Jan 30 15:44:50 <= D {"result":[{"id":"<redacted_id>","zone_id":"<redacted_zone>","zone_name":"<redacted_domain>","name":"<redacted_name>.<redacted_domain>","type":"A","content":"<redacted_ip>","proxiable":true,"proxied":false,"ttl":1,"settings":{},"meta":{"auto_added":false,"managed_by_apps":false,"managed_by_argo_tunnel":false},"comment":null,"tags":[],"
Jan 30 15:44:50 <= D created_on":"2024-01-26T11:58:17.151364Z","modified_on":"2025-01-30T17:23:49.058729Z"}],"success":true,"errors":[],"messages":[],"result_info":{"page":1,"per_page":100,"count":1,"total_count":1,"total_pages":1}}
Jan 30 15:44:50 == I Malformed encoding found in chunked-encoding

root@ASUS4:/tmp/home/root#

```

Vlad-Star, 2025-01-30:
We are still getting the following errors
`Malformed encoding found in chunked-encoding`

I may be wrong... but from what I am reading, the curl callback function maybe called multiple times during the transfer (especially with chunked encoding).

I'd suggest to review **curl_cb()** function from **mdu.c** and adjust it to what's recommended here (i.e. with realloc inside):
https://curl.se/libcurl/c/CURLOPT_WRITEFUNCTION.html

```
static size_t cb(char *data, size_t size, size_t nmemb, void *clientp)
{
  size_t realsize = size * nmemb;
  struct memory *mem = (struct memory *)clientp;
 
  char *ptr = realloc(mem->response, mem->size + realsize + 1);
  if(!ptr)
    return 0;  /* out of memory */
 
  mem->response = ptr;
  memcpy(&(mem->response[mem->size]), data, realsize);
  mem->size += realsize;
  mem->response[mem->size] = 0;
 
  return realsize;
}
```

pedro0311, 2025-01-30:
It's already there (even updated by AI), in V173.
I have (maybe) a stupid idea: could you disable dump
`nvram unset debug_ddns
nvram commit`
and test again V173?

Vlad-Star, 2025-01-30:
v173 after disabling the dump
```
root@ASUS4:/tmp/home/root# awk -f redaction.awk dict.txt /var/log/messages | egrep 'mdu|ddns|curl|wget'
Jan 30 16:54:46 ASUS4 user.debug preinit[1]: *** stop_ddns: IN
Jan 30 16:54:46 ASUS4 user.info preinit[1]: ddns is stopped
Jan 30 16:54:46 ASUS4 user.debug preinit[1]: *** start_ddns: IN
Jan 30 16:54:46 ASUS4 user.info preinit[1]: ddns is started
Jan 30 16:54:46 ASUS4 user.debug ddns-update[4569]: *** ddns_update_main: args:
Jan 30 16:54:46 ASUS4 user.debug ddns-update[4569]: *** update: IN num=[0] dirty=[0] force=[0]
Jan 30 16:54:46 ASUS4 user.debug ddns-update[4569]: *** update: cru d ddns0
Jan 30 16:54:46 ASUS4 user.debug ddns-update[4569]: *** update: cru d ddnsf0
Jan 30 16:54:46 ASUS4 user.debug ddns-update[4569]: *** update: !f_exist(/var/lib/mdu/ddnsx0.msg) - creating ...
Jan 30 16:54:46 ASUS4 user.debug ddns-update[4569]: *** mdu <<<<<<<
Jan 30 16:54:46 ASUS4 daemon.debug mdu[4578]: *** main: IN v2.1 [libcurl]
Jan 30 16:54:46 ASUS4 daemon.debug mdu[4578]: *** main: checking for no WAN mode - true, using custom interface: br1
Jan 30 16:54:46 ASUS4 daemon.debug mdu[4578]: *** check_cookie: IN
Jan 30 16:54:46 ASUS4 daemon.debug mdu[4578]: *** read_tmaddr: IN cachename: /var/lib/mdu/ddnsx0.cache
Jan 30 16:54:46 ASUS4 daemon.debug mdu[4578]: *** check_cookie: no cookie
Jan 30 16:54:46 ASUS4 daemon.debug mdu[4578]: *** main: proceeding DDNS server update [service: cloudflare] ...
Jan 30 16:54:46 ASUS4 daemon.debug mdu[4578]: *** update_cloudflare: IN
Jan 30 16:54:46 ASUS4 daemon.debug mdu[4578]: *** _http_req: IN host=[api.cloudflare.com] query=[/client/v4/zones/<redacted_zone>/dns_records?type=A&name=<redacted_name>.<redacted_domain>&order=name&direction=asc] ssl=[1] header=[User-Agent: Mozilla/5.0 (X11; Linux x86_64; rv:10.0) Gecko/20100101 Firefox/109.0^M Authorization: Bearer <redacted_bearer>^M Content-Type: application/json^M Cache-Control: no-cache] auth=[0] data=[(null)] req=[GET] ifname=[br1]
Jan 30 16:54:46 ASUS4 daemon.debug mdu[4578]: *** curl_setup: IN
Jan 30 16:54:46 ASUS4 daemon.debug mdu[4578]: *** curl_setup: after curl_global_sslset() result=[0]
Jan 30 16:54:46 ASUS4 daemon.debug mdu[4578]: *** curl_setup: after CURLOPT_FOLLOWLOCATION cc=[0]
Jan 30 16:54:46 ASUS4 daemon.debug mdu[4578]: *** curl_setup: after CURLOPT_MAXREDIRS cc=[0]
Jan 30 16:54:46 ASUS4 daemon.debug mdu[4578]: *** _http_req: after curl_setup()
Jan 30 16:54:46 ASUS4 daemon.debug mdu[4578]: *** _http_req: after CURLOPT_HTTP_VERSION cc=[0]
Jan 30 16:54:46 ASUS4 daemon.debug mdu[4578]: *** _http_req: before header=[User-Agent: Mozilla/5.0 (X11; Linux x86_64; rv:10.0) Gecko/20100101 Firefox/109.0^M Authorization: Bearer <redacted_bearer>^M Content-Type: application/json^M Cache-Control: no-cache]
Jan 30 16:54:46 ASUS4 daemon.debug mdu[4578]: *** _http_req: after headers
Jan 30 16:54:46 ASUS4 daemon.debug mdu[4578]: *** _http_req: after CURLOPT_HTTPHEADER cc=[0]
Jan 30 16:54:46 ASUS4 daemon.debug mdu[4578]: *** _http_req: after CURLAUTH_NONE cc=[0]
Jan 30 16:54:46 ASUS4 daemon.debug mdu[4578]: *** _http_req: after CURLOPT_UPLOAD cc=[0]
Jan 30 16:54:46 ASUS4 daemon.debug mdu[4578]: *** _http_req: after CURLOPT_READDATA cc=[0]
Jan 30 16:54:46 ASUS4 daemon.debug mdu[4578]: *** _http_req: after CURLOPT_INFILESIZE cc=[0]
Jan 30 16:54:47 ASUS4 daemon.debug mdu[4578]: *** _http_req: resolving IP of server api.cloudflare.com ...
Jan 30 16:54:47 ASUS4 daemon.debug mdu[4578]: *** _http_req: before curl_easy_perform trys=[4]
Jan 30 16:54:48 ASUS4 daemon.debug mdu[4578]: *** _http_req: after curl_easy_perform r=[0] trys=[4]
Jan 30 16:54:48 ASUS4 daemon.debug mdu[4578]: *** _http_req: return=[0] IP=[104.19.193.29] of host=[api.cloudflare.com]
Jan 30 16:54:48 ASUS4 daemon.debug mdu[4578]: *** _http_req: after CURLOPT_WRITEFUNCTION cc=[0]
Jan 30 16:54:48 ASUS4 daemon.debug mdu[4578]: *** _http_req: after CURLOPT_WRITEDATA cc=[0]
Jan 30 16:54:48 ASUS4 daemon.debug mdu[4578]: *** _http_req: connecting ...
Jan 30 16:54:48 ASUS4 daemon.debug mdu[4578]: *** _http_req: before curl_easy_perform trys=[4]
Jan 30 16:54:49 ASUS4 daemon.debug mdu[4578]: *** _http_req: after curl_easy_perform r=[0] trys=[4]
Jan 30 16:54:49 ASUS4 daemon.debug mdu[4578]: *** _http_req: connected, response=[200] bytes=[563] received=[{"result":[{"id":"<redacted_id>","zone_id":"<redacted_zone>","zone_name":"<redacted_domain>","name":"<redacted_name>.<redacted_domain>","type":"A","content":"<redacted_ip>","proxiable":true,"proxied":false,"ttl":1,"settings":{},"meta":{"auto_added":false,"managed_by_apps":false,"managed_by_argo_tunnel":false},"comment":null,"tags":[],"created_on":"2024-01-26T11:58:17.151364Z","modified_on":"2025-01-
Jan 30 16:54:49 ASUS4 daemon.debug mdu[4578]: *** _http_req: before curl_slist_free_all()
Jan 30 16:54:49 ASUS4 daemon.debug mdu[4578]: *** curl_cleanup: IN
Jan 30 16:54:49 ASUS4 daemon.debug mdu[4578]: *** curl_cleanup: after curl_dfile
Jan 30 16:54:49 ASUS4 daemon.debug mdu[4578]: *** curl_cleanup: after curl_easy_cleanup()
Jan 30 16:54:49 ASUS4 daemon.debug mdu[4578]: *** curl_cleanup: OUT
Jan 30 16:54:49 ASUS4 daemon.debug mdu[4578]: *** _http_req: OUT
Jan 30 16:54:49 ASUS4 daemon.debug mdu[4578]: *** update_cloudflare: after http_req() query=[/client/v4/zones/<redacted_zone>/dns_records?type=A&name=<redacted_name>.<redacted_domain>&order=name&direction=asc] header=[User-Agent: Mozilla/5.0 (X11; Linux x86_64; rv:10.0) Gecko/20100101 Firefox/109.0^M Authorization: Bearer <redacted_bearer>^M Content-T [... cut ...]
Jan 30 16:54:52 ASUS4 user.debug ddns-update[4569]: *** update: cru d ddns2
Jan 30 16:54:52 ASUS4 user.debug ddns-update[4569]: *** update: cru d ddnsf2
Jan 30 16:54:52 ASUS4 user.debug ddns-update[4569]: *** update: msg=''
Jan 30 16:54:52 ASUS4 user.debug ddns-update[4569]: *** update: IN num=[3] dirty=[0] force=[0]
Jan 30 16:54:52 ASUS4 user.debug ddns-update[4569]: *** update: cru d ddns3
Jan 30 16:54:52 ASUS4 user.debug ddns-update[4569]: *** update: cru d ddnsf3
Jan 30 16:54:52 ASUS4 user.debug ddns-update[4569]: *** update: msg=''
root@ASUS4:/tmp/home/root#

```

pedro0311, 2025-01-30:
Could you please do the same with V171 (no dump)?

Vlad-Star, 2025-01-31:
v171 no dump
```
root@ASUS4:/tmp/home/root# awk -f redaction.awk dict.txt /var/log/messages | egrep 'mdu|ddns|curl|wget'
Jan 30 21:40:13 ASUS4 user.debug preinit[1]: *** stop_ddns: IN
Jan 30 21:40:13 ASUS4 user.info preinit[1]: ddns is stopped
Jan 30 21:40:13 ASUS4 user.debug preinit[1]: *** start_ddns: IN
Jan 30 21:40:13 ASUS4 user.info preinit[1]: ddns is started
Jan 30 21:40:13 ASUS4 user.debug ddns-update[1874]: *** ddns_update_main: args:
Jan 30 21:40:13 ASUS4 user.debug ddns-update[1874]: *** update: IN num=[0] dirty=[0] force=[0]
Jan 30 21:40:13 ASUS4 user.debug ddns-update[1874]: *** update: cru d ddns0
Jan 30 21:40:13 ASUS4 user.debug ddns-update[1874]: *** update: cru d ddnsf0
Jan 30 21:40:13 ASUS4 user.debug ddns-update[1874]: *** update: !f_exist(/var/lib/mdu/ddnsx0.msg) - creating ...
Jan 30 21:40:13 ASUS4 user.debug ddns-update[1874]: *** mdu <<<<<<<
Jan 30 21:40:13 ASUS4 daemon.debug mdu[1883]: *** main: IN v2.1 [libcurl]
Jan 30 21:40:13 ASUS4 daemon.debug mdu[1883]: *** main: checking for no WAN mode - true, using custom interface: br1
Jan 30 21:40:13 ASUS4 daemon.debug mdu[1883]: *** check_cookie: IN
Jan 30 21:40:13 ASUS4 daemon.debug mdu[1883]: *** read_tmaddr: IN cachename: /var/lib/mdu/ddnsx0.cache
Jan 30 21:40:13 ASUS4 daemon.debug mdu[1883]: *** check_cookie: no cookie
Jan 30 21:40:13 ASUS4 daemon.debug mdu[1883]: *** main: proceeding DDNS server update [service: cloudflare ] ...
Jan 30 21:40:13 ASUS4 daemon.debug mdu[1883]: *** update_cloudflare: IN
Jan 30 21:40:13 ASUS4 daemon.debug mdu[1883]: *** _http_req: IN host=[api.cloudflare.com] query=[/client/v4/zones/<redacted_zone>/dns_records?type=A&name=<redacted_name>.<redacted_domain>&order=name&direction=asc] ssl=[1] header=[User-Agent: Mozilla/5.0 (X11; Linux x86_64; rv:10.0) Gecko/20100101 Firefox/109.0^M Authorization: Bearer <redacted_bearer>^M Content-Type: application/json^M Cache-Control: no-cache] auth=[0] data=[(null)] req=[GET] ifname=[br1]
Jan 30 21:40:13 ASUS4 daemon.debug mdu[1883]: *** _http_req: resolving IP of server api.cloudflare.com ...
Jan 30 21:40:13 ASUS4 daemon.debug mdu[1883]: *** curl_resolve_ip: IN
Jan 30 21:40:13 ASUS4 daemon.debug mdu[1883]: *** curl_setup: IN
Jan 30 21:40:13 ASUS4 daemon.debug mdu[1883]: *** curl_setup: OUT
Jan 30 21:40:13 ASUS4 daemon.debug mdu[1883]: *** curl_resolve_ip: after curl_setup()
Jan 30 21:40:13 ASUS4 daemon.debug mdu[1883]: *** curl_resolve_ip: after CURLOPT_URL
Jan 30 21:40:13 ASUS4 daemon.debug mdu[1883]: *** curl_resolve_ip: after CURLOPT_HTTPHEADER
Jan 30 21:40:13 ASUS4 daemon.debug mdu[1883]: *** curl_resolve_ip: after CURLOPT_INTERFACE
Jan 30 21:40:13 ASUS4 daemon.debug mdu[1883]: *** curl_resolve_ip: before curl_easy_perform trys=[4]
Jan 30 21:40:14 ASUS4 daemon.debug mdu[1883]: *** curl_resolve_ip: after curl_easy_perform r=[0] trys=[4]
Jan 30 21:40:14 ASUS4 daemon.debug mdu[1883]: *** curl_cleanup: IN
Jan 30 21:40:14 ASUS4 daemon.debug mdu[1883]: *** curl_cleanup: after curl_dfile
Jan 30 21:40:14 ASUS4 daemon.debug mdu[1883]: *** curl_cleanup: after curl_easy_cleanup()
Jan 30 21:40:14 ASUS4 daemon.debug mdu[1883]: *** curl_cleanup: OUT
Jan 30 21:40:14 ASUS4 daemon.debug mdu[1883]: *** curl_resolve_ip: OUT IP=[104.19.192.175]
Jan 30 21:40:14 ASUS4 daemon.debug mdu[1883]: *** _http_req: IP=[104.19.192.175] of host=[api.cloudflare.com]
Jan 30 21:40:14 ASUS4 daemon.debug mdu[1883]: *** _http_req: after curl_resolve_ip
Jan 30 21:40:14 ASUS4 daemon.debug mdu[1883]: *** _http_req: after curl_wbuf
Jan 30 21:40:14 ASUS4 daemon.debug mdu[1883]: *** curl_setup: IN
Jan 30 21:40:14 ASUS4 daemon.debug mdu[1883]: *** curl_setup: OUT
Jan 30 21:40:14 ASUS4 daemon.debug mdu[1883]: *** _http_req: after curl_setup()
Jan 30 21:40:14 ASUS4 daemon.debug mdu[1883]: *** _http_req: after CURLOPT_URL
Jan 30 21:40:15 ASUS4 daemon.debug mdu[1883]: *** _http_req: after CURLOPT_HTTPHEADER
Jan 30 21:40:15 ASUS4 daemon.debug mdu[1883]: *** _http_req: after CURLOPT_INTERFACE
Jan 30 21:40:15 ASUS4 daemon.debug mdu[1883]: *** _http_req: after CURLOPT_HTTPAUTH
Jan 30 21:40:15 ASUS4 daemon.debug mdu[1883]: *** _http_req: after CURLOPT_WRITEDATA
Jan 30 21:40:15 ASUS4 daemon.debug mdu[1883]: *** _http_req: after CURLOPT_UPLOAD
Jan 30 21:40:15 ASUS4 daemon.debug mdu[1883]: *** _http_req: after route_adddel()
Jan 30 21:40:15 ASUS4 daemon.debug mdu[1883]: *** _http_req: before curl_easy_perform trys=[4]
Jan 30 21:40:16 ASUS4 daemon.debug mdu[1883]: *** _http_req: after curl_easy_perform r=[0] trys=[4]
Jan 30 21:40:16 ASUS4 daemon.debug mdu[1883]: *** _http_req: before curl_slist_free_all()
Jan 30 21:40:16 ASUS4 daemon.debug mdu[1883]: *** _http_req: connected, response=[200]
Jan 30 21:40:16 ASUS4 daemon.debug mdu[1883]: *** curl_cleanup: IN
Jan 30 21:40:16 ASUS4 daemon.debug mdu[1883]: *** curl_cleanup: after curl_dfile
Jan 30 21:40:16 ASUS4 daemon.debug mdu[1883]: *** curl_cleanup: after curl_easy_cleanup()
Jan 30 21:40:16 ASUS4 daemon.debug mdu[1883]: *** curl_cleanup: OUT
Jan 30 21:40:16 ASUS4 daemon.debug mdu[1883]: *** update_cloudflare: after http_req() query=[/client/v4/zones/<redacted_zone>/dns_records?type=A&name=<redacted_name>.<redacted_domain>&order=name&direction=asc] header=[User-Agent: Mozilla/5.0 (X11; Linux x86_64; rv:10.0) Gecko/20100101 Firefox/109.0^M Authorization: Bearer <redacted_bearer>^M Content-Type: application/json^M Cache-Control: no-cache]
Jan 30 21:40:16 ASUS4 daemon.debug mdu[1883]: *** update_cloudflare: after http_req(1)
[... cut ...]
Jan 30 21:40:19 ASUS4 user.debug ddns-update[1874]: *** update: cru d ddns2
Jan 30 21:40:19 ASUS4 user.debug ddns-update[1874]: *** update: cru d ddnsf2
Jan 30 21:40:19 ASUS4 user.debug ddns-update[1874]: *** update: msg=''
Jan 30 21:40:19 ASUS4 user.debug ddns-update[1874]: *** update: IN num=[3] dirty=[0] force=[0]
Jan 30 21:40:19 ASUS4 user.debug ddns-update[1874]: *** update: cru d ddns3
Jan 30 21:40:19 ASUS4 user.debug ddns-update[1874]: *** update: cru d ddnsf3
Jan 30 21:40:19 ASUS4 user.debug ddns-update[1874]: *** update: msg=''
root@ASUS4:/tmp/home/root#
```

Vlad-Star, 2025-01-31:
also v171 no dump
```
root@ASUS4:/tmp/home/root# awk -f redaction.awk dict.txt /var/lib/mdu/ddnsx0.cache
1738291219,<redacted_ip>
root@ASUS4:/tmp/home/root# awk -f redaction.awk dict.txt /var/lib/mdu/ddnsx0.conf
user
pass <redacted_bearer>
host pi.<redacted_domain>
addr @1
mx
backmx 0
wildcard 0
url <redacted_zone>
ahash <redacted_zone>
msg /var/lib/mdu/ddnsx0.msg
cookie /var/lib/mdu/ddnsx0.cache
addrcache ddnsx0.extip
root@ASUS4:/tmp/home/root# awk -f redaction.awk dict.txt /var/lib/mdu/ddnsx0.extip
401,<redacted_ip>
root@ASUS4:/tmp/home/root# awk -f redaction.awk dict.txt /var/lib/mdu/ddnsx0.msg
Unknown response (-1).
#RETRY 12 1
root@ASUS4:/tmp/home/root#

```

pedro0311, 2025-01-31:
> I am wondering why this line in .conf file doesn't have a full path (i.e. /var/lib/mdu/ddnsx0.extip)

This is how it was designed back then

AFAICS, there are no differences between version with (write) callback function and just ordinary buffer - the same problems.
Please test V175 without dump: https://bitbucket.org/pedro311/freshtomato-arm/downloads/freshtomato-RT-AC68U-K26ARM-2024.5.175-beta-mdu-log-AIO-64K.trx

pedro0311, 2025-01-31:
OK so we know now that:
1. IP of CF is resolved,
2. Then we have also (probably good) respond from CF of our first request,
3. cloudflare_errorcheck() returns OK,
4. next get_address() returns properly IP,
5. and here is the problem...

So let's test V176: https://bitbucket.org/pedro311/freshtomato-arm/downloads/freshtomato-RT-AC68U-K26ARM-2024.5.176-beta-mdu-log-AIO-64K.trx
Thanks.

Vlad-Star, 2025-01-31:
yep. can you copy the current **mdu.c** to the downloads folder, please? 

v176 no dump
```
root@ASUS4:/tmp/home/root# awk -f redaction.awk dict.txt /var/log/messages | egrep 'mdu|ddns|curl'
Jan 31 12:34:27 ASUS4 user.debug preinit[1]: *** stop_ddns: IN
Jan 31 12:34:27 ASUS4 user.info preinit[1]: ddns is stopped
Jan 31 12:34:27 ASUS4 user.debug preinit[1]: *** start_ddns: IN
Jan 31 12:34:28 ASUS4 user.info preinit[1]: ddns is started
Jan 31 12:34:28 ASUS4 user.debug ddns-update[2200]: *** ddns_update_main: args:
Jan 31 12:34:28 ASUS4 user.debug ddns-update[2200]: *** update: IN num=[0] dirty=[0] force=[0]
Jan 31 12:34:28 ASUS4 user.debug ddns-update[2200]: *** update: cru d ddns0
Jan 31 12:34:28 ASUS4 user.debug ddns-update[2200]: *** update: cru d ddnsf0
Jan 31 12:34:28 ASUS4 user.debug ddns-update[2200]: *** update: !f_exist(/var/lib/mdu/ddnsx0.msg) - creating ...
Jan 31 12:34:28 ASUS4 user.debug ddns-update[2200]: *** mdu <<<<<<<
Jan 31 12:34:28 ASUS4 daemon.debug mdu[2209]: *** main: IN v2.1 [libcurl]
Jan 31 12:34:28 ASUS4 daemon.debug mdu[2209]: *** main: checking for no WAN mode - true, using custom interface: br1
Jan 31 12:34:28 ASUS4 daemon.debug mdu[2209]: *** check_cookie: IN
Jan 31 12:34:28 ASUS4 daemon.debug mdu[2209]: *** read_tmaddr: IN cachename: /var/lib/mdu/ddnsx0.cache
Jan 31 12:34:28 ASUS4 daemon.debug mdu[2209]: *** check_cookie: no cookie
Jan 31 12:34:28 ASUS4 daemon.debug mdu[2209]: *** main: proceeding DDNS server update [service: cloudflare ] ...
Jan 31 12:34:28 ASUS4 daemon.debug mdu[2209]: *** update_cloudflare: IN
Jan 31 12:34:28 ASUS4 daemon.debug mdu[2209]: *** _http_req: IN host=[api.cloudflare.com] query=[/client/v4/zones/<redacted_zone>/dns_records?type=A&name=<redacted_name>.<redacted_domain>&order=name&direction=asc] ssl=[1] header=[User-Agent: Mozilla/5.0 (X11; Linux x86_64; rv:10.0) Gecko/20100101 Firefox/109.0^M Authorization: Bearer <redacted_bearer>^M Content-Type: application/json^M Cache-Control: no-cache] auth=[0] data=[(null)] req=[GET] ifname=[br1]
Jan 31 12:34:28 ASUS4 daemon.debug mdu[2209]: *** _http_req: resolving IP of server api.cloudflare.com ...
Jan 31 12:34:28 ASUS4 daemon.debug mdu[2209]: *** curl_resolve_ip: IN
Jan 31 12:34:28 ASUS4 daemon.debug mdu[2209]: *** curl_setup: IN
Jan 31 12:34:28 ASUS4 daemon.debug mdu[2209]: *** curl_setup: OUT
Jan 31 12:34:28 ASUS4 daemon.debug mdu[2209]: *** curl_resolve_ip: after curl_setup()
Jan 31 12:34:28 ASUS4 daemon.debug mdu[2209]: *** curl_resolve_ip: after CURLOPT_URL
Jan 31 12:34:28 ASUS4 daemon.debug mdu[2209]: *** curl_resolve_ip: after CURLOPT_HTTPHEADER
Jan 31 12:34:28 ASUS4 daemon.debug mdu[2209]: *** curl_resolve_ip: after CURLOPT_INTERFACE
Jan 31 12:34:28 ASUS4 daemon.debug mdu[2209]: *** curl_resolve_ip: before curl_easy_perform trys=[4]
Jan 31 12:34:29 ASUS4 daemon.debug mdu[2209]: *** curl_resolve_ip: after curl_easy_perform r=[0] trys=[4]
Jan 31 12:34:29 ASUS4 daemon.debug mdu[2209]: *** curl_cleanup: IN
Jan 31 12:34:29 ASUS4 daemon.debug mdu[2209]: *** curl_cleanup: after curl_dfile
Jan 31 12:34:29 ASUS4 daemon.debug mdu[2209]: *** curl_cleanup: after curl_easy_cleanup()
Jan 31 12:34:29 ASUS4 daemon.debug mdu[2209]: *** curl_cleanup: OUT
Jan 31 12:34:29 ASUS4 daemon.debug mdu[2209]: *** curl_resolve_ip: OUT IP=[104.19.192.175]
Jan 31 12:34:29 ASUS4 daemon.debug mdu[2209]: *** _http_req: IP=[104.19.192.175] of host=[api.cloudflare.com]
Jan 31 12:34:29 ASUS4 daemon.debug mdu[2209]: *** _http_req: after curl_resolve_ip
Jan 31 12:34:29 ASUS4 daemon.debug mdu[2209]: *** _http_req: after curl_wbuf
Jan 31 12:34:29 ASUS4 daemon.debug mdu[2209]: *** curl_setup: IN
Jan 31 12:34:29 ASUS4 daemon.debug mdu[2209]: *** curl_setup: OUT
Jan 31 12:34:29 ASUS4 daemon.debug mdu[2209]: *** _http_req: after curl_setup()
Jan 31 12:34:29 ASUS4 daemon.debug mdu[2209]: *** _http_req: after CURLOPT_URL
Jan 31 12:34:29 ASUS4 daemon.debug mdu[2209]: *** _http_req: after CURLOPT_HTTPHEADER
Jan 31 12:34:29 ASUS4 daemon.debug mdu[2209]: *** _http_req: after CURLOPT_INTERFACE
Jan 31 12:34:29 ASUS4 daemon.debug mdu[2209]: *** _http_req: after CURLOPT_HTTPAUTH
Jan 31 12:34:29 ASUS4 daemon.debug mdu[2209]: *** _http_req: after CURLOPT_WRITEDATA
Jan 31 12:34:29 ASUS4 daemon.debug mdu[2209]: *** _http_req: after CURLOPT_UPLOAD
Jan 31 12:34:29 ASUS4 daemon.debug mdu[2209]: *** _http_req: after route_adddel()
Jan 31 12:34:29 ASUS4 daemon.debug mdu[2209]: *** _http_req: before curl_easy_perform trys=[4]
Jan 31 12:34:30 ASUS4 daemon.debug mdu[2209]: *** _http_req: after curl_easy_perform r=[0] trys=[4]
Jan 31 12:34:31 ASUS4 daemon.debug mdu[2209]: *** _http_req: before curl_slist_free_all()
Jan 31 12:34:31 ASUS4 daemon.debug mdu[2209]: *** _http_req: connected, response=[200]
Jan 31 12:34:31 ASUS4 daemon.debug mdu[2209]: *** curl_cleanup: IN
Jan 31 12:34:31 ASUS4 daemon.debug mdu[2209]: *** curl_cleanup: after curl_dfile
Jan 31 12:34:31 ASUS4 daemon.debug mdu[2209]: *** curl_cleanup: after curl_easy_cleanup()
Jan 31 12:34:31 ASUS4 daemon.debug mdu[2209]: *** curl_cleanup: OUT
Jan 31 12:34:31 ASUS4 daemon.debug mdu[2209]: *** update_cloudflare: after http_req() query=[/client/v4/zones/<redacted_zone>/dns_records?type=A&name=<redacted_name>.<redacted_domain>&order=name&direction=asc] header=[User-Agent: Mozilla/5.0 (X11; Linux x86_64; rv:10.0) Gecko/20100101 Firefox/109.0^M Authorization: Bearer <redacted_bearer>^M Content-Type: application/json^M Cache-Control: no-cache] return=[200]
[... cut ...]
Jan 31 12:34:33 ASUS4 user.debug ddns-update[2200]: *** update: cru d ddns2
Jan 31 12:34:33 ASUS4 user.debug ddns-update[2200]: *** update: cru d ddnsf2
Jan 31 12:34:33 ASUS4 user.debug ddns-update[2200]: *** update: msg=''
Jan 31 12:34:33 ASUS4 user.debug ddns-update[2200]: *** update: IN num=[3] dirty=[0] force=[0]
Jan 31 12:34:33 ASUS4 user.debug ddns-update[2200]: *** update: cru d ddns3
Jan 31 12:34:33 ASUS4 user.debug ddns-update[2200]: *** update: cru d ddnsf3
Jan 31 12:34:33 ASUS4 user.debug ddns-update[2200]: *** update: msg=''
root@ASUS4:/tmp/home/root#
```

Vlad-Star, 2025-01-31:
in the old **mdu.c** source I am seeing two definitions of the **save_cookie** function - is there a good reason for it? Is this still a case?

(line 31) `static void save_cookie(void);`
and
(line 1993)
```
static void save_cookie(void)
{
...

```

Vlad-Star, 2025-02-01:
Looks like it fails along these lines?
```
	memset(s, 0, sizeof(s));
	snprintf(s, sizeof(s), "%ld,%s", now, c);
	f_write_string(cookie, s, FW_NEWLINE, 0);
```

Vlad-Star, 2025-02-01:
Oh... actually... attempt to change the constant?

```
	const char *c;
...
	if ((c = get_address(0)) == NULL) {

```

pedro0311, 2025-02-01:
Please, try this: https://bitbucket.org/pedro311/freshtomato-arm/downloads/freshtomato-RT-AC68U-K26ARM-2024.5.177-beta-mdu-buffer-log-AIO-64K.trx

Vlad-Star, 2025-02-01:
v177
```
...
Feb  1 12:23:53 ASUS4 daemon.debug mdu[2439]: *** curl_setup: IN
Feb  1 12:23:53 ASUS4 daemon.debug mdu[2439]: *** curl_setup: OUT
Feb  1 12:23:53 ASUS4 daemon.debug mdu[2439]: *** _http_req: after CURLOPT_URL
Feb  1 12:23:53 ASUS4 daemon.debug mdu[2439]: *** _http_req: after CURLOPT_HTTPHEADER
Feb  1 12:23:54 ASUS4 daemon.debug mdu[2439]: *** _http_req: after CURLOPT_INTERFACE
Feb  1 12:23:54 ASUS4 daemon.debug mdu[2439]: *** _http_req: after CURLOPT_HTTPAUTH
Feb  1 12:23:54 ASUS4 daemon.debug mdu[2439]: *** _http_req: after CURLOPT_WRITEDATA
Feb  1 12:23:54 ASUS4 daemon.debug mdu[2439]: *** _http_req: after CURLOPT_UPLOAD
Feb  1 12:23:54 ASUS4 daemon.debug mdu[2439]: *** _http_req: after route_adddel()
Feb  1 12:23:54 ASUS4 daemon.debug mdu[2439]: *** _http_req: before curl_easy_perform trys=[4]
Feb  1 12:23:54 ASUS4 daemon.debug mdu[2439]: *** _http_req: after curl_easy_perform r=[0] trys=[4]
Feb  1 12:23:54 ASUS4 daemon.debug mdu[2439]: *** _http_req: before curl_slist_free_all()
Feb  1 12:23:55 ASUS4 daemon.debug mdu[2439]: *** _http_req: connected, response=[200]
Feb  1 12:23:55 ASUS4 daemon.debug mdu[2439]: *** curl_cleanup: IN
Feb  1 12:23:55 ASUS4 daemon.debug mdu[2439]: *** curl_cleanup: after curl_dfile
Feb  1 12:23:55 ASUS4 daemon.debug mdu[2439]: *** curl_cleanup: after curl_easy_cleanup()
Feb  1 12:23:55 ASUS4 daemon.debug mdu[2439]: *** curl_cleanup: OUT
Feb  1 12:23:55 ASUS4 daemon.debug mdu[2439]: *** get_address: OUT used myexternalip.com service; time,address (843,<redacted_ip>) saved to ddnsx0.extip q=[<redacted_ip>]
Feb  1 12:23:55 ASUS4 daemon.debug mdu[2439]: *** success_msg: IN
Feb  1 12:23:55 ASUS4 daemon.debug mdu[2439]: *** save_cookie: IN
Feb  1 12:23:55 ASUS4 daemon.debug mdu[2439]: *** save_cookie: after now < Y2K
Feb  1 12:23:55 ASUS4 daemon.debug mdu[2439]: *** save_cookie: after cookie
Feb  1 12:23:55 ASUS4 daemon.debug mdu[2439]: *** get_address: IN required=[0]
Feb  1 12:23:55 ASUS4 daemon.debug mdu[2439]: *** read_tmaddr: IN cachename: ddnsx0.extip
Feb  1 12:23:55 ASUS4 user.debug ddns-update[2430]: *** mdu >>>>>>> update: service: cloudflare config: /var/lib/mdu/ddnsx0.conf; exitcode: 1
Feb  1 12:23:55 ASUS4 user.debug ddns-update[2430]: *** update: add scheduler [external checker] ...
Feb  1 12:23:55 ASUS4 user.debug ddns-update[2430]: *** update: msg='retry n=12 errors=1'
Feb  1 12:23:55 ASUS4 user.debug ddns-update[2430]: *** update: cru a ddns0 35 * * * * ddns-update 0
Feb  1 12:23:55 ASUS4 user.debug ddns-update[2430]: *** update: OUT
Feb  1 12:23:55 ASUS4 user.debug ddns-update[2430]: *** update: IN num=[1] dirty=[0] force=[0]
Feb  1 12:23:55 ASUS4 user.debug ddns-update[2430]: *** update: cru d ddns1
Feb  1 12:23:55 ASUS4 user.debug ddns-update[2430]: *** update: cru d ddnsf1
Feb  1 12:23:55 ASUS4 user.debug ddns-update[2430]: *** update: msg=''
Feb  1 12:23:55 ASUS4 user.debug ddns-update[2430]: *** update: IN num=[2] dirty=[0] force=[0]
Feb  1 12:23:55 ASUS4 user.debug ddns-update[2430]: *** update: cru d ddns2
Feb  1 12:23:55 ASUS4 user.debug ddns-update[2430]: *** update: cru d ddnsf2
Feb  1 12:23:55 ASUS4 user.debug ddns-update[2430]: *** update: msg=''
Feb  1 12:23:55 ASUS4 user.debug ddns-update[2430]: *** update: IN num=[3] dirty=[0] force=[0]
Feb  1 12:23:55 ASUS4 user.debug ddns-update[2430]: *** update: cru d ddns3
Feb  1 12:23:55 ASUS4 user.debug ddns-update[2430]: *** update: cru d ddnsf3
Feb  1 12:23:55 ASUS4 user.debug ddns-update[2430]: *** update: msg=''

```

pedro0311, 2025-02-02:
Interesting. And no sigsegv or something?

pedro0311, 2025-02-02:
Another one: https://bitbucket.org/pedro311/freshtomato-arm/downloads/freshtomato-RT-AC68U-K26ARM-2024.5.178-beta-mdu-buffer-log-AIO-64K.trx Thanks.

Vlad-Star, 2025-02-02:
nope, no segfaults

v178
```
root@ASUS4:/tmp/home/root# awk -f redaction.awk dict.txt /var/log/messages | egrep 'mdu|ddns|curl'
Feb  2 09:57:50 ASUS4 user.debug preinit[1]: *** stop_ddns: IN
Feb  2 09:57:50 ASUS4 user.info preinit[1]: ddns is stopped
Feb  2 09:57:50 ASUS4 user.debug preinit[1]: *** start_ddns: IN
Feb  2 09:57:50 ASUS4 user.info preinit[1]: ddns is started
Feb  2 09:57:50 ASUS4 user.debug ddns-update[1831]: *** ddns_update_main: args:
Feb  2 09:57:50 ASUS4 user.debug ddns-update[1831]: *** update: IN num=[0] dirty=[0] force=[0]
Feb  2 09:57:50 ASUS4 user.debug ddns-update[1831]: *** update: cru d ddns0
Feb  2 09:57:50 ASUS4 user.debug ddns-update[1831]: *** update: cru d ddnsf0
Feb  2 09:57:50 ASUS4 user.debug ddns-update[1831]: *** update: !f_exist(/var/lib/mdu/ddnsx0.msg) - creating ...
Feb  2 09:57:50 ASUS4 user.debug ddns-update[1831]: *** mdu <<<<<<<
Feb  2 09:57:50 ASUS4 daemon.debug mdu[1840]: *** main: IN v2.1 [libcurl]
Feb  2 09:57:50 ASUS4 daemon.debug mdu[1840]: *** main: checking for no WAN mode - true, using custom interface: br1
Feb  2 09:57:50 ASUS4 daemon.debug mdu[1840]: *** check_cookie: IN
Feb  2 09:57:50 ASUS4 daemon.debug mdu[1840]: *** read_tmaddr: IN cachename: /var/lib/mdu/ddnsx0.cache
Feb  2 09:57:50 ASUS4 daemon.debug mdu[1840]: *** read_tmaddr: OUT return 0
Feb  2 09:57:50 ASUS4 daemon.debug mdu[1840]: *** check_cookie: no cookie
Feb  2 09:57:50 ASUS4 daemon.debug mdu[1840]: *** main: proceeding DDNS server update [service: cloudflare ] ...
Feb  2 09:57:50 ASUS4 daemon.debug mdu[1840]: *** update_cloudflare: IN
Feb  2 09:57:50 ASUS4 daemon.debug mdu[1840]: *** _http_req: IN host=[api.cloudflare.com] query=[/client/v4/zones/<redacted_zone>/dns_records?type=A&name=<redacted_name>.<redacted_domain>&order=name&direction=asc] ssl=[1] header=[User-Agent: Mozilla/5.0 (X11; Linux x86_64; rv:10.0) Gecko/20100101 Firefox/109.0^M Authorization: Bearer <redacted_bearer>^M Content-Type: application/json^M Cache-Control: no-cache] auth=[0] data=[(null)] req=[GET] ifname=[br1]
Feb  2 09:57:50 ASUS4 daemon.debug mdu[1840]: *** _http_req: resolving IP of server api.cloudflare.com ...
Feb  2 09:57:50 ASUS4 daemon.debug mdu[1840]: *** curl_resolve_ip: IN
Feb  2 09:57:50 ASUS4 daemon.debug mdu[1840]: *** curl_setup: IN
Feb  2 09:57:50 ASUS4 daemon.debug mdu[1840]: *** curl_resolve_ip: before curl_easy_perform trys=[4]
Feb  2 09:57:51 ASUS4 daemon.debug mdu[1840]: *** curl_resolve_ip: after curl_easy_perform r=[0] trys=[4]
Feb  2 09:57:51 ASUS4 daemon.debug mdu[1840]: *** curl_cleanup: IN
Feb  2 09:57:51 ASUS4 daemon.debug mdu[1840]: *** curl_cleanup: after curl_dfile
Feb  2 09:57:51 ASUS4 daemon.debug mdu[1840]: *** curl_cleanup: after curl_easy_cleanup()
Feb  2 09:57:51 ASUS4 daemon.debug mdu[1840]: *** curl_cleanup: OUT
Feb  2 09:57:51 ASUS4 daemon.debug mdu[1840]: *** curl_resolve_ip: OUT IP=[104.19.192.176]
Feb  2 09:57:51 ASUS4 daemon.debug mdu[1840]: *** _http_req: IP=[104.19.192.176] of host=[api.cloudflare.com]
Feb  2 09:57:51 ASUS4 daemon.debug mdu[1840]: *** _http_req: after curl_resolve_ip
Feb  2 09:57:51 ASUS4 daemon.debug mdu[1840]: *** _http_req: after curl_wbuf
Feb  2 09:57:51 ASUS4 daemon.debug mdu[1840]: *** curl_setup: IN
Feb  2 09:57:51 ASUS4 daemon.debug mdu[1840]: *** curl_setup: OUT
Feb  2 09:57:51 ASUS4 daemon.debug mdu[1840]: *** _http_req: after curl_setup()
Feb  2 09:57:51 ASUS4 daemon.debug mdu[1840]: *** _http_req: after CURLOPT_URL
Feb  2 09:57:51 ASUS4 daemon.debug mdu[1840]: *** _http_req: after CURLOPT_HTTPHEADER
Feb  2 09:57:52 ASUS4 daemon.debug mdu[1840]: *** _http_req: after CURLOPT_INTERFACE
Feb  2 09:57:52 ASUS4 daemon.debug mdu[1840]: *** _http_req: after CURLOPT_HTTPAUTH
Feb  2 09:57:52 ASUS4 daemon.debug mdu[1840]: *** _http_req: after CURLOPT_WRITEDATA
Feb  2 09:57:52 ASUS4 daemon.debug mdu[1840]: *** _http_req: after CURLOPT_UPLOAD
Feb  2 09:57:52 ASUS4 daemon.debug mdu[1840]: *** _http_req: after route_adddel()
Feb  2 09:57:52 ASUS4 daemon.debug mdu[1840]: *** _http_req: before curl_easy_perform trys=[4]
Feb  2 09:57:53 ASUS4 daemon.debug mdu[1840]: *** _http_req: after curl_easy_perform r=[0] trys=[4]
Feb  2 09:57:53 ASUS4 daemon.debug mdu[1840]: *** _http_req: before curl_slist_free_all()
Feb  2 09:57:53 ASUS4 daemon.debug mdu[1840]: *** _http_req: connected, response=[200]
Feb  2 09:57:53 ASUS4 daemon.debug mdu[1840]: *** curl_cleanup: IN
Feb  2 09:57:53 ASUS4 daemon.debug mdu[1840]: *** curl_cleanup: after curl_dfile
Feb  2 09:57:53 ASUS4 daemon.debug mdu[1840]: *** curl_cleanup: after curl_easy_cleanup()
Feb  2 09:57:53 ASUS4 daemon.debug mdu[1840]: *** curl_cleanup: OUT
Feb  2 09:57:53 ASUS4 daemon.debug mdu[1840]: *** update_cloudflare: after http_req() query=[/client/v4/zones/<redacted_zone>/dns_records?type=A&name=<redacted_name>.<redacted_domain>&order=name&direction=asc] header=[User-Agent: Mozilla/5.0 (X11; Linux x86_64; rv:10.0) Gecko/20100101 Firefox/109.0^M Authorization: Bearer <redacted_bearer>^M Content-Type: application/json^M Cache-Control: no-cache] return=[200]
Feb  2 09:57:53 ASUS4 daemon.debug mdu[1840]: *** update_cloudflare: before cloudflare_errorcheck(1)
Feb  2 09:57:53 ASUS4 daemon.debug mdu[1840]: *** cloudflare_errorcheck: IN code=[200]
Feb  2 09:57:53 ASUS4 daemon.debug mdu[1840]: *** cloudflare_errorcheck: after body[n] strlen(body)=[491] body=[{"result":[{"id":"<redacted_id>","name":"<redacted_name>.<redacted_domain>","type":"A","content":"<redacted_ip>","proxiable":true,"proxied":false,"ttl":1,"se [... cut ...]
Feb  2 09:57:55 ASUS4 user.debug ddns-update[1831]: *** update: msg=''
Feb  2 09:57:55 ASUS4 user.debug ddns-update[1831]: *** update: IN num=[3] dirty=[0] force=[0]
Feb  2 09:57:55 ASUS4 user.debug ddns-update[1831]: *** update: cru d ddns3
Feb  2 09:57:55 ASUS4 user.debug ddns-update[1831]: *** update: cru d ddnsf3
Feb  2 09:57:55 ASUS4 user.debug ddns-update[1831]: *** update: msg=''
root@ASUS4:/tmp/home/root#
```
v176 was able to get out of read_tmaddr() and get_address(), causing error in save_cookie(). v177/v178 seem to be failing within read_tmaddr().

Vlad-Star, 2025-02-02:
also v178
```
root@ASUS4:/tmp/home/root# awk -f redaction.awk dict.txt /var/lib/mdu/ddnsx0.extip
403,<redacted_ip>
root@ASUS4:/tmp/home/root#
```

Vlad-Star, 2025-02-02:
Sorry for another noob question - does the **f_read_string()** function auto-close the file handle after reading from it?

pedro0311, 2025-02-02:
> Sorry for another noob question - does the f_read_string() function auto-close the file handle after reading from it? Same question about f_write_string() for writing

Yes, here:
https://github.com/FreshTomato-Project/freshtomato-arm/blob/arm-master/release/src-rt-6.x.4708/router/shared/files.c#L74
https://github.com/FreshTomato-Project/freshtomato-arm/blob/arm-master/release/src-rt-6.x.4708/router/shared/files.c#L42

As you can see, I fixed path to ddnsxX.extip file, but it doesn't change anything.
I really am clueless now...
Will think about it a little later.

pedro0311, 2025-02-02:
In the meantime, you can test v179: https://bitbucket.org/pedro311/freshtomato-arm/downloads/freshtomato-RT-AC68U-K26ARM-2024.5.179-beta-mdu-buffer-log-AIO-64K.trx

Vlad-Star, 2025-02-02:
v179
```
root@ASUS4:/tmp/home/root# awk -f redaction.awk dict.txt /var/log/messages | egrep 'mdu|ddns|curl' | tail -50
Feb  2 13:05:44 ASUS4 daemon.debug mdu[1948]: *** _http_req: after CURLOPT_URL
Feb  2 13:05:44 ASUS4 daemon.debug mdu[1948]: *** _http_req: after CURLOPT_HTTPHEADER
Feb  2 13:05:44 ASUS4 daemon.debug mdu[1948]: *** _http_req: after CURLOPT_INTERFACE
Feb  2 13:05:44 ASUS4 daemon.debug mdu[1948]: *** _http_req: after CURLOPT_HTTPAUTH
Feb  2 13:05:44 ASUS4 daemon.debug mdu[1948]: *** _http_req: after CURLOPT_WRITEDATA
Feb  2 13:05:44 ASUS4 daemon.debug mdu[1948]: *** _http_req: after CURLOPT_UPLOAD
Feb  2 13:05:44 ASUS4 daemon.debug mdu[1948]: *** _http_req: after route_adddel()
Feb  2 13:05:44 ASUS4 daemon.debug mdu[1948]: *** _http_req: before curl_easy_perform trys=[4]
Feb  2 13:05:45 ASUS4 daemon.debug mdu[1948]: *** _http_req: after curl_easy_perform r=[0] trys=[4]
Feb  2 13:05:45 ASUS4 daemon.debug mdu[1948]: *** _http_req: before curl_slist_free_all()
Feb  2 13:05:45 ASUS4 daemon.debug mdu[1948]: *** _http_req: connected, response=[200]
Feb  2 13:05:45 ASUS4 daemon.debug mdu[1948]: *** curl_cleanup: IN
Feb  2 13:05:45 ASUS4 daemon.debug mdu[1948]: *** curl_cleanup: after curl_dfile
Feb  2 13:05:45 ASUS4 daemon.debug mdu[1948]: *** curl_cleanup: after curl_easy_cleanup()
Feb  2 13:05:45 ASUS4 daemon.debug mdu[1948]: *** curl_cleanup: OUT
Feb  2 13:05:45 ASUS4 daemon.debug mdu[1948]: *** get_address: OUT used trackip.net service; time,address (527,<redacted_ip>) saved to /var/lib/mdu/ddnsx0.extip q=[<redacted_ip>]
Feb  2 13:05:45 ASUS4 daemon.debug mdu[1948]: *** success_msg: IN
Feb  2 13:05:45 ASUS4 daemon.debug mdu[1948]: *** save_cookie: IN
Feb  2 13:05:45 ASUS4 daemon.debug mdu[1948]: *** save_cookie: after now < Y2K
Feb  2 13:05:45 ASUS4 daemon.debug mdu[1948]: *** save_cookie: after cookie
Feb  2 13:05:45 ASUS4 daemon.debug mdu[1948]: *** get_address: IN required=[0]
Feb  2 13:05:45 ASUS4 daemon.debug mdu[1948]: *** read_tmaddr: IN cachename: /var/lib/mdu/ddnsx0.extip
Feb  2 13:05:45 ASUS4 daemon.debug mdu[1948]: *** read_tmaddr: after f_read_string
Feb  2 13:05:45 ASUS4 daemon.debug mdu[1948]: *** read_tmaddr: s=527,<redacted_ip> tm=527 addr=<redacted_ip>
Feb  2 13:05:45 ASUS4 daemon.debug mdu[1948]: *** read_tmaddr: OUT return 1
Feb  2 13:05:45 ASUS4 daemon.debug mdu[1948]: *** get_address: OUT using cached address <redacted_ip> from /var/lib/mdu/ddnsx0.extip. Expires in 298 seconds
Feb  2 13:05:45 ASUS4 daemon.debug mdu[1948]: *** save_cookie: after get_address(0)
Feb  2 13:05:45 ASUS4 daemon.debug mdu[1948]: *** save_cookie: OUT cookie=[1738519545,<redacted_ip>]
Feb  2 13:05:45 ASUS4 daemon.debug mdu[1948]: *** success_msg: msg=[Update successful.]
Feb  2 13:05:45 ASUS4 daemon.debug mdu[1948]: *** success_msg: do_exit=[0]
Feb  2 13:05:45 ASUS4 daemon.debug mdu[1948]: *** update_cloudflare: after get_address(1) addr=[<redacted_ip>] r=[0]
Feb  2 13:05:45 ASUS4 daemon.debug mdu[1948]: *** update_cloudflare: after [found = strstr(body, find)]1
Feb  2 13:05:45 ASUS4 daemon.err mdu[1948]: Unknown response (-1).
Feb  2 13:05:45 ASUS4 user.debug ddns-update[1939]: *** mdu >>>>>>> update: service: cloudflare config: /var/lib/mdu/ddnsx0.conf; exitcode: 1
Feb  2 13:05:45 ASUS4 user.debug ddns-update[1939]: *** update: add scheduler [external checker] ...
Feb  2 13:05:45 ASUS4 user.debug ddns-update[1939]: *** update: msg='retry n=12 errors=1'
Feb  2 13:05:45 ASUS4 user.debug ddns-update[1939]: *** update: cru a ddns0 17 * * * * ddns-update 0
Feb  2 13:05:45 ASUS4 user.debug ddns-update[1939]: *** update: OUT
Feb  2 13:05:45 ASUS4 user.debug ddns-update[1939]: *** update: IN num=[1] dirty=[0] force=[0]
Feb  2 13:05:45 ASUS4 user.debug ddns-update[1939]: *** update: cru d ddns1
Feb  2 13:05:45 ASUS4 user.debug ddns-update[1939]: *** update: cru d ddnsf1
Feb  2 13:05:45 ASUS4 user.debug ddns-update[1939]: *** update: msg=''
Feb  2 13:05:45 ASUS4 user.debug ddns-update[1939]: *** update: IN num=[2] dirty=[0] force=[0]
Feb  2 13:05:45 ASUS4 user.debug ddns-update[1939]: *** update: cru d ddns2
Feb  2 13:05:45 ASUS4 user.debug ddns-update[1939]: *** update: cru d ddnsf2
Feb  2 13:05:45 ASUS4 user.debug ddns-update[1939]: *** update: msg=''
Feb  2 13:05:45 ASUS4 user.debug ddns-update[1939]: *** update: IN num=[3] dirty=[0] force=[0]
Feb  2 13:05:45 ASUS4 user.debug ddns-update[1939]: *** update: cru d ddns3
Feb  2 13:05:45 ASUS4 user.debug ddns-update[1939]: *** update: cru d ddnsf3
Feb  2 13:05:45 ASUS4 user.debug ddns-update[1939]: *** update: msg=''
root@ASUS4:/tmp/home/root# awk -f redaction.awk dict.txt /var/lib/mdu/ddnsx0.extip
527,<redacted_ip>
root@ASUS4:/tmp/home/root#

```

Vlad-Star, 2025-02-02:
with **debug_ddns=1**
```
root@ASUS4:/tmp/home/root# awk -f redaction.awk dict.txt /tmp/mdu0-cloudflare.txt
Feb 02 13:41:41 == I TLSv1.3 (OUT), TLS handshake, Client hello (1)
Feb 02 13:41:41 == I TLSv1.3 (IN), TLS handshake, Server hello (2)
Feb 02 13:41:41 == I TLSv1.3 (IN), TLS handshake, Encrypted Extensions (8)
Feb 02 13:41:41 == I TLSv1.3 (IN), TLS handshake, Certificate (11)
Feb 02 13:41:41 == I TLSv1.3 (IN), TLS handshake, CERT verify (15)
Feb 02 13:41:41 == I TLSv1.3 (IN), TLS handshake, Finished (20)
Feb 02 13:41:41 == I TLSv1.3 (OUT), TLS change cipher, Change cipher spec (1)
Feb 02 13:41:41 == I TLSv1.3 (OUT), TLS handshake, Finished (20)
Feb 02 13:41:41 => H GET /client/v4/zones/<redacted_zone>/dns_records?type=A&name=<redacted_name>.<redacted_domain>&order=name&direction=asc HTTP/1.1
Feb 02 13:41:41 => H Host: api.cloudflare.com
Feb 02 13:41:41 => H Accept: */*
Feb 02 13:41:41 => H User-Agent: Mozilla/5.0 (X11; Linux x86_64; rv:10.0) Gecko/20100101 Firefox/109.0
Feb 02 13:41:41 => H Authorization: Bearer <redacted_bearer>
Feb 02 13:41:41 => H Content-Type: application/json
Feb 02 13:41:41 => H Cache-Control: no-cache
Feb 02 13:41:41 => H
Feb 02 13:41:42 == I TLSv1.3 (IN), TLS handshake, Newsession Ticket (4)
Feb 02 13:41:42 == I TLSv1.3 (IN), TLS handshake, Newsession Ticket (4)
Feb 02 13:41:42 <= H HTTP/1.1 200 OK
Feb 02 13:41:42 <= H Date: Sun, 02 Feb 2025 18:41:42 GMT
Feb 02 13:41:42 <= H Content-Type: application/json
Feb 02 13:41:42 <= H Transfer-Encoding: chunked
Feb 02 13:41:42 <= H Connection: keep-alive
Feb 02 13:41:42 <= H CF-Ray: 90bc511d2d6faff8-ATL
Feb 02 13:41:42 <= H CF-Cache-Status: DYNAMIC
Feb 02 13:41:42 <= H Set-Cookie: __cflb=0H28vgHxwvgAQtjUGUFqYFDiSDreGJnUkoSFYBYDceh; SameSite=Lax; path=/; expires=Sun, 02-Feb-25 21:11:43 GMT; HttpOnly
Feb 02 13:41:42 <= H Vary: Accept-Encoding
Feb 02 13:41:42 <= H cf-auditlog-id: 0194c7f7-ee4d-759f-8312-fa99bc7397d3
Feb 02 13:41:42 <= H Set-Cookie: __cf_bm=IFXqlzVldZnoiRQqAp0.SHP9lscJowV.t8QByprUzTI-1738521702-1.0.1.1-tVuz73gCZ6eosQ8jNElpeo1M1_Jr3IrPOG8yruIvuWsvJFRLtGPH.nHUfmNIaCmUdyeHrAYF6cEzdRnrdDsNOw; path=/; expires=Sun, 02-Feb-25 19:11:42 GMT; domain=.api.cloudflare.com; HttpOnly; Secure; SameSite=None
Feb 02 13:41:42 <= H Set-Cookie: __cfruid=10976e808d41cf8d4913e779aec66575c7786d8e-1738521702; path=/; domain=.api.cloudflare.com; HttpOnly; Secure; SameSite=None
Feb 02 13:41:42 <= H Set-Cookie: _cfuvid=20nmQ3VxHc5Xl9iz.taUHlSVZ_mPM29EwtANUDGeGFg-1738521702221-0.0.1.1-604800000; path=/; domain=.api.cloudflare.com; HttpOnly; Secure; SameSite=None
Feb 02 13:41:42 <= H Server: cloudflare
Feb 02 13:41:42 <= H
Feb 02 13:41:42 <= D 1eb
Feb 02 13:41:42 <= D {"result":[{"id":"<redacted_id>","name":"<redacted_name>.<redacted_domain>","type":"A","content":"<redacted_ip>","proxiable":true,"proxied":false,"ttl":1,"settings":{},"meta":{"auto_added":false,"managed_by_apps":false,"managed_by_argo_tunnel":false},"comment":null,"tags":[],"created_on":"2024-01-26T11:58:17.151364Z","modified_on":"2025-01-30T17:2
Feb 02 13:41:42 <= D 3:49.058729Z"}],"success":true,"errors":[],"messages":[],"result_info":{"page":1,"per_page":100,"count":1,"total_count":1,"total_pages":1}}
Feb 02 13:41:42 == I Malformed encoding found in chunked-encoding
Feb 02 13:41:43 == I TLSv1.3 (OUT), TLS handshake, Client hello (1)
Feb 02 13:41:43 == I TLSv1.3 (IN), TLS handshake, Server hello (2)
Feb 02 13:41:43 == I TLSv1.3 (IN), TLS handshake, Encrypted Extensions (8)
Feb 02 13:41:43 == I TLSv1.3 (IN), TLS handshake, Certificate (11)
Feb 02 13:41:43 == I TLSv1.3 (IN), TLS handshake, CERT verify (15)
Feb 02 13:41:43 == I TLSv1.3 (IN), TLS handshake, Finished (20)
Feb 02 13:41:43 == I TLSv1.3 (OUT), TLS change cipher, Change cipher spec (1)
Feb 02 13:41:43 == I TLSv1.3 (OUT), TLS handshake, Finished (20)
Feb 02 13:41:43 => H GET /client/v4/zones/<redacted_zone>/dns_records?type=A&name=<redacted_name>.<redacted_domain>&order=name&direction=asc HTTP/1.1
Feb 02 13:41:43 => H Host: api.cloudflare.com
Feb 02 13:41:43 => H Accept: */*
Feb 02 13:41:43 => H User-Agent: Mozilla/5.0 (X11; Linux x86_64; rv:10.0) Gecko/20100101 Firefox/109.0
Feb 02 13:41:43 => H Authorization: Bearer <redacted_bearer>
Feb 02 13:41:43 => H Content-Type: application/json
Feb 02 13:41:43 => H Cache-Control: no-cache
Feb 02 13:41:43 => H
Feb 02 13:41:43 == I TLSv1.3 (IN), TLS handshake, Newsession Ticket (4)
Feb 02 13:41:43 == I TLSv1.3 (IN), TLS handshake, Newsession Ticket (4)
Feb 02 13:41:43 <= H HTTP/1.1 200 OK
Feb 02 13:41:43 <= H Date: Sun, 02 Feb 2025 18:41:43 GMT
Feb 02 13:41:43 <= H Content-Type: application/json
Feb 02 13:41:43 <= H Transfer-Encoding: chunked
Feb 02 13:41:43 <= H Connection: keep-alive
Feb 02 13:41:43 <= H CF-Ray: 90bc51281a78b03e-ATL
Feb 02 13:41:43 <= H CF-Cache-Status: DYNAMIC
Feb 02 13:41:43 <= H Set-Cookie: __cflb=0H28vgHxwvgAQtjUGUFqYFDiSDreGJnUm8qbZoBCqBj; SameSite=Lax; path=/; expires=Sun, 02-Feb-25 21:11:44 GMT; HttpOnly
Feb 02 13:41:43 <= H Vary: Accept-Encoding
Feb 02 13:41:43 <= H cf-auditlog-id: 0194c7f7-f52a-7829-98e9-8aa040b883a2
Feb 02 13:41:43 <= H Set-Cookie: __cf_bm=YBqRH.fw12yZwcin82ZykIq4J5IOr4PJ209tcc5MR9s-1738521703-1.0.1.1-.V_KMfXz3HEhLj0x25ZMLAyh8jTDyhQFNJEkq9gZvsOGCFNYLSFjqAM5XB4vaHODW7uMRJQ3j9FfTALCcUl8DA; path=/; expires=Sun, 02-Feb-25 19:11:43 GMT; domain=.api.cloudflare.com; HttpOnly; Secure; SameSite=None
[... cut ...]
Feb 02 13:41:43 <= D {"result":[{"id":"<redacted_id>","name":"<redacted_name>.<redacted_domain>","type":"A","content":"<redacted_ip>","proxiable":true,"proxied":false,"ttl":1,"settings":{},"meta":{"auto_added":false,"managed_by_apps":false,"managed_by_argo_tunnel":false},"comment":null,"tags":[],"created_on":"2024-01-26T11:58:17.151364Z","modified_on":"2025-01-30T17:2
Feb 02 13:41:43 <= D 3:49.058729Z"}],"success":true,"errors":[],"messages":[],"result_info":{"page":1,"per_page":100,"count":1,"total_count":1,"total_pages":1}}
Feb 02 13:41:43 == I Malformed encoding found in chunked-encoding

root@ASUS4:/tmp/home/root#
```
I think "content" doesn't get to the body variable

pedro0311, 2025-02-02:
> v179

Could you attach a log a little earlier, where it is:

`cloudflare_errorcheck: after body[n] /.../`

And this one is with dump enabled or disabled?

pedro0311, 2025-02-02:
> I think "content" doesn't get to the body variable

Let's see how it looks like there: https://bitbucket.org/pedro311/freshtomato-arm/downloads/freshtomato-RT-AC68U-K26ARM-2024.5.180-beta-mdu-buffer-log-AIO-64K.trx

Vlad-Star, 2025-02-02:
v179 with dump enabled
```
root@ASUS4:/tmp/home/root# nvram get debug_ddns
1
root@ASUS4:/tmp/home/root# awk -f redaction.awk dict.txt /var/log/messages
Feb  2 14:54:11 ASUS4 user.debug preinit[1]: *** stop_ddns: IN
Feb  2 14:54:12 ASUS4 user.info preinit[1]: ddns is stopped
Feb  2 14:54:12 ASUS4 user.debug preinit[1]: *** start_ddns: IN
Feb  2 14:54:12 ASUS4 user.info preinit[1]: ddns is started
Feb  2 14:54:12 ASUS4 user.debug ddns-update[23793]: *** ddns_update_main: args:
Feb  2 14:54:12 ASUS4 user.debug ddns-update[23793]: *** update: IN num=[0] dirty=[0] force=[0]
Feb  2 14:54:12 ASUS4 user.debug ddns-update[23793]: *** update: cru d ddns0
Feb  2 14:54:12 ASUS4 user.debug ddns-update[23793]: *** update: cru d ddnsf0
Feb  2 14:54:12 ASUS4 user.debug ddns-update[23793]: *** update: !f_exist(/var/lib/mdu/ddnsx0.msg) - creating ...
Feb  2 14:54:12 ASUS4 user.debug ddns-update[23793]: *** mdu <<<<<<<
Feb  2 14:54:12 ASUS4 daemon.debug mdu[23802]: *** main: IN v2.1 [libcurl]
Feb  2 14:54:12 ASUS4 daemon.debug mdu[23802]: *** main: checking for no WAN mode - true, using custom interface: br1
Feb  2 14:54:12 ASUS4 daemon.debug mdu[23802]: *** check_cookie: IN
Feb  2 14:54:12 ASUS4 daemon.debug mdu[23802]: *** read_tmaddr: IN cachename: /var/lib/mdu/ddnsx0.cache
Feb  2 14:54:12 ASUS4 daemon.debug mdu[23802]: *** read_tmaddr: OUT return 0
Feb  2 14:54:12 ASUS4 daemon.debug mdu[23802]: *** check_cookie: no cookie
Feb  2 14:54:12 ASUS4 daemon.debug mdu[23802]: *** main: proceeding DDNS server update [service: cloudflare ] ...
Feb  2 14:54:12 ASUS4 daemon.debug mdu[23802]: *** update_cloudflare: IN
Feb  2 14:54:12 ASUS4 daemon.debug mdu[23802]: *** _http_req: IN host=[api.cloudflare.com] query=[/client/v4/zones/<redacted_zone>/dns_records?type=A&name=<redacted_name>.<redacted_domain>&order=name&direction=asc] ssl=[1] header=[User-Agent: Mozilla/5.0 (X11; Linux x86_64; rv:10.0) Gecko/20100101 Firefox/109.0^M Authorization: Bearer <redacted_bearer>^M Content-Type: application/json^M Cache-Control: no-cache] auth=[0] data=[(null)] req=[GET] ifname=[br1]
Feb  2 14:54:12 ASUS4 daemon.debug mdu[23802]: *** _http_req: resolving IP of server api.cloudflare.com ...
Feb  2 14:54:12 ASUS4 daemon.debug mdu[23802]: *** curl_resolve_ip: IN
Feb  2 14:54:12 ASUS4 daemon.debug mdu[23802]: *** curl_setup: IN
Feb  2 14:54:12 ASUS4 daemon.debug mdu[23802]: *** curl_setup: OUT
Feb  2 14:54:12 ASUS4 daemon.debug mdu[23802]: *** curl_resolve_ip: after curl_setup()
Feb  2 14:54:12 ASUS4 daemon.debug mdu[23802]: *** curl_resolve_ip: after CURLOPT_URL
Feb  2 14:54:12 ASUS4 daemon.debug mdu[23802]: *** curl_resolve_ip: after CURLOPT_HTTPHEADER
Feb  2 14:54:12 ASUS4 daemon.debug mdu[23802]: *** curl_resolve_ip: after CURLOPT_INTERFACE
Feb  2 14:54:12 ASUS4 daemon.debug mdu[23802]: *** curl_resolve_ip: before curl_easy_perform trys=[4]
Feb  2 14:54:13 ASUS4 daemon.debug mdu[23802]: *** curl_resolve_ip: after curl_easy_perform r=[56] trys=[4]
Feb  2 14:54:13 ASUS4 daemon.debug mdu[23802]: *** curl_cleanup: IN
Feb  2 14:54:13 ASUS4 daemon.debug mdu[23802]: *** curl_cleanup: after curl_dfile
Feb  2 14:54:13 ASUS4 daemon.debug mdu[23802]: *** curl_cleanup: after curl_easy_cleanup()
Feb  2 14:54:13 ASUS4 daemon.debug mdu[23802]: *** curl_cleanup: OUT
Feb  2 14:54:13 ASUS4 daemon.debug mdu[23802]: *** curl_resolve_ip: OUT IP=[104.19.192.29]
Feb  2 14:54:13 ASUS4 daemon.debug mdu[23802]: *** _http_req: IP=[104.19.192.29] of host=[api.cloudflare.com]
Feb  2 14:54:13 ASUS4 daemon.debug mdu[23802]: *** _http_req: after curl_resolve_ip
Feb  2 14:54:13 ASUS4 daemon.debug mdu[23802]: *** _http_req: after curl_wbuf
Feb  2 14:54:13 ASUS4 daemon.debug mdu[23802]: *** curl_setup: IN
Feb  2 14:54:13 ASUS4 daemon.debug mdu[23802]: *** curl_setup: OUT
Feb  2 14:54:13 ASUS4 daemon.debug mdu[23802]: *** _http_req: after curl_setup()
Feb  2 14:54:13 ASUS4 daemon.debug mdu[23802]: *** _http_req: after CURLOPT_URL
Feb  2 14:54:13 ASUS4 daemon.debug mdu[23802]: *** _http_req: after CURLOPT_HTTPHEADER
Feb  2 14:54:13 ASUS4 daemon.debug mdu[23802]: *** _http_req: after CURLOPT_INTERFACE
Feb  2 14:54:13 ASUS4 daemon.debug mdu[23802]: *** _http_req: after CURLOPT_HTTPAUTH
Feb  2 14:54:13 ASUS4 daemon.debug mdu[23802]: *** _http_req: after CURLOPT_WRITEDATA
Feb  2 14:54:13 ASUS4 daemon.debug mdu[23802]: *** _http_req: after CURLOPT_UPLOAD
Feb  2 14:54:13 ASUS4 daemon.debug mdu[23802]: *** _http_req: after route_adddel()
Feb  2 14:54:14 ASUS4 daemon.debug mdu[23802]: *** _http_req: before curl_easy_perform trys=[4]
Feb  2 14:54:14 ASUS4 daemon.debug mdu[23802]: *** _http_req: after curl_easy_perform r=[56] trys=[4]
Feb  2 14:54:15 ASUS4 daemon.debug mdu[23802]: *** _http_req: before curl_slist_free_all()
Feb  2 14:54:15 ASUS4 daemon.debug mdu[23802]: *** _http_req: connected, response=[200]
Feb  2 14:54:15 ASUS4 daemon.debug mdu[23802]: *** curl_cleanup: IN
Feb  2 14:54:15 ASUS4 daemon.debug mdu[23802]: *** curl_cleanup: after curl_dfile
Feb  2 14:54:15 ASUS4 daemon.debug mdu[23802]: *** curl_cleanup: after curl_easy_cleanup()
Feb  2 14:54:15 ASUS4 daemon.debug mdu[23802]: *** curl_cleanup: OUT
Feb  2 14:54:15 ASUS4 daemon.debug mdu[23802]: *** update_cloudflare: after http_req() query=[/client/v4/zones/<redacted_zone>/dns_records?type=A&name=<redacted_name>.<redacted_domain>&order=name&direction=asc] header=[User-Agent: Mozilla/5.0 (X11; Linux x86_64; rv:10.0) Gecko/20100101 Firefox/109.0^M Authorization: Bearer <redacted_bearer>^M Content-Type: application/json^M [... cut ...]
Feb 02 14:54:14 <= D {"result":[{"id":"<redacted_id>","name":"<redacted_name>.<redacted_domain>","type":"A","content":"<redacted_ip>","proxiable":true,"proxied":false,"ttl":1,"settings":{},"meta":{"auto_added":false,"managed_by_apps":false,"managed_by_argo_tunnel":false},"comment":null,"tags":[],"created_on":"2024-01-26T11:58:17.151364Z","modified_on":"2025-01-30T17:2
Feb 02 14:54:14 <= D 3:49.058729Z"}],"success":true,"errors":[],"messages":[],"result_info":{"page":1,"per_page":100,"count":1,"total_count":1,"total_pages":1}}
Feb 02 14:54:14 == I Malformed encoding found in chunked-encoding

root@ASUS4:/tmp/home/root#
```

Vlad-Star, 2025-02-02:
v179 with dump disabled
```
root@ASUS4:/tmp/home/root# nvram unset debug_ddns
root@ASUS4:/tmp/home/root# nvram commit
root@ASUS4:/tmp/home/root# rm -f /var/lib/mdu/ddnsx0.*
root@ASUS4:/tmp/home/root# rm -f /tmp/mdu0-cloudflare.txt
root@ASUS4:/tmp/home/root# rm -f /var/log/messages
root@ASUS4:/tmp/home/root# awk -f redaction.awk dict.txt /var/log/messages
Feb  2 14:58:05 ASUS4 user.debug preinit[1]: *** stop_ddns: IN
Feb  2 14:58:05 ASUS4 user.info preinit[1]: ddns is stopped
Feb  2 14:58:05 ASUS4 user.debug preinit[1]: *** start_ddns: IN
Feb  2 14:58:05 ASUS4 user.info preinit[1]: ddns is started
Feb  2 14:58:05 ASUS4 user.debug ddns-update[25809]: *** ddns_update_main: args:
Feb  2 14:58:05 ASUS4 user.debug ddns-update[25809]: *** update: IN num=[0] dirty=[0] force=[0]
Feb  2 14:58:05 ASUS4 user.debug ddns-update[25809]: *** update: cru d ddns0
Feb  2 14:58:05 ASUS4 user.debug ddns-update[25809]: *** update: cru d ddnsf0
Feb  2 14:58:05 ASUS4 user.debug ddns-update[25809]: *** update: !f_exist(/var/lib/mdu/ddnsx0.msg) - creating ...
Feb  2 14:58:05 ASUS4 user.debug ddns-update[25809]: *** mdu <<<<<<<
Feb  2 14:58:05 ASUS4 daemon.debug mdu[25818]: *** main: IN v2.1 [libcurl]
Feb  2 14:58:05 ASUS4 daemon.debug mdu[25818]: *** main: checking for no WAN mode - true, using custom interface: br1
Feb  2 14:58:05 ASUS4 daemon.debug mdu[25818]: *** check_cookie: IN
Feb  2 14:58:05 ASUS4 daemon.debug mdu[25818]: *** read_tmaddr: IN cachename: /var/lib/mdu/ddnsx0.cache
Feb  2 14:58:05 ASUS4 daemon.debug mdu[25818]: *** read_tmaddr: OUT return 0
Feb  2 14:58:05 ASUS4 daemon.debug mdu[25818]: *** check_cookie: no cookie
Feb  2 14:58:05 ASUS4 daemon.debug mdu[25818]: *** main: proceeding DDNS server update [service: cloudflare ] ...
Feb  2 14:58:05 ASUS4 daemon.debug mdu[25818]: *** update_cloudflare: IN
Feb  2 14:58:05 ASUS4 daemon.debug mdu[25818]: *** _http_req: IN host=[api.cloudflare.com] query=[/client/v4/zones/<redacted_zone>/dns_records?type=A&name=<redacted_name>.<redacted_domain>&order=name&direction=asc] ssl=[1] header=[User-Agent: Mozilla/5.0 (X11; Linux x86_64; rv:10.0) Gecko/20100101 Firefox/109.0^M Authorization: Bearer <redacted_bearer>^M Content-Type: application/json^M Cache-Control: no-cache] auth=[0] data=[(null)] req=[GET] ifname=[br1]
Feb  2 14:58:05 ASUS4 daemon.debug mdu[25818]: *** _http_req: resolving IP of server api.cloudflare.com ...
Feb  2 14:58:05 ASUS4 daemon.debug mdu[25818]: *** curl_resolve_ip: IN
Feb  2 14:58:05 ASUS4 daemon.debug mdu[25818]: *** curl_setup: IN
Feb  2 14:58:05 ASUS4 daemon.debug mdu[25818]: *** curl_setup: OUT
Feb  2 14:58:05 ASUS4 daemon.debug mdu[25818]: *** curl_resolve_ip: after curl_setup()
Feb  2 14:58:05 ASUS4 daemon.debug mdu[25818]: *** curl_resolve_ip: after CURLOPT_URL
Feb  2 14:58:05 ASUS4 daemon.debug mdu[25818]: *** curl_resolve_ip: after CURLOPT_HTTPHEADER
Feb  2 14:58:05 ASUS4 daemon.debug mdu[25818]: *** curl_resolve_ip: after CURLOPT_INTERFACE
Feb  2 14:58:05 ASUS4 daemon.debug mdu[25818]: *** curl_resolve_ip: before curl_easy_perform trys=[4]
Feb  2 14:58:06 ASUS4 daemon.debug mdu[25818]: *** curl_resolve_ip: after curl_easy_perform r=[0] trys=[4]
Feb  2 14:58:06 ASUS4 daemon.debug mdu[25818]: *** curl_cleanup: IN
Feb  2 14:58:06 ASUS4 daemon.debug mdu[25818]: *** curl_cleanup: after curl_dfile
Feb  2 14:58:06 ASUS4 daemon.debug mdu[25818]: *** curl_cleanup: after curl_easy_cleanup()
Feb  2 14:58:06 ASUS4 daemon.debug mdu[25818]: *** curl_cleanup: OUT
Feb  2 14:58:06 ASUS4 daemon.debug mdu[25818]: *** curl_resolve_ip: OUT IP=[104.19.192.177]
Feb  2 14:58:06 ASUS4 daemon.debug mdu[25818]: *** _http_req: IP=[104.19.192.177] of host=[api.cloudflare.com]
Feb  2 14:58:06 ASUS4 daemon.debug mdu[25818]: *** _http_req: after curl_resolve_ip
Feb  2 14:58:06 ASUS4 daemon.debug mdu[25818]: *** _http_req: after curl_wbuf
Feb  2 14:58:06 ASUS4 daemon.debug mdu[25818]: *** curl_setup: IN
Feb  2 14:58:06 ASUS4 daemon.debug mdu[25818]: *** curl_setup: OUT
Feb  2 14:58:06 ASUS4 daemon.debug mdu[25818]: *** _http_req: after curl_setup()
Feb  2 14:58:06 ASUS4 daemon.debug mdu[25818]: *** _http_req: after CURLOPT_URL
Feb  2 14:58:06 ASUS4 daemon.debug mdu[25818]: *** _http_req: after CURLOPT_HTTPHEADER
Feb  2 14:58:06 ASUS4 daemon.debug mdu[25818]: *** _http_req: after CURLOPT_INTERFACE
Feb  2 14:58:06 ASUS4 daemon.debug mdu[25818]: *** _http_req: after CURLOPT_HTTPAUTH
Feb  2 14:58:06 ASUS4 daemon.debug mdu[25818]: *** _http_req: after CURLOPT_WRITEDATA
Feb  2 14:58:06 ASUS4 daemon.debug mdu[25818]: *** _http_req: after CURLOPT_UPLOAD
Feb  2 14:58:07 ASUS4 daemon.debug mdu[25818]: *** _http_req: after route_adddel()
Feb  2 14:58:07 ASUS4 daemon.debug mdu[25818]: *** _http_req: before curl_easy_perform trys=[4]
Feb  2 14:58:08 ASUS4 daemon.debug mdu[25818]: *** _http_req: after curl_easy_perform r=[0] trys=[4]
Feb  2 14:58:08 ASUS4 daemon.debug mdu[25818]: *** _http_req: before curl_slist_free_all()
Feb  2 14:58:08 ASUS4 daemon.debug mdu[25818]: *** _http_req: connected, response=[200]
Feb  2 14:58:08 ASUS4 daemon.debug mdu[25818]: *** curl_cleanup: IN
Feb  2 14:58:08 ASUS4 daemon.debug mdu[25818]: *** curl_cleanup: after curl_dfile
Feb  2 14:58:08 ASUS4 daemon.debug mdu[25818]: *** curl_cleanup: after curl_easy_cleanup()
Feb  2 14:58:08 ASUS4 daemon.debug mdu[25818]: *** curl_cleanup: OUT
[... cut ...]
Feb  2 14:58:11 ASUS4 user.debug ddns-update[25809]: *** update: cru d ddnsf2
Feb  2 14:58:11 ASUS4 user.debug ddns-update[25809]: *** update: msg=''
Feb  2 14:58:11 ASUS4 user.debug ddns-update[25809]: *** update: IN num=[3] dirty=[0] force=[0]
Feb  2 14:58:12 ASUS4 user.debug ddns-update[25809]: *** update: cru d ddns3
Feb  2 14:58:12 ASUS4 user.debug ddns-update[25809]: *** update: cru d ddnsf3
Feb  2 14:58:12 ASUS4 user.debug ddns-update[25809]: *** update: msg=''
root@ASUS4:/tmp/home/root#
```

Vlad-Star, 2025-02-02:
v180 with dump disabled
```
root@ASUS4:/tmp/home/root# awk -f redaction.awk dict.txt /var/log/messages | egrep 'mdu|ddns|curl'
Feb  2 15:04:51 ASUS4 user.debug preinit[1]: *** stop_ddns: IN
Feb  2 15:04:51 ASUS4 user.info preinit[1]: ddns is stopped
Feb  2 15:04:51 ASUS4 user.debug preinit[1]: *** start_ddns: IN
Feb  2 15:04:51 ASUS4 user.info preinit[1]: ddns is started
Feb  2 15:04:51 ASUS4 user.debug ddns-update[1802]: *** ddns_update_main: args:
Feb  2 15:04:51 ASUS4 user.debug ddns-update[1802]: *** update: IN num=[0] dirty=[0] force=[0]
Feb  2 15:04:51 ASUS4 user.debug ddns-update[1802]: *** update: cru d ddns0
Feb  2 15:04:51 ASUS4 user.debug ddns-update[1802]: *** update: cru d ddnsf0
Feb  2 15:04:51 ASUS4 user.debug ddns-update[1802]: *** update: !f_exist(/var/lib/mdu/ddnsx0.msg) - creating ...
Feb  2 15:04:51 ASUS4 user.debug ddns-update[1802]: *** mdu <<<<<<<
Feb  2 15:04:51 ASUS4 daemon.debug mdu[1811]: *** main: IN v2.1 [libcurl]
Feb  2 15:04:51 ASUS4 daemon.debug mdu[1811]: *** main: checking for no WAN mode - true, using custom interface: br1
Feb  2 15:04:51 ASUS4 daemon.debug mdu[1811]: *** check_cookie: IN
Feb  2 15:04:51 ASUS4 daemon.debug mdu[1811]: *** read_tmaddr: IN cachename: /var/lib/mdu/ddnsx0.cache
Feb  2 15:04:51 ASUS4 daemon.debug mdu[1811]: *** read_tmaddr: OUT return 0
Feb  2 15:04:51 ASUS4 daemon.debug mdu[1811]: *** check_cookie: no cookie
Feb  2 15:04:51 ASUS4 daemon.debug mdu[1811]: *** main: proceeding DDNS server update [service: cloudflare ] ...
Feb  2 15:04:51 ASUS4 daemon.debug mdu[1811]: *** update_cloudflare: IN
Feb  2 15:04:51 ASUS4 daemon.debug mdu[1811]: *** _http_req: IN host=[api.cloudflare.com] query=[/client/v4/zones/<redacted_zone>/dns_records?type=A&name=<redacted_name>.<redacted_domain>&order=name&direction=asc] ssl=[1] header=[User-Agent: Mozilla/5.0 (X11; Linux x86_64; rv:10.0) Gecko/20100101 Firefox/109.0^M Authorization: Bearer <redacted_bearer>^M Content-Type: application/json^M Cache-Control: no-cache] auth=[0] data=[(null)] req=[GET] ifname=[br1]
Feb  2 15:04:51 ASUS4 daemon.debug mdu[1811]: *** _http_req: resolving IP of server api.cloudflare.com ...
Feb  2 15:04:51 ASUS4 daemon.debug mdu[1811]: *** curl_resolve_ip: IN
Feb  2 15:04:51 ASUS4 daemon.debug mdu[1811]: *** curl_setup: IN
Feb  2 15:04:52 ASUS4 daemon.debug mdu[1811]: *** curl_resolve_ip: before curl_easy_perform trys=[4]
Feb  2 15:04:53 ASUS4 daemon.debug mdu[1811]: *** curl_resolve_ip: after curl_easy_perform r=[0] trys=[4]
Feb  2 15:04:53 ASUS4 daemon.debug mdu[1811]: *** curl_cleanup: IN
Feb  2 15:04:53 ASUS4 daemon.debug mdu[1811]: *** curl_cleanup: after curl_dfile
Feb  2 15:04:53 ASUS4 daemon.debug mdu[1811]: *** curl_cleanup: after curl_easy_cleanup()
Feb  2 15:04:53 ASUS4 daemon.debug mdu[1811]: *** curl_cleanup: OUT
Feb  2 15:04:53 ASUS4 daemon.debug mdu[1811]: *** curl_resolve_ip: OUT IP=[104.19.192.174]
Feb  2 15:04:53 ASUS4 daemon.debug mdu[1811]: *** _http_req: IP=[104.19.192.174] of host=[api.cloudflare.com]
Feb  2 15:04:53 ASUS4 daemon.debug mdu[1811]: *** _http_req: after curl_resolve_ip
Feb  2 15:04:53 ASUS4 daemon.debug mdu[1811]: *** curl_setup: IN
Feb  2 15:04:53 ASUS4 daemon.debug mdu[1811]: *** curl_setup: OUT
Feb  2 15:04:53 ASUS4 daemon.debug mdu[1811]: *** _http_req: connected, response=[200]
Feb  2 15:04:53 ASUS4 daemon.debug mdu[1811]: *** curl_cleanup: IN
Feb  2 15:04:53 ASUS4 daemon.debug mdu[1811]: *** curl_cleanup: after curl_dfile
Feb  2 15:04:53 ASUS4 daemon.debug mdu[1811]: *** curl_cleanup: after curl_easy_cleanup()
Feb  2 15:04:53 ASUS4 daemon.debug mdu[1811]: *** curl_cleanup: OUT
Feb  2 15:04:53 ASUS4 daemon.debug mdu[1811]: *** _http_req: OUT code=[200]
Feb  2 15:04:53 ASUS4 daemon.debug mdu[1811]: *** update_cloudflare: after http_req() query=[/client/v4/zones/<redacted_zone>/dns_records?type=A&name=<redacted_name>.<redacted_domain>&order=name&direction=asc] header=[User-Agent: Mozilla/5.0 (X11; Linux x86_64; rv:10.0) Gecko/20100101 Firefox/109.0^M Authorization: Bearer <redacted_bearer>^M Content-Type: application/json^M Cache-Control: no-cache] return=[200]
Feb  2 15:04:53 ASUS4 daemon.debug mdu[1811]: *** update_cloudflare: before cloudflare_errorcheck(1)
Feb  2 15:04:53 ASUS4 daemon.debug mdu[1811]: *** cloudflare_errorcheck: IN code=[200]
Feb  2 15:04:53 ASUS4 daemon.debug mdu[1811]: *** cloudflare_errorcheck: after body[n] strlen(body)=[491] body=[{"result":[{"id":"<redacted_id>","name":"<redacted_name>.<redacted_domain>","type":"A","content":"<redacted_ip>","proxiable":true,"proxied":false,"ttl":1,"settings":{},"meta":{"auto_added":false,"managed_by_apps":false,"managed_by_argo_tunnel":false},"comment":null,"tags":[],"created_on":"2024-01-26T11:58:17.151364Z","modified_on":"2025-01-30T17:23:49.058729Z"}],"success":true,"errors":[],"messages":[],"resu
Feb  2 15:04:53 ASUS4 daemon.debug mdu[1811]: *** cloudflare_errorcheck: code == 200
Feb  2 15:04:53 ASUS4 daemon.debug mdu[1811]: *** cloudflare_errorcheck: success:true
Feb  2 15:04:53 ASUS4 daemon.debug mdu[1811]: *** cloudflare_errorcheck: total_count:!0
Feb  2 15:04:53 ASUS4 daemon.debug mdu[1811]: *** update_cloudflare: after cloudflare_errorcheck(1) r=[0]
Feb  2 15:04:53 ASUS4 daemon.debug mdu[1811]: *** get_address: IN required=[1]
Feb  2 15:04:53 ASUS4 daemon.debug mdu[1811]: *** read_tmaddr: IN cachename: /var/lib/mdu/ddnsx0.extip
Feb  2 15:04:53 ASUS4 daemon.debug mdu[1811]: *** read_tmaddr: OUT return 0
[... cut ...]
Feb  2 15:04:54 ASUS4 user.debug ddns-update[1802]: *** update: msg=''
Feb  2 15:04:54 ASUS4 user.debug ddns-update[1802]: *** update: IN num=[3] dirty=[0] force=[0]
Feb  2 15:04:54 ASUS4 user.debug ddns-update[1802]: *** update: cru d ddns3
Feb  2 15:04:54 ASUS4 user.debug ddns-update[1802]: *** update: cru d ddnsf3
Feb  2 15:04:54 ASUS4 user.debug ddns-update[1802]: *** update: msg=''
root@ASUS4:/tmp/home/root# awk -f redaction.awk dict.txt /var/lib/mdu/ddnsx0.extip
404,<redacted_ip>
root@ASUS4:/tmp/home/root#
```

Vlad-Star, 2025-02-02:
v180 with dump enabled. Looks like it doesn't write to the cookie file due to error encoding, not sure if worth to review at all
```
root@ASUS4:/tmp/home/root# nvram set debug_ddns=1
root@ASUS4:/tmp/home/root# nvram commit
root@ASUS4:/tmp/home/root# rm -f /var/lib/mdu/ddnsx0.*
root@ASUS4:/tmp/home/root# rm -f /tmp/mdu0-cloudflare.txt
root@ASUS4:/tmp/home/root# rm -f /var/log/messages
root@ASUS4:/tmp/home/root# awk -f redaction.awk dict.txt /var/log/messages | egrep 'mdu|ddns|curl'
Feb  2 15:06:17 ASUS4 user.debug preinit[1]: *** stop_ddns: IN
Feb  2 15:06:17 ASUS4 user.info preinit[1]: ddns is stopped
Feb  2 15:06:17 ASUS4 user.debug preinit[1]: *** start_ddns: IN
Feb  2 15:06:17 ASUS4 user.info preinit[1]: ddns is started
Feb  2 15:06:17 ASUS4 user.debug ddns-update[2599]: *** ddns_update_main: args:
Feb  2 15:06:17 ASUS4 user.debug ddns-update[2599]: *** update: IN num=[0] dirty=[0] force=[0]
Feb  2 15:06:17 ASUS4 user.debug ddns-update[2599]: *** update: cru d ddns0
Feb  2 15:06:17 ASUS4 user.debug ddns-update[2599]: *** update: cru d ddnsf0
Feb  2 15:06:17 ASUS4 user.debug ddns-update[2599]: *** update: !f_exist(/var/lib/mdu/ddnsx0.msg) - creating ...
Feb  2 15:06:17 ASUS4 user.debug ddns-update[2599]: *** mdu <<<<<<<
Feb  2 15:06:17 ASUS4 daemon.debug mdu[2608]: *** main: IN v2.1 [libcurl]
Feb  2 15:06:17 ASUS4 daemon.debug mdu[2608]: *** main: checking for no WAN mode - true, using custom interface: br1
Feb  2 15:06:17 ASUS4 daemon.debug mdu[2608]: *** check_cookie: IN
Feb  2 15:06:17 ASUS4 daemon.debug mdu[2608]: *** read_tmaddr: IN cachename: /var/lib/mdu/ddnsx0.cache
Feb  2 15:06:17 ASUS4 daemon.debug mdu[2608]: *** read_tmaddr: OUT return 0
Feb  2 15:06:17 ASUS4 daemon.debug mdu[2608]: *** check_cookie: no cookie
Feb  2 15:06:17 ASUS4 daemon.debug mdu[2608]: *** main: proceeding DDNS server update [service: cloudflare ] ...
Feb  2 15:06:17 ASUS4 daemon.debug mdu[2608]: *** update_cloudflare: IN
Feb  2 15:06:17 ASUS4 daemon.debug mdu[2608]: *** _http_req: IN host=[api.cloudflare.com] query=[/client/v4/zones/<redacted_zone>/dns_records?type=A&name=<redacted_name>.<redacted_domain>&order=name&direction=asc] ssl=[1] header=[User-Agent: Mozilla/5.0 (X11; Linux x86_64; rv:10.0) Gecko/20100101 Firefox/109.0^M Authorization: Bearer <redacted_bearer>^M Content-Type: application/json^M Cache-Control: no-cache] auth=[0] data=[(null)] req=[GET] ifname=[br1]
Feb  2 15:06:17 ASUS4 daemon.debug mdu[2608]: *** _http_req: resolving IP of server api.cloudflare.com ...
Feb  2 15:06:17 ASUS4 daemon.debug mdu[2608]: *** curl_resolve_ip: IN
Feb  2 15:06:17 ASUS4 daemon.debug mdu[2608]: *** curl_setup: IN
Feb  2 15:06:17 ASUS4 daemon.debug mdu[2608]: *** curl_setup: OUT
Feb  2 15:06:17 ASUS4 daemon.debug mdu[2608]: *** curl_resolve_ip: after curl_setup()
Feb  2 15:06:17 ASUS4 daemon.debug mdu[2608]: *** curl_resolve_ip: after CURLOPT_URL
Feb  2 15:06:17 ASUS4 daemon.debug mdu[2608]: *** curl_resolve_ip: after CURLOPT_HTTPHEADER
Feb  2 15:06:17 ASUS4 daemon.debug mdu[2608]: *** curl_resolve_ip: after CURLOPT_INTERFACE
Feb  2 15:06:18 ASUS4 daemon.debug mdu[2608]: *** curl_resolve_ip: before curl_easy_perform trys=[4]
Feb  2 15:06:19 ASUS4 daemon.debug mdu[2608]: *** curl_resolve_ip: after curl_easy_perform r=[56] trys=[4]
Feb  2 15:06:19 ASUS4 daemon.debug mdu[2608]: *** curl_cleanup: IN
Feb  2 15:06:19 ASUS4 daemon.debug mdu[2608]: *** curl_cleanup: after curl_dfile
Feb  2 15:06:19 ASUS4 daemon.debug mdu[2608]: *** curl_cleanup: after curl_easy_cleanup()
Feb  2 15:06:19 ASUS4 daemon.debug mdu[2608]: *** curl_cleanup: OUT
Feb  2 15:06:19 ASUS4 daemon.debug mdu[2608]: *** curl_resolve_ip: OUT IP=[104.19.192.176]
Feb  2 15:06:19 ASUS4 daemon.debug mdu[2608]: *** _http_req: IP=[104.19.192.176] of host=[api.cloudflare.com]
Feb  2 15:06:19 ASUS4 daemon.debug mdu[2608]: *** _http_req: after curl_resolve_ip
Feb  2 15:06:19 ASUS4 daemon.debug mdu[2608]: *** curl_setup: IN
Feb  2 15:06:19 ASUS4 daemon.debug mdu[2608]: *** curl_setup: OUT
Feb  2 15:06:19 ASUS4 daemon.debug mdu[2608]: *** _http_req: connected, response=[200]
Feb  2 15:06:19 ASUS4 daemon.debug mdu[2608]: *** curl_cleanup: IN
Feb  2 15:06:19 ASUS4 daemon.debug mdu[2608]: *** curl_cleanup: after curl_dfile
Feb  2 15:06:19 ASUS4 daemon.debug mdu[2608]: *** curl_cleanup: after curl_easy_cleanup()
Feb  2 15:06:19 ASUS4 daemon.debug mdu[2608]: *** curl_cleanup: OUT
Feb  2 15:06:19 ASUS4 daemon.debug mdu[2608]: *** _http_req: OUT code=[200]
Feb  2 15:06:19 ASUS4 daemon.debug mdu[2608]: *** update_cloudflare: after http_req() query=[/client/v4/zones/<redacted_zone>/dns_records?type=A&name=<redacted_name>.<redacted_domain>&order=name&direction=asc] header=[User-Agent: Mozilla/5.0 (X11; Linux x86_64; rv:10.0) Gecko/20100101 Firefox/109.0^M Authorization: Bearer <redacted_bearer>^M Content-Type: application/json^M Cache-Control: no-cache] return=[200]
Feb  2 15:06:19 ASUS4 daemon.debug mdu[2608]: *** update_cloudflare: before cloudflare_errorcheck(1)
Feb  2 15:06:19 ASUS4 daemon.debug mdu[2608]: *** cloudflare_errorcheck: IN code=[200]
Feb  2 15:06:19 ASUS4 daemon.debug mdu[2608]: *** cloudflare_errorcheck: after body[n] strlen(body)=[352] body=[{"result":[{"id":"<redacted_id>","name":"<redacted_name>.<redacted_domain>","type":"A","content":"<redacted_ip>","proxiable":true,"proxied":false,"ttl":1,"settings":{},"meta":{"auto_added":false,"managed_by_apps":false,"managed_by_argo_tunnel":false},"comment":null,"tags":[],"created_on":"2024-01-26T11:58:17.151364Z","modifie [... cut ...]
Feb 02 15:06:19 <= D {"result":[{"id":"<redacted_id>","name":"<redacted_name>.<redacted_domain>","type":"A","content":"<redacted_ip>","proxiable":true,"proxied":false,"ttl":1,"settings":{},"meta":{"auto_added":false,"managed_by_apps":false,"managed_by_argo_tunnel":false},"comment":null,"tags":[],"created_on":"2024-01-26T11:58:17.151364Z","modified_on":"2025-01-30T17:2
Feb 02 15:06:19 <= D 3:49.058729Z"}],"success":true,"errors":[],"messages":[],"result_info":{"page":1,"per_page":100,"count":1,"total_count":1,"total_pages":1}}
Feb 02 15:06:19 == I Malformed encoding found in chunked-encoding
root@ASUS4:/tmp/home/root#
```

pedro0311, 2025-02-02:
Oh, damn, I added log not in the place where I should. Will soon upload v181.

pedro0311, 2025-02-02:
Done: https://bitbucket.org/pedro311/freshtomato-arm/downloads/freshtomato-RT-AC68U-K26ARM-2024.5.181-beta-mdu-buffer-log-AIO-64K.trx

Vlad-Star, 2025-02-02:
v181 no dump
```
root@ASUS4:/tmp/home/root# awk -f redaction.awk dict.txt /var/log/messages | egrep 'mdu|ddns|curl'
Feb  2 16:18:04 ASUS4 user.debug preinit[1]: *** stop_ddns: IN
Feb  2 16:18:04 ASUS4 user.info preinit[1]: ddns is stopped
Feb  2 16:18:04 ASUS4 user.debug preinit[1]: *** start_ddns: IN
Feb  2 16:18:04 ASUS4 user.info preinit[1]: ddns is started
Feb  2 16:18:04 ASUS4 user.debug ddns-update[1667]: *** ddns_update_main: args:
Feb  2 16:18:04 ASUS4 user.debug ddns-update[1667]: *** update: IN num=[0] dirty=[0] force=[0]
Feb  2 16:18:04 ASUS4 user.debug ddns-update[1667]: *** update: cru d ddns0
Feb  2 16:18:04 ASUS4 user.debug ddns-update[1667]: *** update: cru d ddnsf0
Feb  2 16:18:04 ASUS4 user.debug ddns-update[1667]: *** update: !f_exist(/var/lib/mdu/ddnsx0.msg) - creating ...
Feb  2 16:18:04 ASUS4 user.debug ddns-update[1667]: *** mdu <<<<<<<
Feb  2 16:18:04 ASUS4 daemon.debug mdu[1676]: *** main: IN v2.1 [libcurl]
Feb  2 16:18:04 ASUS4 daemon.debug mdu[1676]: *** main: checking for no WAN mode - true, using custom interface: br1
Feb  2 16:18:04 ASUS4 daemon.debug mdu[1676]: *** check_cookie: IN
Feb  2 16:18:04 ASUS4 daemon.debug mdu[1676]: *** read_tmaddr: IN cachename: /var/lib/mdu/ddnsx0.cache
Feb  2 16:18:04 ASUS4 daemon.debug mdu[1676]: *** read_tmaddr: OUT return 0
Feb  2 16:18:04 ASUS4 daemon.debug mdu[1676]: *** check_cookie: no cookie
Feb  2 16:18:04 ASUS4 daemon.debug mdu[1676]: *** main: proceeding DDNS server update [service: cloudflare ] ...
Feb  2 16:18:04 ASUS4 daemon.debug mdu[1676]: *** update_cloudflare: IN
Feb  2 16:18:04 ASUS4 daemon.debug mdu[1676]: *** _http_req: IN host=[api.cloudflare.com] query=[/client/v4/zones/<redacted_zone>/dns_records?type=A&name=<redacted_name>.<redacted_domain>&order=name&direction=asc] ssl=[1] header=[User-Agent: Mozilla/5.0 (X11; Linux x86_64; rv:10.0) Gecko/20100101 Firefox/109.0^M Authorization: Bearer <redacted_bearer>^M Content-Type: application/json^M Cache-Control: no-cache] auth=[0] data=[(null)] req=[GET] ifname=[br1]
Feb  2 16:18:04 ASUS4 daemon.debug mdu[1676]: *** _http_req: resolving IP of server api.cloudflare.com ...
Feb  2 16:18:04 ASUS4 daemon.debug mdu[1676]: *** curl_resolve_ip: IN
Feb  2 16:18:04 ASUS4 daemon.debug mdu[1676]: *** curl_setup: IN
Feb  2 16:18:04 ASUS4 daemon.debug mdu[1676]: *** curl_setup: OUT
Feb  2 16:18:04 ASUS4 daemon.debug mdu[1676]: *** curl_resolve_ip: after curl_setup()
Feb  2 16:18:04 ASUS4 daemon.debug mdu[1676]: *** curl_resolve_ip: after CURLOPT_URL
Feb  2 16:18:04 ASUS4 daemon.debug mdu[1676]: *** curl_resolve_ip: after CURLOPT_HTTPHEADER
Feb  2 16:18:04 ASUS4 daemon.debug mdu[1676]: *** curl_resolve_ip: after CURLOPT_INTERFACE
Feb  2 16:18:04 ASUS4 daemon.debug mdu[1676]: *** curl_resolve_ip: before curl_easy_perform trys=[4]
Feb  2 16:18:06 ASUS4 daemon.debug mdu[1676]: *** curl_resolve_ip: after curl_easy_perform r=[0] trys=[4]
Feb  2 16:18:06 ASUS4 daemon.debug mdu[1676]: *** curl_cleanup: IN
Feb  2 16:18:06 ASUS4 daemon.debug mdu[1676]: *** curl_cleanup: after curl_dfile
Feb  2 16:18:06 ASUS4 daemon.debug mdu[1676]: *** curl_cleanup: after curl_easy_cleanup()
Feb  2 16:18:06 ASUS4 daemon.debug mdu[1676]: *** curl_cleanup: OUT
Feb  2 16:18:06 ASUS4 daemon.debug mdu[1676]: *** curl_resolve_ip: OUT IP=[104.19.192.174]
Feb  2 16:18:06 ASUS4 daemon.debug mdu[1676]: *** _http_req: IP=[104.19.192.174] of host=[api.cloudflare.com]
Feb  2 16:18:06 ASUS4 daemon.debug mdu[1676]: *** _http_req: after curl_resolve_ip
Feb  2 16:18:06 ASUS4 daemon.debug mdu[1676]: *** curl_setup: IN
Feb  2 16:18:06 ASUS4 daemon.debug mdu[1676]: *** curl_setup: OUT
Feb  2 16:18:07 ASUS4 daemon.debug mdu[1676]: *** _http_req: connected, response=[200]
Feb  2 16:18:07 ASUS4 daemon.debug mdu[1676]: *** curl_cleanup: IN
Feb  2 16:18:07 ASUS4 daemon.debug mdu[1676]: *** curl_cleanup: after curl_dfile
Feb  2 16:18:07 ASUS4 daemon.debug mdu[1676]: *** curl_cleanup: after curl_easy_cleanup()
Feb  2 16:18:07 ASUS4 daemon.debug mdu[1676]: *** curl_cleanup: OUT
Feb  2 16:18:07 ASUS4 daemon.debug mdu[1676]: *** _http_req: OUT code=[200]
Feb  2 16:18:07 ASUS4 daemon.debug mdu[1676]: *** update_cloudflare: after http_req() query=[/client/v4/zones/<redacted_zone>/dns_records?type=A&name=<redacted_name>.<redacted_domain>&order=name&direction=asc] header=[User-Agent: Mozilla/5.0 (X11; Linux x86_64; rv:10.0) Gecko/20100101 Firefox/109.0^M Authorization: Bearer <redacted_bearer>^M Content-Type: application/json^M Cache-Control: no-cache] return=[200]
Feb  2 16:18:07 ASUS4 daemon.debug mdu[1676]: *** update_cloudflare: before cloudflare_errorcheck(1)
Feb  2 16:18:07 ASUS4 daemon.debug mdu[1676]: *** cloudflare_errorcheck: IN code=[200]
Feb  2 16:18:07 ASUS4 daemon.debug mdu[1676]: *** cloudflare_errorcheck: after body[n] strlen(body)=[491] body=[{"result":[{"id":"<redacted_id>","name":"<redacted_name>.<redacted_domain>","type":"A","content":"<redacted_ip>","proxiable":true,"proxied":false,"ttl":1,"settings":{},"meta":{"auto_added":false,"managed_by_apps":false,"managed_by_argo_tunnel":false},"comment":null,"tags":[],"created_on":"2024-01-26T11:58:17.151364Z","modified_on":"2025-01-30T17:23:49.058729Z"}],"success":true,"errors":[],"messages":[],"resu
Feb  2 16:18:07 ASUS4 daemon.debug mdu[1676]: *** cloudflare_errorcheck: code == 200
Feb  2 16:18:07 ASUS4 daemon.debug mdu[1676]: *** cloudflare_errorcheck: success:true
Feb  2 16:18:07 ASUS4 daemon.debug mdu[1676]: *** cloudflare_errorcheck: total_count:!0
[... cut ...]
Feb  2 16:18:07 ASUS4 user.debug ddns-update[1667]: *** update: msg=''
Feb  2 16:18:07 ASUS4 user.debug ddns-update[1667]: *** update: IN num=[3] dirty=[0] force=[0]
Feb  2 16:18:07 ASUS4 user.debug ddns-update[1667]: *** update: cru d ddns3
Feb  2 16:18:07 ASUS4 user.debug ddns-update[1667]: *** update: cru d ddnsf3
Feb  2 16:18:07 ASUS4 user.debug ddns-update[1667]: *** update: msg=''
root@ASUS4:/tmp/home/root# awk -f redaction.awk dict.txt /var/lib/mdu/ddnsx0.extip
385,<redacted_ip>
root@ASUS4:/tmp/home/root#
```

pedro0311, 2025-02-02:
Ha!
`Feb  2 16:18:07 ASUS4 daemon.debug mdu[1676]: *** update_cloudflare: body=[<redacted_ip>]`
As I thought, body (blob) is overwritten every time by get_address so I really don't know, how it worked at all.
Fix is on its way.

pedro0311, 2025-02-02:
https://bitbucket.org/pedro311/freshtomato-arm/downloads/freshtomato-RT-AC68U-K26ARM-2024.5.182-beta-mdu-buffer-log-AIO-64K.trx

Vlad-Star, 2025-02-02:
v182. Yay, it worked! Thanks man for nailing it down, as well as your patience!
```
root@ASUS4:/tmp/home/root# awk -f redaction.awk dict.txt /var/log/messages | egrep 'mdu|ddns|curl'
Feb  2 17:39:23 ASUS4 user.debug preinit[1]: *** stop_ddns: IN
Feb  2 17:39:23 ASUS4 user.info preinit[1]: ddns is stopped
Feb  2 17:39:23 ASUS4 user.debug preinit[1]: *** start_ddns: IN
Feb  2 17:39:23 ASUS4 user.info preinit[1]: ddns is started
Feb  2 17:39:23 ASUS4 user.debug ddns-update[1876]: *** ddns_update_main: args:
Feb  2 17:39:23 ASUS4 user.debug ddns-update[1876]: *** update: IN num=[0] dirty=[0] force=[0]
Feb  2 17:39:23 ASUS4 user.debug ddns-update[1876]: *** update: cru d ddns0
Feb  2 17:39:23 ASUS4 user.debug ddns-update[1876]: *** update: cru d ddnsf0
Feb  2 17:39:23 ASUS4 user.debug ddns-update[1876]: *** update: !f_exist(/var/lib/mdu/ddnsx0.msg) - creating ...
Feb  2 17:39:23 ASUS4 user.debug ddns-update[1876]: *** mdu <<<<<<<
Feb  2 17:39:23 ASUS4 daemon.debug mdu[1885]: *** main: IN v2.1 [libcurl]
Feb  2 17:39:23 ASUS4 daemon.debug mdu[1885]: *** main: checking for no WAN mode - true, using custom interface: br1
Feb  2 17:39:23 ASUS4 daemon.debug mdu[1885]: *** check_cookie: IN
Feb  2 17:39:23 ASUS4 daemon.debug mdu[1885]: *** read_tmaddr: IN cachename: /var/lib/mdu/ddnsx0.cache
Feb  2 17:39:23 ASUS4 daemon.debug mdu[1885]: *** read_tmaddr: OUT return 0
Feb  2 17:39:23 ASUS4 daemon.debug mdu[1885]: *** check_cookie: no cookie
Feb  2 17:39:23 ASUS4 daemon.debug mdu[1885]: *** main: proceeding DDNS server update [service: cloudflare ] ...
Feb  2 17:39:23 ASUS4 daemon.debug mdu[1885]: *** update_cloudflare: IN
Feb  2 17:39:23 ASUS4 daemon.debug mdu[1885]: *** _http_req: IN host=[api.cloudflare.com] query=[/client/v4/zones/<redacted_zone>/dns_records?type=A&name=<redacted_name>.<redacted_domain>&order=name&direction=asc] ssl=[1] header=[User-Agent: Mozilla/5.0 (X11; Linux x86_64; rv:10.0) Gecko/20100101 Firefox/109.0^M Authorization: Bearer <redacted_bearer>^M Content-Type: application/json^M Cache-Control: no-cache] auth=[0] data=[(null)] req=[GET] ifname=[br1]
Feb  2 17:39:23 ASUS4 daemon.debug mdu[1885]: *** _http_req: resolving IP of server api.cloudflare.com ...
Feb  2 17:39:23 ASUS4 daemon.debug mdu[1885]: *** curl_resolve_ip: IN
Feb  2 17:39:23 ASUS4 daemon.debug mdu[1885]: *** curl_setup: IN
Feb  2 17:39:23 ASUS4 daemon.debug mdu[1885]: *** curl_setup: OUT
Feb  2 17:39:23 ASUS4 daemon.debug mdu[1885]: *** curl_resolve_ip: after curl_setup()
Feb  2 17:39:23 ASUS4 daemon.debug mdu[1885]: *** curl_resolve_ip: after CURLOPT_URL
Feb  2 17:39:23 ASUS4 daemon.debug mdu[1885]: *** curl_resolve_ip: after CURLOPT_HTTPHEADER
Feb  2 17:39:23 ASUS4 daemon.debug mdu[1885]: *** curl_resolve_ip: after CURLOPT_INTERFACE
Feb  2 17:39:23 ASUS4 daemon.debug mdu[1885]: *** curl_resolve_ip: before curl_easy_perform trys=[4]
Feb  2 17:39:24 ASUS4 daemon.debug mdu[1885]: *** curl_resolve_ip: after curl_easy_perform r=[0] trys=[4]
Feb  2 17:39:24 ASUS4 daemon.debug mdu[1885]: *** curl_cleanup: IN
Feb  2 17:39:24 ASUS4 daemon.debug mdu[1885]: *** curl_cleanup: after curl_dfile
Feb  2 17:39:24 ASUS4 daemon.debug mdu[1885]: *** curl_cleanup: after curl_easy_cleanup()
Feb  2 17:39:24 ASUS4 daemon.debug mdu[1885]: *** curl_cleanup: OUT
Feb  2 17:39:24 ASUS4 daemon.debug mdu[1885]: *** curl_resolve_ip: OUT IP=[104.19.193.29]
Feb  2 17:39:24 ASUS4 daemon.debug mdu[1885]: *** _http_req: IP=[104.19.193.29] of host=[api.cloudflare.com]
Feb  2 17:39:24 ASUS4 daemon.debug mdu[1885]: *** _http_req: after curl_resolve_ip
Feb  2 17:39:24 ASUS4 daemon.debug mdu[1885]: *** curl_setup: IN
Feb  2 17:39:24 ASUS4 daemon.debug mdu[1885]: *** curl_setup: OUT
Feb  2 17:39:25 ASUS4 daemon.debug mdu[1885]: *** _http_req: connected, response=[200]
Feb  2 17:39:25 ASUS4 daemon.debug mdu[1885]: *** curl_cleanup: IN
Feb  2 17:39:25 ASUS4 daemon.debug mdu[1885]: *** curl_cleanup: after curl_dfile
Feb  2 17:39:25 ASUS4 daemon.debug mdu[1885]: *** curl_cleanup: after curl_easy_cleanup()
Feb  2 17:39:25 ASUS4 daemon.debug mdu[1885]: *** curl_cleanup: OUT
Feb  2 17:39:25 ASUS4 daemon.debug mdu[1885]: *** _http_req: OUT code=[200]
Feb  2 17:39:25 ASUS4 daemon.debug mdu[1885]: *** update_cloudflare: after http_req() query=[/client/v4/zones/<redacted_zone>/dns_records?type=A&name=<redacted_name>.<redacted_domain>&order=name&direction=asc] header=[User-Agent: Mozilla/5.0 (X11; Linux x86_64; rv:10.0) Gecko/20100101 Firefox/109.0^M Authorization: Bearer <redacted_bearer>^M Content-Type: application/json^M Cache-Control: no-cache] return=[200]
Feb  2 17:39:25 ASUS4 daemon.debug mdu[1885]: *** update_cloudflare: before cloudflare_errorcheck(1)
Feb  2 17:39:25 ASUS4 daemon.debug mdu[1885]: *** cloudflare_errorcheck: IN code=[200]
Feb  2 17:39:25 ASUS4 daemon.debug mdu[1885]: *** cloudflare_errorcheck: after body[n] strlen(body)=[491] body=[{"result":[{"id":"<redacted_id>","name":"<redacted_name>.<redacted_domain>","type":"A","content":"<redacted_ip>","proxiable":true,"proxied":false,"ttl":1,"settings":{},"meta":{"auto_added":false,"managed_by_apps":false,"managed_by_argo_tunnel":false},"comment":null,"tags":[],"created_on":"2024-01-26T11:58:17.151364Z","modified_on":"2025-01-30T17:23:49.058729Z"}],"success":true,"errors":[],"messages":[],"resu
Feb  2 17:39:25 ASUS4 daemon.debug mdu[1885]: *** cloudflare_errorcheck: code == 200
Feb  2 17:39:25 ASUS4 daemon.debug mdu[1885]: *** cloudflare_errorcheck: success:true
[... cut ...]
Feb  2 17:39:26 ASUS4 user.debug ddns-update[1876]: *** update: cru d ddns2
Feb  2 17:39:26 ASUS4 user.debug ddns-update[1876]: *** update: cru d ddnsf2
Feb  2 17:39:26 ASUS4 user.debug ddns-update[1876]: *** update: msg=''
Feb  2 17:39:26 ASUS4 user.debug ddns-update[1876]: *** update: IN num=[3] dirty=[0] force=[0]
Feb  2 17:39:26 ASUS4 user.debug ddns-update[1876]: *** update: cru d ddns3
Feb  2 17:39:26 ASUS4 user.debug ddns-update[1876]: *** update: cru d ddnsf3
Feb  2 17:39:26 ASUS4 user.debug ddns-update[1876]: *** update: msg=''
root@ASUS4:/tmp/home/root#
```

Vlad-Star, 2025-02-03:
Just confirmed that CF IP update is working from the command prompt with the following command
```
curl https://api.cloudflare.com/client/v4/zones/$ZONE_ID/dns_records/$DNS_RECORD_ID \
    -s \
    -X PUT \
    -H "Authorization: Bearer $CLOUDFLARE_API_KEY" \
    -H 'Content-Type: application/json' \
    -d '{
      "content": "'$IP'",
      "name": "'$FQDN'",
      "proxied": false,
      "type": "A"
    }'
```

pedro0311, 2025-02-03:
OK, so please change this IP address again and test: https://bitbucket.org/pedro311/freshtomato-arm/downloads/freshtomato-RT-AC68U-K26ARM-2024.5.183-beta-mdu-buffer-log-AIO-64K.trx, because I see missing logs in v182 probably due to busybox limits (I put usleep in a few places).

Vlad-Star, 2025-02-03:
v183
```
root@ASUS4:/tmp/home/root# awk -f redaction.awk dict.txt /var/log/messages | egrep 'mdu|ddns|curl'
Feb  3 10:21:25 ASUS4 user.debug preinit[1]: *** stop_ddns: IN
Feb  3 10:21:25 ASUS4 user.info preinit[1]: ddns is stopped
Feb  3 10:21:25 ASUS4 user.debug preinit[1]: *** start_ddns: IN
Feb  3 10:21:25 ASUS4 user.info preinit[1]: ddns is started
Feb  3 10:21:25 ASUS4 user.debug ddns-update[3074]: *** ddns_update_main: args:
Feb  3 10:21:25 ASUS4 user.debug ddns-update[3074]: *** update: IN num=[0] dirty=[0] force=[0]
Feb  3 10:21:25 ASUS4 user.debug ddns-update[3074]: *** update: cru d ddns0
Feb  3 10:21:25 ASUS4 user.debug ddns-update[3074]: *** update: cru d ddnsf0
Feb  3 10:21:25 ASUS4 user.debug ddns-update[3074]: *** update: !f_exist(/var/lib/mdu/ddnsx0.msg) - creating ...
Feb  3 10:21:25 ASUS4 user.debug ddns-update[3074]: *** mdu <<<<<<<
Feb  3 10:21:25 ASUS4 daemon.debug mdu[3083]: *** main: IN v2.1 [libcurl]
Feb  3 10:21:25 ASUS4 daemon.debug mdu[3083]: *** main: checking for no WAN mode - true, using custom interface: br1
Feb  3 10:21:25 ASUS4 daemon.debug mdu[3083]: *** check_cookie: IN
Feb  3 10:21:25 ASUS4 daemon.debug mdu[3083]: *** read_tmaddr: IN cachename: /var/lib/mdu/ddnsx0.cache
Feb  3 10:21:25 ASUS4 daemon.debug mdu[3083]: *** check_cookie: no cookie
Feb  3 10:21:25 ASUS4 daemon.debug mdu[3083]: *** main: proceeding DDNS server update [service: cloudflare ] ...
Feb  3 10:21:25 ASUS4 daemon.debug mdu[3083]: *** update_cloudflare: IN
Feb  3 10:21:25 ASUS4 daemon.debug mdu[3083]: *** _http_req: IN host=[api.cloudflare.com] query=[/client/v4/zones/<redacted_zone>/dns_records?type=A&name=<redacted_name>.<redacted_domain>&order=name&direction=asc] ssl=[1] header=[User-Agent: Mozilla/5.0 (X11; Linux x86_64; rv:10.0) Gecko/20100101 Firefox/109.0^M Authorization: Bearer <redacted_bearer>^M Content-Type: application/json^M Cache-Control: no-cache] auth=[0] data=[(null)] req=[GET] ifname=[br1]
Feb  3 10:21:26 ASUS4 daemon.debug mdu[3083]: *** _http_req: resolving IP of server api.cloudflare.com ...
Feb  3 10:21:26 ASUS4 daemon.debug mdu[3083]: *** curl_resolve_ip: IN
Feb  3 10:21:26 ASUS4 daemon.debug mdu[3083]: *** curl_setup: IN
Feb  3 10:21:26 ASUS4 daemon.debug mdu[3083]: *** curl_setup: OUT
Feb  3 10:21:26 ASUS4 daemon.debug mdu[3083]: *** curl_cleanup: IN
Feb  3 10:21:26 ASUS4 daemon.debug mdu[3083]: *** curl_cleanup: OUT
Feb  3 10:21:26 ASUS4 daemon.debug mdu[3083]: *** curl_resolve_ip: OUT IP=[104.19.193.29]
Feb  3 10:21:26 ASUS4 daemon.debug mdu[3083]: *** _http_req: IP=[104.19.193.29] of host=[api.cloudflare.com]
Feb  3 10:21:26 ASUS4 daemon.debug mdu[3083]: *** _http_req: after curl_resolve_ip
Feb  3 10:21:26 ASUS4 daemon.debug mdu[3083]: *** _http_req: after curl_wbuf
Feb  3 10:21:26 ASUS4 daemon.debug mdu[3083]: *** curl_setup: IN
Feb  3 10:21:26 ASUS4 daemon.debug mdu[3083]: *** curl_setup: OUT
Feb  3 10:21:26 ASUS4 daemon.debug mdu[3083]: *** _http_req: after curl_setup()
Feb  3 10:21:27 ASUS4 daemon.debug mdu[3083]: *** _http_req: after CURLOPT_URL
Feb  3 10:21:27 ASUS4 daemon.debug mdu[3083]: *** _http_req: after CURLOPT_HTTPHEADER
Feb  3 10:21:27 ASUS4 daemon.debug mdu[3083]: *** _http_req: after CURLOPT_INTERFACE
Feb  3 10:21:27 ASUS4 daemon.debug mdu[3083]: *** _http_req: after CURLOPT_HTTPAUTH
Feb  3 10:21:27 ASUS4 daemon.debug mdu[3083]: *** _http_req: after CURLOPT_WRITEDATA
Feb  3 10:21:27 ASUS4 daemon.debug mdu[3083]: *** _http_req: after CURLOPT_UPLOAD
Feb  3 10:21:27 ASUS4 daemon.debug mdu[3083]: *** _http_req: after route_adddel()
Feb  3 10:21:27 ASUS4 daemon.debug mdu[3083]: *** _http_req: before curl_easy_perform trys=[4]
Feb  3 10:21:28 ASUS4 daemon.debug mdu[3083]: *** _http_req: after curl_easy_perform r=[0] trys=[4]
Feb  3 10:21:28 ASUS4 daemon.debug mdu[3083]: *** _http_req: before curl_slist_free_all()
Feb  3 10:21:28 ASUS4 daemon.debug mdu[3083]: *** _http_req: connected, response=[200]
Feb  3 10:21:28 ASUS4 daemon.debug mdu[3083]: *** curl_cleanup: IN
Feb  3 10:21:28 ASUS4 daemon.debug mdu[3083]: *** curl_cleanup: OUT
Feb  3 10:21:28 ASUS4 daemon.debug mdu[3083]: *** update_cloudflare: after http_req() query=[/client/v4/zones/<redacted_zone>/dns_records?type=A&name=<redacted_name>.<redacted_domain>&order=name&direction=asc] header=[User-Agent: Mozilla/5.0 (X11; Linux x86_64; rv:10.0) Gecko/20100101 Firefox/109.0^M Authorization: Bearer <redacted_bearer>^M Content-Type: application/json^M Cache-Control: no-cache] return=[200]
Feb  3 10:21:29 ASUS4 daemon.debug mdu[3083]: *** update_cloudflare: before cloudflare_errorcheck(1)
Feb  3 10:21:29 ASUS4 daemon.debug mdu[3083]: *** cloudflare_errorcheck: IN code=[200]
Feb  3 10:21:29 ASUS4 daemon.debug mdu[3083]: *** cloudflare_errorcheck: after body[n] strlen(body)=[491] body=[{"result":[{"id":"<redacted_id>","name":"<redacted_name>.<redacted_domain>","type":"A","content":"<redacted_wrong_ip>","proxiable":true,"proxied":false,"ttl":1,"settings":{},"meta":{"auto_added":false,"managed_by_apps":false,"managed_by_argo_tunnel":false},"comment":null,"tags":[],"created_on":"2024-01-26T11:58:17.151364Z","modified_on":"2025-02-03T15:20:22.174748Z"}],"success":true,"errors":[],"messages":[],"resu
Feb  3 10:21:29 ASUS4 daemon.debug mdu[3083]: *** cloudflare_errorcheck: code == 200
Feb  3 10:21:29 ASUS4 daemon.debug mdu[3083]: *** cloudflare_errorcheck: success:true
Feb  3 10:21:29 ASUS4 daemon.debug mdu[3083]: *** cloudflare_errorcheck: total_count:!0
[... cut ...]
Feb  3 10:21:44 ASUS4 user.debug ddns-update[3074]: *** update: msg=''
Feb  3 10:21:44 ASUS4 user.debug ddns-update[3074]: *** update: IN num=[3] dirty=[0] force=[0]
Feb  3 10:21:44 ASUS4 user.debug ddns-update[3074]: *** update: cru d ddns3
Feb  3 10:21:44 ASUS4 user.debug ddns-update[3074]: *** update: cru d ddnsf3
Feb  3 10:21:44 ASUS4 user.debug ddns-update[3074]: *** update: msg=''
root@ASUS4:/tmp/home/root# awk -f redaction.awk dict.txt /var/lib/mdu/ddnsx0.extip
546,<redacted_ip>
root@ASUS4:/tmp/home/root#
```

Vlad-Star, 2025-02-03:
Side note - looks like this IP service is no longer working (not sure if temporary or permanent)
```
Feb  3 06:16:22 ASUS4 daemon.debug mdu[21949]: *** curl_resolve_ip: error - (libcurl error (7) - Failed to connect to ip.tyk.nu port 80 after 122 ms: Error.)
Feb  3 06:16:22 ASUS4 daemon.debug mdu[21949]: *** get_address: libcurl error (7) - Failed to connect to ip.tyk.nu port 80 after 122 ms: Error. (ip.tyk.nu) - trying another one ...

```

```
root@ASUS4:/tmp/home/root# wget ip.tyk.nu
Connecting to ip.tyk.nu (95.216.101.209:80)
wget: can't connect to remote host (95.216.101.209): Connection refused
root@ASUS4:/tmp/home/root#
```

pedro0311, 2025-02-03:
```
curl https://api.cloudflare.com/client/v4/zones/$ZONE_ID/dns_records/$DNS_RECORD_ID \
    -s \
    -X PUT \
    -H "Authorization: Bearer $CLOUDFLARE_API_KEY" \
    -H 'Content-Type: application/json' \
    -d '{
      "content": "'$IP'",
      "name": "'$FQDN'",
      "proxied": false,
      "type": "A"
    }'
```

In 'content' and 'name' are double " and ' needed or just " will suffice? Can you check it?

Vlad-Star, 2025-02-03:
double quotes are only needed for shell to substitute defined variables ($IP and $FQDN), otherwise it should work with single quotes, let me quickly test

Vlad-Star, 2025-02-03:
Cloudflare API (DNS update): https://developers.cloudflare.com/api/resources/dns/subresources/records/methods/update/

pedro0311, 2025-02-03:
As I can see, there is an auth with API Email + API Key but your working curl example is with "Authorization: Bearer API_KEY"?

Vlad-Star, 2025-02-03:
it's working both ways, but let me correct the terms, just to be clear

API TOKEN here (which needs to be generated through the UI):
`-H "Authorization: Bearer $CLOUDFLARE_API_TOKEN"`

or

E-Mail + Global API key here ():
```
    -H "X-Auth-Email: $CLOUDFLARE_EMAIL"
    -H "X-Auth-Key: $CLOUDFLARE_GLOBAL_API_KEY" 
```

Here is what they suggest: `When possible, use API tokens instead of Global API keys.`

pedro0311, 2025-02-03:
OK, now I understand. Let's just make it work, and then it can be converted to the preferred version.
https://bitbucket.org/pedro311/freshtomato-arm/downloads/freshtomato-RT-AC68U-K26ARM-2024.5.184-beta-mdu-buffer-log-AIO-64K.trx

Vlad-Star, 2025-02-03:
Basically, this is the page for both (if you still have the CF account):
https://dash.cloudflare.com/profile/api-tokens

Vlad-Star, 2025-02-03:
v184 works (and this IS the preferred method)
```
root@ASUS4:/tmp/home/root# awk -f redaction.awk dict.txt /var/log/messages | egrep 'mdu|ddns|curl'
Feb  3 12:19:45 ASUS4 user.debug preinit[1]: *** stop_ddns: IN
Feb  3 12:19:45 ASUS4 user.info preinit[1]: ddns is stopped
Feb  3 12:19:45 ASUS4 user.debug preinit[1]: *** start_ddns: IN
Feb  3 12:19:45 ASUS4 user.info preinit[1]: ddns is started
Feb  3 12:19:45 ASUS4 user.debug ddns-update[2214]: *** ddns_update_main: args:
Feb  3 12:19:45 ASUS4 user.debug ddns-update[2214]: *** update: IN num=[0] dirty=[0] force=[0]
Feb  3 12:19:45 ASUS4 user.debug ddns-update[2214]: *** update: cru d ddns0
Feb  3 12:19:45 ASUS4 user.debug ddns-update[2214]: *** update: cru d ddnsf0
Feb  3 12:19:45 ASUS4 user.debug ddns-update[2214]: *** update: !f_exist(/var/lib/mdu/ddnsx0.msg) - creating ...
Feb  3 12:19:45 ASUS4 user.debug ddns-update[2214]: *** mdu <<<<<<<
Feb  3 12:19:45 ASUS4 daemon.debug mdu[2223]: *** main: IN v2.1 [libcurl]
Feb  3 12:19:45 ASUS4 daemon.debug mdu[2223]: *** main: checking for no WAN mode - true, using custom interface: br1
Feb  3 12:19:45 ASUS4 daemon.debug mdu[2223]: *** check_cookie: IN
Feb  3 12:19:45 ASUS4 daemon.debug mdu[2223]: *** read_tmaddr: IN cachename: /var/lib/mdu/ddnsx0.cache
Feb  3 12:19:45 ASUS4 daemon.debug mdu[2223]: *** check_cookie: no cookie
Feb  3 12:19:45 ASUS4 daemon.debug mdu[2223]: *** main: proceeding DDNS server update [service: cloudflare ] ...
Feb  3 12:19:45 ASUS4 daemon.debug mdu[2223]: *** update_cloudflare: IN
Feb  3 12:19:46 ASUS4 daemon.debug mdu[2223]: *** _http_req: IN host=[api.cloudflare.com] query=[/client/v4/zones/<redacted_zone>/dns_records?type=A&name=<redacted_name>.<redacted_domain>&order=name&direction=asc] ssl=[1] header=[User-Agent: Mozilla/5.0 (X11; Linux x86_64; rv:10.0) Gecko/20100101 Firefox/109.0^M Authorization: Bearer <redacted_bearer>^M Content-Type: application/json^M Cache-Control: no-cache] auth=[0] data=[(null)] req=[GET] ifname=[br1]
Feb  3 12:19:46 ASUS4 daemon.debug mdu[2223]: *** _http_req: resolving IP of server api.cloudflare.com ...
Feb  3 12:19:46 ASUS4 daemon.debug mdu[2223]: *** curl_resolve_ip: IN
Feb  3 12:19:46 ASUS4 daemon.debug mdu[2223]: *** curl_setup: IN
Feb  3 12:19:46 ASUS4 daemon.debug mdu[2223]: *** curl_setup: OUT
Feb  3 12:19:47 ASUS4 daemon.debug mdu[2223]: *** curl_cleanup: IN
Feb  3 12:19:47 ASUS4 daemon.debug mdu[2223]: *** curl_cleanup: OUT
Feb  3 12:19:47 ASUS4 daemon.debug mdu[2223]: *** curl_resolve_ip: OUT IP=[104.19.192.175]
Feb  3 12:19:47 ASUS4 daemon.debug mdu[2223]: *** _http_req: IP=[104.19.192.175] of host=[api.cloudflare.com]
Feb  3 12:19:47 ASUS4 daemon.debug mdu[2223]: *** _http_req: after curl_resolve_ip
Feb  3 12:19:47 ASUS4 daemon.debug mdu[2223]: *** _http_req: after curl_wbuf
Feb  3 12:19:47 ASUS4 daemon.debug mdu[2223]: *** curl_setup: IN
Feb  3 12:19:47 ASUS4 daemon.debug mdu[2223]: *** curl_setup: OUT
Feb  3 12:19:47 ASUS4 daemon.debug mdu[2223]: *** _http_req: after curl_setup()
Feb  3 12:19:47 ASUS4 daemon.debug mdu[2223]: *** _http_req: after CURLOPT_URL
Feb  3 12:19:47 ASUS4 daemon.debug mdu[2223]: *** _http_req: after CURLOPT_HTTPHEADER
Feb  3 12:19:47 ASUS4 daemon.debug mdu[2223]: *** _http_req: after CURLOPT_INTERFACE
Feb  3 12:19:47 ASUS4 daemon.debug mdu[2223]: *** _http_req: after CURLOPT_HTTPAUTH
Feb  3 12:19:47 ASUS4 daemon.debug mdu[2223]: *** _http_req: after CURLOPT_WRITEDATA
Feb  3 12:19:47 ASUS4 daemon.debug mdu[2223]: *** _http_req: after CURLOPT_UPLOAD
Feb  3 12:19:47 ASUS4 daemon.debug mdu[2223]: *** _http_req: after route_adddel()
Feb  3 12:19:47 ASUS4 daemon.debug mdu[2223]: *** _http_req: before curl_easy_perform trys=[4]
Feb  3 12:19:48 ASUS4 daemon.debug mdu[2223]: *** _http_req: after curl_easy_perform r=[0] trys=[4]
Feb  3 12:19:49 ASUS4 daemon.debug mdu[2223]: *** _http_req: before curl_slist_free_all()
Feb  3 12:19:49 ASUS4 daemon.debug mdu[2223]: *** _http_req: connected, response=[200]
Feb  3 12:19:49 ASUS4 daemon.debug mdu[2223]: *** curl_cleanup: IN
Feb  3 12:19:49 ASUS4 daemon.debug mdu[2223]: *** curl_cleanup: OUT
Feb  3 12:19:49 ASUS4 daemon.debug mdu[2223]: *** update_cloudflare: after http_req() query=[/client/v4/zones/<redacted_zone>/dns_records?type=A&name=<redacted_name>.<redacted_domain>&order=name&direction=asc] header=[User-Agent: Mozilla/5.0 (X11; Linux x86_64; rv:10.0) Gecko/20100101 Firefox/109.0^M Authorization: Bearer <redacted_bearer>^M Content-Type: application/json^M Cache-Control: no-cache] return=[200]
Feb  3 12:19:49 ASUS4 daemon.debug mdu[2223]: *** update_cloudflare: before cloudflare_errorcheck(1)
Feb  3 12:19:49 ASUS4 daemon.debug mdu[2223]: *** cloudflare_errorcheck: IN code=[200]
Feb  3 12:19:49 ASUS4 daemon.debug mdu[2223]: *** cloudflare_errorcheck: after body[n] strlen(body)=[491] body=[{"result":[{"id":"<redacted_id>","name":"<redacted_name>.<redacted_domain>","type":"A","content":"<redacted_wrong_ip>","proxiable":true,"proxied":false,"ttl":1,"settings":{},"meta":{"auto_added":false,"managed_by_apps":false,"managed_by_argo_tunnel":false},"comment":null,"tags":[],"created_on":"2024-01-26T11:58:17.151364Z","modified_on":"2025-02-03T17:19:38.405524Z"}],"success":true,"errors":[],"messages":[],"resu
Feb  3 12:19:49 ASUS4 daemon.debug mdu[2223]: *** cloudflare_errorcheck: code == 200
Feb  3 12:19:49 ASUS4 daemon.debug mdu[2223]: *** cloudflare_errorcheck: success:true
Feb  3 12:19:49 ASUS4 daemon.debug mdu[2223]: *** cloudflare_errorcheck: total_count:!0
[... cut ...]
Feb  3 12:19:55 ASUS4 user.debug ddns-update[2214]: *** update: cru d ddns2
Feb  3 12:19:55 ASUS4 user.debug ddns-update[2214]: *** update: cru d ddnsf2
Feb  3 12:19:55 ASUS4 user.debug ddns-update[2214]: *** update: msg=''
Feb  3 12:19:55 ASUS4 user.debug ddns-update[2214]: *** update: IN num=[3] dirty=[0] force=[0]
Feb  3 12:19:55 ASUS4 user.debug ddns-update[2214]: *** update: cru d ddns3
Feb  3 12:19:55 ASUS4 user.debug ddns-update[2214]: *** update: cru d ddnsf3
Feb  3 12:19:55 ASUS4 user.debug ddns-update[2214]: *** update: msg=''
root@ASUS4:/tmp/home/root#
```

[14 more replies]

closed 2025-02-11 by pedro0311