FreshTomato-Project/freshtomato-arm #56: Option to Log "Successful" WebGUI Logins to the System Log

LakePath, 2025-01-31:
In the System Log... every-now-and-then... I'll see an "**_unsuccessful_**" login, only to wonder if a "**_successful_**" login had been achieved shortly thereafter without my knowledge.

Would it be possible to add an Option to enable "Successful" WebGUI Logins in Administration > Logging ?

Suggested by @wellloaded [here](https://www.linksysinfo.org/index.php?threads/about-logins-logs-and-protection.78742/#post-356351:~:text=2) and discussed [further here](https://www.linksysinfo.org/index.php?threads/about-logins-logs-and-protection.78742/post-355494).

HommeOursPorc, 2025-08-13:
I looked at the source and tried to find where the logs are coming from... It look like it is coming from the httpd binary, but it is not in Busybox/httpd source???

```
root@AC68U-Relais:/usr/sbin# strings httpd | grep -n  bad
254:bad password attempt (GUI) from: %s
```

Where is this code??

HommeOursPorc, 2025-08-15:
@pedro0311 would adding this: 
```
logmsg(LOG_INFO, "successful login (GUI) from: %s", client_addr);
```
At line ~566 would work?
https://github.com/FreshTomato-Project/freshtomato-arm/blob/f75bceea62077da3c3d70e91cde759d71a344433/release/src-rt-6.x.4708/router/httpd/httpd.c#L566

Like this: 
```
	if (auth == AUTH_BAD) {
		auth_fail(cl, 1);
		return;
	}

	for (handler = &mime_handlers[0]; handler->pattern; handler++) {
		if (match(handler->pattern, file)) {
			if ((handler->auth) && (auth != AUTH_OK)) {
				auth_fail(cl, 0);
				return;
			}

			// Log successful authentication for protected resources
			if ((handler->auth) && (auth == AUTH_OK)) {
				logmsg(LOG_INFO, "successful login (GUI) from: %s", client_addr);
			}

			if (handler->input)
				handler->input(file, cl, boundary);

			eat_garbage();

			if (handler->mime_type != NULL)
				send_header(200, NULL, handler->mime_type, handler->cache);

			if (handler->output)
				handler->output(file);

			return;
		}
	}

	if (auth != AUTH_OK) {
		auth_fail(cl, (auth == AUTH_NONE ? 0 : 1));
		return;
	}

	send_error(404, NULL, NULL);
}
```

Yes.... This is the AI answer... But it might have a good idea for once 🤣🤞

What do you think?

HommeOursPorc, 2025-08-15:
Ok. I'll make myself a build environment and do some test. Eventually, I hope, I will get it to work.

pedro0311, 2025-08-15:
No, use AI. It helped me (after 2 weeks of problems) with this: https://github.com/FreshTomato-Project/freshtomato-arm/commit/0f9e33516d001360d18c0a010a432892670c7b02
It's all about asking the question correctly (everything has to be configured correctly one by one) and correcting his answer (he's wrong as hell most of the time).

HommeOursPorc, 2025-08-15:
> No, use AI. It helped me (after 2 weeks of problems) with this: [0f9e335](https://github.com/FreshTomato-Project/freshtomato-arm/commit/0f9e33516d001360d18c0a010a432892670c7b02) It's all about asking the question correctly (everything has to be configured correctly one by one) and correcting his answer (he's wrong as hell most of the time).

Ok I used Gemini instead of GitHub Copilot

Gemini says the correct place is at line ~298
https://github.com/FreshTomato-Project/freshtomato-arm/blob/f75bceea62077da3c3d70e91cde759d71a344433/release/src-rt-6.x.4708/router/httpd/httpd.c#L298

This: 
```
if (strcmp(pass, p) == 0)
    return AUTH_OK;
}
```

Need to be changed to this: 
```
if (strcmp(pass, p) == 0) {
    client_addr = get_client_addr();
    logmsg(LOG_NOTICE, "successful login for user '%s' from %s", authinfo, client_addr);
    return AUTH_OK;
}
```
 Gemini says:
// This is the correct place to add the successful login message.
// It is only triggered when both username and password are correct.

By placing the logmsg inside the if (strcmp(pass, p) == 0) block, you can be sure it will only run when the authentication is successful. This is the correct, efficient way to add the log message **without flooding your system logs**.

I'll try to set a test environment today 🙂

wellloaded, 2025-08-15:
I agree with @pedro0311 the so called "bad answers" are slowly disappearing from AI.
So in this context it's important to remember there's no such a thing as a bad answer (99% of the times at least) but rather a not so good/descriptive question.

Human beings get annoyed when questions are too long, AI instead requires this to work properly. Just add all your conditions to the question it will eventually work.

HTH

HommeOursPorc, 2025-08-16:
> I'm not sure about 'client_addr'... At one place Gemini says
> `client_addr = get_client_addr();`
> and at an other place it says
> `get_client_addr();`

ok... the first one doesn't work

````
   httpd 44/104
make[5]: Entering directory '/root/freshtomato-arm/release/src-rt-6.x.4708/router/httpd'
 [httpd] CC httpd.o
httpd.c: In function ‘auth_check’:
httpd.c:298:19: error: assignment to expression with array type
       client_addr = get_client_addr();
                   ^
make[5]: *** [Makefile:63: httpd.o] Error 1
make[5]: Leaving directory '/root/freshtomato-arm/release/src-rt-6.x.4708/router/httpd'
make[4]: *** [Makefile:4463: httpd] Error 2
make[4]: Leaving directory '/root/freshtomato-arm/release/src-rt-6.x.4708/router'
make[3]: *** [build.mak:7: build_all] Error 2
make[3]: Leaving directory '/root/freshtomato-arm/release/src-rt-6.x.4708'
make[2]: *** [Makefile:1162: bin] Error 2
make[2]: Leaving directory '/root/freshtomato-arm/release/src-rt-6.x.4708'
make[1]: *** [Makefile:1168: e] Error 2
make[1]: Leaving directory '/root/freshtomato-arm/release/src-rt-6.x.4708'
make: *** [targets.mak:2: ac68e] Error 2
root@debian-vbox:~/freshtomato-arm/release/src-rt-6.x.4708#
````

HommeOursPorc, 2025-08-16:
LOL

it doesn't work. It spam the logs again :(

[image]

HommeOursPorc, 2025-08-16:
## Plan for Tracking Successful Basic Auth Logins (Without Cookies)

### 1. Track Successful Logins Per (Username, IP)
- Maintain a server-side table or file that records (username, IP, last_successful_login_timestamp).

### 2. On Each Authenticated Request
- When credentials are valid:
  - Check if (username, IP) exists and if the timestamp is recent (e.g., within the last 30 minutes).
    - If not present or expired:  
      - Log the successful login.
      - Store/update the (username, IP, timestamp) entry.
    - If present and not expired:  
      - Do not log (user is already “logged in”).

### 3. Expiry Mechanism (Timeout)
- On every check, remove/ignore entries older than the configured timeout (e.g., 30 minutes).
- This ensures if a user closes the browser and comes back later, a new login will be logged.

### 4. On Logout (if implemented)
- If your web GUI provides a logout button:
  - When logout is clicked, remove the (username, IP) entry from your tracking table/file.

### 5. Where to Store the Table?
- For minimal persistence, use a temporary file (e.g., /tmp/httpd_login_success).
- If you prefer, use in-memory static data (but be aware this won’t persist across process restarts).

---

## Example Flow

1. User logs in successfully from IP 1.2.3.4 as “admin”.
2. You check if (“admin”, 1.2.3.4) is in your temp file and not expired.
   - If not, log it and store (“admin”, 1.2.3.4, now).
   - If yes and not expired, do nothing.
3. On each request, repeat the check.
   - If >30 minutes have passed, treat as new login.

---

## Implementation Steps

1. **Add a function** to check and update the temp file (or memory) for (username, IP) and timestamp.
2. **Modify httpd.c** where successful Basic Auth is detected:
   - Call your function before logging.
   - Only log if the function says this is a new login.
3. **Periodically clean up** expired entries (this can be done on each check for simplicity).

---

## Pseudo C Code

```c
#define LOGIN_TIMEOUT 1800 // 30 minutes

struct LoginEntry {
    char username[64];
    char ip[32];
    time_t last_login;
};

bool should_log_success(const char* username, const char* ip) {
    // 1. Read temp file or in-memory table
    // 2. For each entry, if entry is expired, remove it
    // 3. If (username, ip) is found and not expired, return false
    // 4. Otherwise, add/update entry and return true
}

void on_successful_login(const char* username, const char* ip) {
    if (should_log_success(username, ip)) {
        syslog(LOG_NOTICE, "SUCCESSFUL LOGIN FROM %s FOR %s", ip, username);
    }
}
```

---

## Advantages

- No client-side or protocol changes.
- No extra user interaction.
- Minimal code changes, and robust for your use case.

---

I think what is complicated here is the timeout. The successful login message will be logged every 30 minutes... Not that bad, but that is I close the browser and reopen it? The GUI will ask for my username and password, but the "log record file (or nvram variable)" will still be inside the 30 minutes window, so it won't be logged...??

Is there a function that make the prompt to ask for the username/password???

Hmmm 🤔

HommeOursPorc, 2025-09-06:
Excuse my ignorance, I found this: 
https://stackoverflow.com/questions/34609521/custom-html-login-form-in-http-basic-auth

> Since you are using an AJAX call, you could intercept the 401 status code from the server and redirect the user to a custom login form. For example let's suppose that you were using jQuery and trying to access the protected with Basic Authentication API endpoint https://httpbin.org/basic-auth/user/passwd:
>
>
 ```
$.ajax({
    url: 'https://httpbin.org/basic-auth/user/passwd',
    type: 'GET'
}).then(function(result) {
    alert('success');
}).fail(function(xhr) {
   if (xhr.status === 401) {
       // we are not authenticated => we must redirect the user to our custom login form
       window.location.href = '/my-custom-login-form';
   }
});
```
>
> Once you have collected the username and password you will know how to construct the correct authentication header on subsequent requests to your API:
>
``` $.ajax({
    url: 'https://httpbin.org/basic-auth/user/passwd',
    type: 'GET',
    headers: {
        Authorization: 'Basic dXNlcjpwYXNzd2Q='
    }
})...
```

@pedro0311 can something like this be useful to us?

If I understand correctly, we create a login page instead of using the browser pop-up. Then the page will create the basic auth... No idea how that works, sorry 😐 
But I suppose that login in from a page instead of the pop-up could give us the control we need to log the successful logins? 

Honestly I have no idea of what I'm talking about. 
Just trying to bring ideas. 

Hopefully it will somehow help a little bit 🙂

wellloaded, 2025-09-07:
Most of the open source implementations use this approach (login in webpage) and it's more elegant as well as it can be provided in a webpage with css applied.

What I suggest @HommeOursPorc is to squeeze AI (I would personally use Gemini for this) and see if you can create an initial bogus login page, this can then be added to a test build and move forward from there. The next step is to address the auth cookie part of the code, and relevant browser popup login integration, I don't consider this to be extremely difficult, but it will require time and dedication for sure.