FreshTomato-Project/freshtomato-arm #58: compile problem on ubuntu 24.04.1 LTS

blu66blu66, 2025-02-10:
configure.ac:296: error: possibly undefined macro: AC_SYS_YEAR2038_RECOMMENDED
      If this token and others are legitimate, please use m4_pattern_allow.
      See the Autoconf documentation.
make[4]: *** [Makefile:3918: util-linux/stamp-h1] Error 1
make[4]: Leaving directory '/home/danut/Desktop/freshtomato-arm/release/src-rt-6.x.4708/router'
make[3]: *** [build.mak:7: build_all] Error 2
make[3]: Leaving directory '/home/danut/Desktop/freshtomato-arm/release/src-rt-6.x.4708'
make[2]: *** [Makefile:1148: bin] Error 2
make[2]: Leaving directory '/home/danut/Desktop/freshtomato-arm/release/src-rt-6.x.4708'
make[1]: *** [Makefile:1166: z] Error 2
make[1]: Leaving directory '/home/danut/Desktop/freshtomato-arm/release/src-rt-6.x.4708'
make: *** [targets.mak:11: n18z] Error 2

pedro0311, 2025-02-11:
What is your autotools version? i.e.:
`autoconf --version`

closed 2025-02-11 by pedro0311

blu66blu66, 2025-02-11:
> What is your autotools version? i.e.: `autoconf --version`

autoconf (GNU Autoconf) 2.71

commit 35e9cd7a (2025-02-11): Revert "util-linux: update to 2.40.4" (close #58)

This reverts commit f7de9c3a9a3eda1b38f2804830e089c4362dc7bc.

Waiting for newer autotools...