FreshTomato-Project/freshtomato-arm #71: problem with firmware upgrade 2025.1

blu66blu66, 2025-02-26:
Won` upgrade ...maybe is from commit 
https://github.com/FreshTomato-Project/freshtomato-arm/commit/e7d6d3575382a989cbe148a089bd59d5726d4905
...i don`t know....please check

pedro0311, 2025-02-26:
What are you talking about?

blu66blu66, 2025-02-26:
i have compilled untill yesterday ....but now i can't upgrade to latest firmware. is freesing. my rooter is ok . but it won`t upgrade. last commit it was 
[www: admin-upgrade.asp: fix typo in](https://github.com/FreshTomato-Project/freshtomato-arm/commit/e7d6d3575382a989cbe148a089bd59d5726d4905) https://github.com/FreshTomato-Project/freshtomato-arm/commit/99335ef07f3465ec8f7eb3200f2179fcfd2ff107
after that won`t upgrade...

pedro0311, 2025-02-26:
Sorry, but  no one said that you can upgrade on every commit of FT with no problem. You should use official release.
Sometimes, even in linux repo are problems, so you should be ready for this.
Compilation with no approved commits are always risky, so sorry mate...

blu66blu66, 2025-02-27:
...revert to asus firmware with recovery then clear nvram then install 2025.1, afther that restore my configuration
everything seems to be ok !

closed 2025-02-27 by pedro0311