FreshTomato-Project/freshtomato-arm #73: Unable to connect to specifically LAN0 (br0) via Virtual Wireless Interface with some VLAN setups

alejandro-n-rivera, 2025-03-04:
This is reliably reproducible on my Netgear R8000 router using FreshTomato 2025.1 (VPN version).
[freshtomato-R8000-K26ARM7-2025.1-VPN-64K]

I'm not sure that it's router-model-specific (hence I left it out in the issue title).

**Steps to reproduce:**
(Assumes you already have a version of FreshTomato flashed)

1. Clear NVRAM:
   a. Go to Administration > Configuration > Restore Default Configuration 
   b. Select "Erase all data in NVRAM memory (thorough)" > Click "OK"

2. Flash FreshTomato 2025.1 (VPN version)
   a. Go to Administration > Upgrade > Check "Erase all data in NVRAM." checkbox (for good measure)
   b. Choose File > Choose "freshtomato-R8000-K26ARM7-2025.1-VPN-64K.trx" > Upgrade
   c. Wait for this to complete
   
3. Use a device to connect to default SSID "FreshTomato50-1"
   a. This SSID should be for "Wireless eth1 (wl1) / 5 GHz" by default
   b. You can verify this towards the bottom of the Status > Overview page
   c. Verify you've connected successfully from your device
   d. NOTE: I had my router connected to a WAN already before (unsure if necessary)
   e. This is just to verify we're able to connect to eth1 (wl1)

4. Create wl1.1 Virtual Wireless Interface
   a. Go to Advanced > Virtual Wireless > fill out fields:

        Interface: wl1.1
        Enabled: <checked>
        SSID: [your choice]
        Mode: Access Point
        Bridge: LAN0 (br0)
        WFilter: Disabled

   b. Click "Add"
   c. You should be taken to wl1.1 tab
   d. Leave all values as default:

        Enable Interface: <checked>
        AP Isolation: <unchecked>
        Mac Address: 00:00:00:00:00:00  (this gets generated after creation)
        Wireless Mode: Access Point
        SSID: <auto-populated with your choice>
        Broadcast: <checked>
        Wireless Filter: Disable filter on that interface
        Security: Disabled

   e. Click "Overview"
   f. You should be taken back to Overview tab
   g. Click "Save"
   h. Wait for "Settings saved. Some services are being restarted..." message to disappear

5. Verify wl1.1 / 5 GHz interface is "Up (LAN0)"
   a. Go to Status > Overview
   b. Verify towards bottom of the page that "Wireless wl1.1 / 5 GHz" exists
   c. Verify its Interface Status is "Up (LAN0)"

6. Use a device to connect to Virtual Wireless SSID ([your choice] name):
   a. Verify you've connected successfully from your device
   b. This is just to verify we're able to connect to wl1.1
   c. (At this point I rebooted router and re-confirmed successful connection--probably not necessary)

7. Create new LAN bridge (br1)
   a. Go to Basic > Network
   b. Under LAN section, fill out:

        Bridge: 1
        STP: <unchecked>
        IP Address: 192.168.2.1
        Netmask: 255.255.255.0
        DHCP: <checked>
        IP Range: <Leave as default> (192.168.2.2 / 192.168.2.254)
        Lease Time: 1440

   c. Click "Add"
   d. Scroll down and click "Save"
   e. Wait for "Settings saved. Some services are being restarted..." message to disappear
   
8. At this point, your device may disconnect briefly, but it should be able to reconnect to wl1.1 successfully

9. Create new VLAN for br1 
   a. Go to Advanced > VLAN
   b. Uncheck Port 2 for VLAN 1 > Click "OK" (you may choose another port if you wish)
   c. Click "Save" at the bottom of the page
   d. Fill out:

        VLAN: 3
        VID: 3
        Check Port 2 (or one you chose above)
        Ethernet to bridge mapping: LAN1 (br1)

   e. Click "Add"
   f. Click "Save" at the bottom of the page
   g. Wait for your router to reboot

10. Note at this point that you're still able to connect to Virtual Wireless Interface wl1.1 (SSID [your choice])

11. Bridge wl1 (SSID "FreshTomato50-1") with LAN1(br1)
   a. Go back to Advanced > VLAN
   b. Under VLAN Wireless, change "Bridge eth1 (wl1) / 5 GHz" to "LAN1 (br1)"
   c. NOTE this is for default SSID "FreshTomato50-1" (NOT SSID [your choice])
   d. Click "Save" at the bottom of the page
   e. Wait for your router to reboot again

12. Note at this point that you're no longer able to connect to Virtual Wireless Interface wl1.1 (SSID [your choice])

I'm unsure what's causing the issue above, but I'm willing to help in any way I can.
I do have programming experience and *some* networking experience, so I will be trying to self-investigate as best I can.

**Some more notes:**

- After all steps above, I'm still able to connect to default SSID "FreshTomato50-1" (eth1 (wl1) on LAN1 (br1))
- Rebooted again after step 12 for good measure, but issue persists
- Weirdly, if I create another LAN2 (br2) (w/ DHCP 192.168.3.1/24) and associate wl1.1 with that and reboot, then I'm able to connect to it...
- ...but if I try to associate wl1.1 back with LAN0 (br0) and reboot again, same issue arises.
- This rules out the issue just being that wl1 and wl1.1 can't be associated with different bridges/LANs
- I can even associate wl1.1 with LAN1 (br1) and reboot and reconnect. But still not LAN0 (br0).
- Seems to be something specific about the Virutal Wireless Interface and LAN0 (br0)...
- On other wireless interfaces "eth2 (wl0) / 2.4 GHz" and "eth3 (wl2) / 5 GHz", I'm still able to connect to them while they're associated with LAN0 (br0)
- If I associate eth1 (wl1) back to LAN0 (br0) so that BOTH wl1 and wl1.1 (virtual) interfaces are associated with it, THEN I'm able to connect to LAN0 (br0) with wl1.1 (virtual)
- Also, of course, I verified this with multiple devices (2 different laptops and a smartphone). Behavior was the same with all of them.
- Note that there was no wireless security (e.g., WPA2-Personal or otherwise) used in these steps, and DHCP was always in use. I've found that these make no difference (adding security or trying static IPs).

**Workaround I found for now:**

Strangely, I can associate eth1 (wl1) with LAN0 (br0) and wl1.1 with LAN1 (br1), and have no issues connecting to both wl1 and wl1.1 with this setup (just not vice-versa).

pedro0311, 2025-03-04:
Does this issue also occur in earlier FT versions - 2024.3 for example?

alejandro-n-rivera, 2025-03-04:
> Does this issue also occur in earlier FT versions - 2024.3 for example?

I'm not sure yet, but I will try flashing 2024.3 and report back here. Should be able to test in the next 5 hours or so.

alejandro-n-rivera, 2025-03-04:
> Does this issue also occur in earlier FT versions - 2024.3 for example?

@pedro0311 Can confirm that I'm able to reproduce this same behavior following the steps above on FreshTomato 2024.3 (VPN) [freshtomato-R8000-K26ARM7-2024.3-VPN-64K]

I've followed the steps exactly as outlined (with the exception of using the 2024.3 .trx to flash).

I am happy to test even older firmware versions if you'd like/if it would be helpful.

Also, thank you for looking at this!

pedro0311, 2025-03-04:
> I am happy to test even older firmware versions if you'd like/if it would be helpful.

I'm curious if this issue also occurs in the old @Shibby v140 build: https://tomato.groov.pl/download/K26ARM7/140-MultiWAN/tomato-R8000-ARM--140-VPN-64K.zip
Can you check?
But remember to clean NVRAM.

alejandro-n-rivera, 2025-03-04:
Ok, checked with this shibby build and can confirm that I CANNOT reproduce with this on that firmware. 

Followed the same steps 1-12 as above, except using tomato-R8000-ARM--140-VPN-64K.trx, and did NOT encounter the same issue.

> But remember to clean NVRAM.

Did this before flash, during flash (checkbox when doing the "Upgrade" procedure), and after flash for good measure (just in case).

I noted some differences here just for my own notes and also as sort of proof that I did follow those same steps with the older shibby version:

Step 3. SSID "Tomato50-1" instead for wl1 by default
Step 4a. No "WFilter" field when creating Virtual Wireless Interface wl1.1
Step 4d. No "AP Isolation" checkbox, No "Wireless Filter"
Step 5c. Interface Status is "Up (LAN)" instead of "Up (LAN0)"
Steps 7e./8 No "Settings saved. Some services are being restarted..." message, but do see momentary disconnect/successful reconnect to wl1.1 on my device
Step 9c. Column name is "Bridge" instead of "Ethernet to bridge mapping"
Step 11. "wl1" is just referred to as "eth1"; No SSID names next to dropdowns
Step 12. After this reboot, I actually am still able to connect to wl1.1 (virtual wireless interface)

HommeOursPorc, 2025-03-04:
When you say can't connect, you mean the device cannot connect to the SSID or you are not receiving an IP? Did you try setting a static IP in the device.

I'm no VLAN guru and maybe I'm doing something wrong, but I had a similar issue with VLAN without the wireless part. After setting some VLANs, it look like dnsmasq DHCP server stop working for br0.

Look at this thread on LI:
https://www.linksysinfo.org/index.php?threads/need-help-with-router-accesspoint-configuration-and-vlan.79039/

I wrote: 
```
Observation: RT-N12 can't reach the internet and doesn't get time.... I tried a fiew things but nothing works...
Also, for some strange reason, DHCP on port 1 (192.168.1.0/24) stopped working after saving the VLANs configs... The only way I can access the RT-N12 via port 1 is with a static address.... Weird...
```

Maybe it is related?

I didn't mean to hijack this issue and I apologize in advance if it is not related. I hope this helps.

alejandro-n-rivera, 2025-03-04:
> I didn't mean to hijack this issue and I apologize in advance if it is not related. 

No, no, I appreciate the discussion!

> When you say can't connect, you mean the device cannot connect to the SSID or you are not receiving an IP?

Yeah, to clarify, I can't even connect to the SSID assigned to wl1.1.
I also tried with a static IP (both with DHCP on or off for br0) but this made no difference.

It seems to be more than just not being able to get an IP since, even when specifying a static one, it's like my devices can't negotiate a connection at all if that makes sense (they fail to connect to that SSID/virtual interface wl1.1 period). 

I tried this on 2 laptops (one Windows and one Linux) and an Android smartphone, and all 3 exhibited the same behavior (didn't matter if I tried with a static IP or not).

Also, the other interfaces (Ethernet Port 1 on my router or eth2 (wl0) or eth3 (wl2)) I'm assigned an IP (when DHCP enabled) on br0 just fine. So wired or wireless (5 GHz or 2.4 GHz), it doesn't matter. It seems to only be happening for that virtual wireless interface wl1.1.

And also weird:
If wl1.1 is on br0 AND wl1 is on br0, no issues connecting to wl1.1
If wl1.1 is on br0 and wl1 is on br1--ISSUE HERE, can't connect to wl1.1
If wl1.1 is on br2 and wl1 is on br1, no issues connecting to wl1.1
If wl1.1 is on br1 and wl1 is on br0, no issues connecting to wl1.1 (my current workaround)
Throughout all this, there's never an issue connecting to wl1

And finally, just confirmed (as discussed above) that this is also reproducible on FreshTomato 2024.3, but NOT reproducible on the older shibby Tomato firmware linked by pedro0311

HommeOursPorc, 2025-03-04:
Thanks for clarifying. So it is a different issue from the one I experienced. Please disregard my prior message.

alejandro-n-rivera, 2025-03-04:
Ok, I will update here! It will likely be a bit before I am able to continue testing (possibly a day or two), but I will check if it's reproducible on FreshTomato 2022.1.

If it's not reproducible in 2022.1, I'll do some investigation to see if I can pinpoint on which specific version of FreshTomato this functionality broke, which should hopefully help narrow down the cause issue.

Will update here as soon as possible. Thank you again for looking at this @pedro0311! 😄

alejandro-n-rivera, 2025-03-16:
Sorry for the extended delay. Life has a way of...happening.

I was finally able to dedicate some time to test this, and I've found that it IS reproducible in version FreshTomato 2022.1 (VPN).

I will do some research (perhaps using shibby builds after v140) to see if I can possibly narrow down when this bug may have been introduced. As soon as I have another update, I will update here.

Thanks for the patience and for the help looking into this!

pedro0311, 2025-03-17:
Thanks for testing.
It's a good idea to check also v140 by @shibby: https://tomato.groov.pl/download/K26ARM7/140-MultiWAN/
Just remember to clean nvram and use (everywhere) Ctrl+F5 because of changes in asp scripts.

pedro0311, 2025-07-18:
Will be fixed in 2025.4: https://github.com/FreshTomato-Project/freshtomato-arm/commit/e4e4ea56131b7cba9f46cf843f3fed1741bb371e