FreshTomato-Project/freshtomato-arm #75: UPnP IGD isn't showing forwarded ports

MadClown, 2025-03-05:
UPnP works properly but whenever I go into the UPnP IGD properties on windows 11 it doesn't show any forwarded ports. However it shows the forwarded ports in the router UPnP settings page. I'm using 2025.1 on a R7000.

pedro0311, 2025-03-05:
Are you sure?
Remember that a program requesting port forwarding does so when it starts and then at specified time intervals.

[image]

closed 2025-03-05 by pedro0311

MadClown, 2025-03-05:
Yes UPnP actually does forward ports fine that's not the issue I'm experiencing. It's when you go through the Windows UPnP IGD properties to view the forwarded ports that is not showing anything forwarded. I believe it might have started with the 2024.4 release.

MarsTomato, 2025-07-24:
@Self-Hosting-Group Could you show us the patch that could help this case? thx & BR

MarsTomato, 2025-07-31:
> This can easily be reproduced.
> 
>     1. Open windows explorer network view
> 
>     2. Right-click on the router and select Properties -> Settings...
> 
>     3. Add a port map for the local client with a port >1024 and then click OK
> 
>     4. Reopen the user interface. The port map will no longer be displayed without applying the suggested patch
> 
> 
> Note: The port map is created correctly even without the patch (visible in the router UI), but it is no longer displayed after creation (as the router now returns a leaseduration of >0 when compiled with IGDv2). Therefore, it cannot be edited or deleted in the windows UI.
> 
> I hope this helps.

thx 👍 , yes - very much

Do you see any negativ effects because of this change?
i tested add, delete & change with Win11 --> working/showing correct at the FT GUI (maximum lease 168 hours)

MarsTomato, 2025-08-01:
> Great, thanks for accepting the fix! However, it applies from the introduction of XP onwards, not just to >= win 10. I'll suggest a more precise commit message (PR?) and get back to you regarding your question.

perfect, just provide me a more precise commit message here and i will adjust it (No PR needed).
I only tested Windows 10 & 11, since probably no one is using Vista, 7, XP, 2000 anymore ... ;-)