FreshTomato-Project/freshtomato-arm #76: R7000 - 2025.1 - Torrent service keeps stopping

AlexPaterson77, 2025-03-12:
I always keep my Netgear R7000 updated with the latest version of FreshTomato. 

Whenever i update the firmware, i reset the Nvram, reboot, update with nvram reset, reboot.
I’m having an issue with the Transmission client since version 2024.4, tried downgrading to 2024.1 and the problem didn’t show anymore.
The issue is simple: Transmission tries to start but after some seconds, it stops.
I enabled Transmissions log at debug level to no avail, nothing strange seems to happen there. The service simply stops.
I also enabled the syslog: it doesn’t log anything.
Any suggestion ?

pedro0311, 2025-03-12:
Well, the syslog with transmission log level set to debug would be needed anyway...
What says:
```ps | grep [t]ransmission```
when it's down?
Can you open transmission GUI at this time?

pedro0311, 2025-03-12:
On the other hand, there is a watchdog added for transmission, so it should be restarted in max. 5 minutes after its die.

AlexPaterson77, 2025-03-13:
[messages.txt](https://github.com/user-attachments/files/19228691/messages.txt)

In attachment the syslog with two different service starts and the comment when the service stopped.
ps | grep [t]ransmission does indeed show a daemon up and running, though the web interface is still unavailable and torrents can not get downloaded, as the usual lifespan of transmission service being up is from 5 to 20 seconds, nothing more.

5570 root     40600 S    /usr/bin/transmission-daemon -g /tmp/mnt/New_Volume/Downloads/.settings -e /var/log/transmission.log

pedro0311, 2025-03-20:
I've got it:

```
Mar 20 16:51:30 router kern.warn kernel: transmission-da/11375: potentially unexpected fatal signal 11.
Mar 20 16:51:30 router kern.warn kernel: Pid: 11375, comm:      transmission-da
Mar 20 16:51:30 router kern.warn kernel: CPU: 1    Tainted: P             (2.6.36.4brcmarm #6)
Mar 20 16:51:30 router kern.warn kernel: PC is at 0x404a82a0
Mar 20 16:51:30 router kern.warn kernel: LR is at 0x53908
Mar 20 16:51:30 router kern.warn kernel: pc : [<404a82a0>]    lr : [<00053908>]    psr: 20000010
Mar 20 16:51:30 router kern.warn kernel: sp : 417bf5c4  ip : 404a8298  fp : 0009cdac
Mar 20 16:51:30 router kern.warn kernel: r10: 00000000  r9 : 00000001  r8 : 00000000
Mar 20 16:51:30 router kern.warn kernel: r7 : 00000000  r6 : 00000014  r5 : 00000000  r4 : 00000000
Mar 20 16:51:30 router kern.warn kernel: r3 : df8818af  r2 : 00000014  r1 : 00000014  r0 : 00000000
Mar 20 16:51:30 router kern.warn kernel: Flags: nzCv  IRQs on  FIQs on  Mode USER_32  ISA ARM  Segment user
Mar 20 16:51:30 router kern.warn kernel: Control: 10c53c7d  Table: 8df3804a  DAC: 00000015
```

AlexPaterson77, 2025-03-20:
Great @pedro0311 

In case you need anything else, do get in touch :)
I can test some snapshots.

pedro0311, 2025-03-20:
> I’m having an issue with the Transmission client since version 2024.4, tried downgrading to 2024.1 and the problem didn’t show anymore.

What about 2024.2 and 2024.3?

AlexPaterson77, 2025-03-20:
I confirm:

2024.2 and 2024.3 work correctly. I'm going to post again in case a prolonged uptime will do something funky

AlexPaterson77, 2025-03-20:
Maybe... From the changelog

- transmission: patches: revert commit 66dbc261 (add ARC4 implementation inside transmission, disable it in openssl)

pedro0311, 2025-03-26:
> transmission: patches: revert commit https://github.com/FreshTomato-Project/freshtomato-arm/commit/66dbc261c84d7719d4f3c3814a12603be0b9b78d (add ARC4 implementation inside transmission, disable it in openssl)

Nope, it was tested no changes.

pedro0311, 2025-03-27:
Are you using a VPN or AIO image?

AlexPaterson77, 2025-03-27:
Hi PedroI’m using AIO.SimoneInviato da iPhone

pedro0311, 2025-03-28:
Oh, I was too fast :(
But I have another idea.

HommeOursPorc, 2025-10-24:
Apparently using the Entware version is OK:
https://www.linksysinfo.org/index.php?threads/finally-fixed-transmission-web-crash-issue.79436/

AlexPaterson77, 2026-01-21:
`Jan 21 16:54:07 ROUTER user.info preinit[1]: starting transmission-daemon ...
Jan 21 16:54:18 ROUTER user.info preinit[8045]: transmission-daemon started
Jan 21 16:54:23 ROUTER daemon.info dnsmasq-dhcp[1108]: DHCPREQUEST(br0) 192.168.0.43 84:94:37:d5:6b:e6
Jan 21 16:54:23 ROUTER daemon.info dnsmasq-dhcp[1108]: DHCPACK(br0) 192.168.0.43 84:94:37:d5:6b:e6 Pans-Air
Jan 21 16:54:23 ROUTER daemon.info dnsmasq-dhcp[1108]: RTR-SOLICIT(br0) 84:94:37:d5:6b:e6
Jan 21 16:54:28 ROUTER daemon.info dnsmasq-dhcp[1108]: RTR-SOLICIT(br0) 84:94:37:d5:6b:e6
Jan 21 16:55:01 ROUTER daemon.notice mountd[1082]: authenticated mount request from 192.168.0.15:758 for /tmp/mnt/Volume (/tmp/mnt/Volume)

RESTART

Jan 21 16:56:59 ROUTER daemon.info dnsmasq-dhcp[1108]: DHCPDISCOVER(br0) 94:58:cb:c1:72:5c
Jan 21 16:56:59 ROUTER daemon.info dnsmasq-dhcp[1108]: DHCPOFFER(br0) 192.168.0.6 94:58:cb:c1:72:5c
Jan 21 16:56:59 ROUTER daemon.info dnsmasq-dhcp[1108]: DHCPDISCOVER(br0) 94:58:cb:c1:72:5c
Jan 21 16:56:59 ROUTER daemon.info dnsmasq-dhcp[1108]: DHCPOFFER(br0) 192.168.0.6 94:58:cb:c1:72:5c
Jan 21 16:56:59 ROUTER daemon.info dnsmasq-dhcp[1108]: DHCPREQUEST(br0) 192.168.0.6 94:58:cb:c1:72:5c
Jan 21 16:56:59 ROUTER daemon.info dnsmasq-dhcp[1108]: DHCPACK(br0) 192.168.0.6 94:58:cb:c1:72:5c
Jan 21 16:57:09 ROUTER authpriv.info dropbear[9423]: Child connection from 176.120.22.17:51004

Jan 21 16:57:19 ROUTER user.info preinit[1]: starting transmission-daemon ...
Jan 21 16:57:30 ROUTER user.info preinit[9508]: transmission-daemon started

DOWN

Jan 21 16:57:59 ROUTER daemon.info dnsmasq-dhcp[1108]: RTR-SOLICIT(br0) 94:27:70:bb:af:c1
Jan 21 16:58:05 ROUTER authpriv.info dropbear[9859]: Child connection from 176.120.22.17:25632
Jan 21 16:58:06 ROUTER authpriv.warn dropbear[9859]: Bad password attempt for 'root' from 176.120.22.17:25632
Jan 21 16:58:06 ROUTER authpriv.info dropbear[9859]: Exit before auth from <176.120.22.17:25632>: (user 'root', 1 fails): Error reading: Connection reset by peer
Jan 21 16:58:07 ROUTER authpriv.info dropbear[9901]: Child connection from 176.120.22.17:25638
Jan 21 16:58:08 ROUTER authpriv.warn dropbear[9901]: User 'admin' has invalid shell, rejected
Jan 21 16:58:09 ROUTER authpriv.info dropbear[9901]: Exit before auth from <176.120.22.17:25638>: (user 'admin', 1 fails): Error reading: Connection reset by peer
Jan 21 16:58:09 ROUTER authpriv.info dropbear[9423]: Exit before auth from <176.120.22.17:51004>: Timeout before auth
`

pedro0311, 2026-02-26:
Good to hear, thank you.

lancethepants, 2026-02-27:
If it's a matter of resources/memory, I wonder if mounting a swap partition would help.

AlexPaterson77, 2026-03-19:
I managed to start the optware transmission-daemon manually by triggering this command from bash

/opt/etc/init.d/S88transmission start

That version of the client is now working correctly and stays up, even though it does not start by just changing the Transmission binary path in the FT GUI, you need to start it manually.
Two things to fix:

- Understand why setting the GUI to use the binary on /opt/bin does not start the daemon and logs an error saying "check settings.json"

- Change the default binaries on the R7000 AIO build of fresh tomato with the one from optware. Default hangs, optware works.

pedro0311, 2026-05-13:
TLDR: no.