FreshTomato-Project/freshtomato-arm #77: Connecting to virtual WiFi not working when MAC address is spoofed.

Lalarian, 2025-03-19:
Hello!

My [R7000](https://amzn.to/32SEjYr) is on latest FreshTomato.
Already for some versions before, I have noticed that when I try spoofing a WiFi-Mac for my virtuakl WiFi AP's (Advanced -> MAC Address), then I cannot connect to the AP anymore.

Mobile devices and Linux notebook/PCs simply try connecting a few times and then stop, without any notification.
A Windows 10 client simply said 'Autorization error'.
As soon as I return the MAC to the original, it's working again.
I also tried changing just the UAA part of the MAC address, but it still did not work.

For making sure I also rebooted the router after changing the MAC, to no success.
I have multiple virtual WLANs for 2.4 GHz as well as 5.0 GHz, maybe it's connected to this issue ..

Browsing the liniksysinfo forum I learned about the wl command. And using this command I found the following issue with the MAC spoofing active (meanin: Another MAC address than the default MAC being set ):

wl dump
BSS Config 3: "temp"

    enable 1 up 1 wlif 0xcf156800 "wl0.3"
    wsec 0x4 auth 0 wsec_index -1 wep_algo 0
    current_bss->BSSID 00:00:00:00:00:00
    current_bss->SSID ""

As soon as I change the mac of a virtual WiFi the BSSID is set to 00:00:00:00:00:00 instead of the MAC provided in the MAC address input form.
For example, the MAC I have actually set wl0.3 to right now is:
C6:04:15:3E:0E:E8

And using the 'nvram show | grep wl0.3' command, I got this for the NVRAM variable for wl0.3'S MAC:
wl0.3_hwaddr=CA:6F:E8:82:E0:B5

So the input form is working, but instead of the given MAC, it's set to all zeroes.

pedro0311, 2025-03-19:
Maybe @MarsTomato our WL master ;) could say something about it here?

MarsTomato, 2025-03-21:
You should (always) leave the **first 3 octets** alone if you need to adjust the mac addr 🙂
--> try to adjust the remaining octets, working?

Lalarian, 2025-03-23:
Hello!
-I tried changing all octets.
-I tried changing only the last three.
-I tried changing just the last one.
-I also tried using the premade "OUI + Random LLA" button which just changes the last three octets.
-I tried using only even hex digits (which also seems to be a requirement for the OUI part and which is what the OUI+LAA button does besides randomizing the LAA part)

MarsTomato, 2025-07-19:
> Commit [e4e4ea5](https://github.com/FreshTomato-Project/freshtomato-arm/commit/e4e4ea56131b7cba9f46cf843f3fed1741bb371e) may fix this in 2025.4 should have this fix.

you can it it with the following images for example https://github.com/MarsTomato/freshtomato-arm/releases/tag/19072025