FreshTomato-Project/freshtomato-arm #80: Dynamic DNS does not update, no messages in logs regarding failed synchronisation

quickishfm, 2025-03-27:
On my Netgear R7000 running `FreshTomato 2024.5 K26ARM USB AIO-64K`, the dynamic DNS setting is set to Daily (`Every 1 days`) yet it did not update for over a week. This happened before on 2024.1 but I upgraded thinking it might fix the issue.

As of currently I have not rebooted the router, as I want to debug why this is happening. I have logging turned on but the logs have nothing regarding Dynamic DNS (only logs being when DDNS was turned on and off).

How can this issue be debugged so I can see why the DDNS has not been updating? I have SSH access to the router and it has not been updated since the screenshot so I can poke around over SSH if need be.

Attached is an image of my DDNS settings

[image]

HommeOursPorc, 2025-03-27:
There was some update for mdu in 2025.1.

```
- mdu: the eNom service has its cert fixed
- mdu: fix memory leak in curl_dump_cb() callback functi
on
- mdu: make update_cloudflare() work again
```

Can you please upgrade to 2025.2 and test again. 2025.1 had an issue with dnsmasq, I advise to skip 2025.1 and upgrade to 2025.2 🙂

quickishfm, 2025-03-28:
Thanks for the prompt response! I've upgraded to 2025.02 and I'll keep an eye on it over the next few weeks to see how it goes. Shall I close this for now, or leave it open until I can verify in a few weeks that it works?

HommeOursPorc, 2025-03-28:
I'm not a dev, just a random "helper". Please leave it open and report back when possible. I'll leave it to @pedro0311 if this need to be kept open or closed.

pedro0311, 2025-03-28:
I'm closing this as it concerns an outdated version of FT and may cause confusion.

closed 2025-03-28 by pedro0311