FreshTomato-Project/freshtomato-arm #81: Advanced > VLAN - Trailing '/' in link doesn't redirect to the correct forum post

HommeOursPorc, 2025-03-29:
https://github.com/FreshTomato-Project/freshtomato-arm/blob/c6f57c96a3e9d6433fa17106cf35765e8c2d18a0/release/src-rt-6.x.4708/router/www/advanced-vlan.asp#L1063

At the line above ^^ the link should not have a trailing '/'

http://www.linksysinfo.org/index.php?threads/can-vlan-gui-port-order-be-corrected.70160/#post-247634**/** <-- '/' should be removed. 

I wanted to do a PR, but I don't know how to do it with the web interface? I can do it directly, I think I need to clone the repo first but I couldn't find how to clone from GitHub to GitHub via the web interface.

closed 2025-04-02 by pedro0311

pedro0311, 2025-04-03:
@HommeOursPorc : Don't you have an edit link in the top right corner?

[image]

HommeOursPorc, 2025-04-03:
> [@HommeOursPorc](https://github.com/HommeOursPorc) : Don't you have an edit link in the top right corner?
> 
> [image]

Yes? Clicking it give me 2 choices, edit in place or open with GitHub.dev.

Clicking edit in place in your repo brings me to the fork page. I didn't check what GitHub.dev is. I'm on my phone so not very practical.

HommeOursPorc, 2025-04-03:
> [@HommeOursPorc](https://github.com/HommeOursPorc) : Don't you have an edit link in the top right corner?
> 
> [image]

Ohhhhh ok!

[image]

I will look into this the next time. Thanks! 😃

commit ec253967 (2025-04-02): GUI: Advanced: VLAN: fix link in Notes (close #81)