FreshTomato-Project/freshtomato-arm #85: LAN DHCP IP not assigned to clients when configured wireless client SSID is not present/working

wellloaded, 2025-04-07:
How to reproduce:

Set up FT in wireless client with the whatever second radio in AP.
Now either power-off the main router/AP and reboot FT, or simply move the FT device + client/s to a place where the parent wireless network is not reachable. e.g. the device needs to power on and the configured parent wireless SSID is unreachable.

At this point, any client trying to connect to FT **on the LAN side** will not get an IP.

MarsTomato, 2025-04-10:
Not sure if i understand the problem correct. But i did create a test setup with wireless client and i was able to get a LAN IP even if i did remove the AP (for WL client)
|AP| <- interrupt for testing ->|WL Client + LAN| <-remove/add-> LAN Client

wellloaded, 2025-04-10:
Thank you for taking the time to test this.
One thing I didn't mention is: once you disconnect/power-off the main AP you should also restart FT, this means FT will boot up without a WC wan.

I can't unfortunately test this from where I am right now, but that's what I did before opening the ticket. Essentially I have a WC set up in one site, then I moved to another site, I just wanted to log in and couldn't get an IP from FT, so I had to set it up statically.

MarsTomato, 2025-04-11:
> Thank you for taking the time to test this. One thing I didn't mention is: once you disconnect/power-off the main AP you should also restart FT, this means FT will boot up without a WC wan.
> 
> I can't unfortunately test this from where I am right now, but that's what I did before opening the ticket. Essentially I have a WC set up in one site, then I moved to another site, I just wanted to log in on FT to set the new wireless details of the new site, but I couldn't get an IP from FT in this scenario, so I had to set it up statically on the laptop.

I also did reboot the WL client router (AP wireless lan disabled) + unplugged my laptop. After the router finally booted up, i plugged the lan cable back in and did a get a LAN IP. Internet was of course not working.

wellloaded, 2025-04-11:
Thanks for the feedback. I will re-test this as soon as I get a WiFi into Internet available. e.g. next week.

HommeOursPorc, 2025-06-14:
@wellloaded with which version did this occurred? Is this still a problem with the latest version?

Linked to this (no DHCP if WAN is down)?
https://github.com/FreshTomato-Project/freshtomato-mips/issues/6

HommeOursPorc, 2025-06-18:
I tried to replicate this issue. Everything works correctly on both ARM and MIPS for version 2025.2
I did got the problem on MIPS 2024.5 once, but it seems OK on 2025.2

closed 2025-07-01 by pedro0311