FreshTomato-Project/freshtomato-arm #88: Suggested text edits to VLAN menu Notes section

whateverusernameisnottaken, 2025-04-17:
I suggest the following edits to the VLAN menu's Notes section:

* VLAN - the VLAN's unique identifier. Valid settings are 0 to 15. This is locally significant and should be considered as a GUI reference only.
* VID - The label that identifies to which VLAN the frame belongs. Valid entries are 1 to 4094. When set, binds VLAN & VID (Such as: VLAN: 4 to VID: 4) on a port basis.
The VLAN and VID number of reference can match but needn't. The special VID "0" means no label and therefor no VLAN configuration applied to frames.
* Ports 1 to 4 and WAN - specify which Ethernet ports will be members of this VLAN.
* Tagged - enables 802.1Q tagging of Ethernet frames on a particular port. This allows multiple VLANs to be configured one given port.
* Default - the VLAN ID assigned to untagged frames received by the router.
* Ethernet-to-bridge mapping - determines the binding between VLAN and physical interfaces such as WANx or LANx. A mapping with "none" is considered unused.

For the text below it, I would do these edits:

It's wise to cross-check settings in the Basic/Network and Advanced/Virtual Wireless menus.
Be mindful that some FreshTomato releases might use VID: 0 as default others VID: 1.
One VID must be assigned to the WAN bridge.
One VID must be selected as the default VLAN.
To prevent 802.1Q compatibility issues, avoid using VID: "0", as 802.1Q specifies that frames with a tag of "0" do not belong to any VLAN. The tag contains only user priority information.
If you trunk with other vendors, note that some discourage from using VID: "1" as it may be specially reserved (for management purposes).

pedro0311, 2025-06-09:
It would be nice, if @lancethepants and/or @HommeOursPorc could review these changes.

HommeOursPorc, 2025-06-09:
I'll look it up soon. 

First thing I noticed:

"Be mindful that some FreshTomato releases might use VID: 0 as default others** VID: 1 for **LAN**."

Not sure how to write it though 🤔

Maybe "as default for LAN, others VID: 1"

"VLAN - the VLAN's unique identifier. Valid settings are 0 to 15. This is locally significant and should be considered as a GUI reference only."

GUI reference? Is this some kind of limitation? Max 16 VLANs? 

"VID - The label that identifies to which VLAN the frame belongs. Valid entries are 1 to 4094. When set, binds VLAN & VID (Such as: VLAN: 4 to VID: 4) on a port basis."

Label... Is this the correct term? Should we use "tag" instead?

"The VLAN and VID number of reference can match but needn't. The special VID "0" means no label and therefore no VLAN configuration applied to frames."

Since English is my second language, I find the first sentence here to be a bit weird. It is probably correct, but for non English speaker it might be hard to understand. Maybe "doesn't need to match" or "doesn't need to be the same".

"Bridge $wireless_if to $lan_bridge - For each (physical or virtual) wireless interface, specify to what LAN bridge it will be mapped"

Specify to"what" or to "which" ?

Bridge the $wireless_if to the selected $lan_bridge - map the wireless interface to the selected LAN

Just some ideas. Let me know what you guys think 🙂

HommeOursPorc, 2025-06-09:
This was a mess 😂. I'll start over.

> I suggest the following edits to the VLAN menu's Notes section:
> 
>     * VLAN - the VLAN's unique identifier. Valid settings are 0 to 15. This is locally significant and should be considered as a GUI reference only.
> 
I think this is good. I would like to know if we are effectively limited to 16 VLANs in the GUI? Maybe we should add this to the notes my?

>     * VID - The label that identifies to which VLAN the frame belongs. Valid entries are 1 to 4094. When set, binds VLAN & VID (Such as: VLAN: 4 to VID: 4) on a port basis.
> 
Not sure about the term label. I looked around rfc-editor.org and some Cisco stuff and I don't see the term label. I understand the idea though. Google Gemini call it a "numerical label or identifier". So I suppose label is ok.

>       The VLAN and VID number of reference can match but needn't. 
> 
I think this should be formulated differently. 
"The VLAN number of reference and the VID can be the same or different number, they can but they do not need to match."

Thoughts?

>      The special VID "0" means no label and therefore no VLAN configuration applied to frames.
> 
From Wikipedia: The reserved value 0x000 indicates that the frame does not carry a VLAN ID; in this case, the 802.1Q tag specifies only a priority (in PCP and DEI fields) and is referred to as a priority tag.

I'm not sure if this mean that the frame is untagged?? Maybe this could be reformulated somehow...

>     * Ports 1 to 4 and WAN - specify which Ethernet ports will be members of this VLAN.
> 
>     * Tagged - enables 802.1Q tagging of Ethernet frames on a particular port. This allows multiple VLANs to be configured one given port.
> 
Configured on one given port? Configured To one given port? 🤔 

>     * Default - the VLAN ID assigned to untagged frames received by the router.
> 
>     * Ethernet-to-bridge mapping - determines the binding between VLAN and physical interfaces such as WANx or LANx. A mapping with "none" is considered unused.
> 
> 
> This sentence doesn't make sense as is: Bridge $wireless_if to $lan_bridge - For each wireless interface define (physical or virtual) specify to what LAN bridge this has to map
> 
> Did you mean: "Bridge $wireless_if to $lan_bridge - For each (physical or virtual) wireless interface, specify to what LAN bridge it will be mapped"
> 
Bridge $wireless_if to the selected $lan_bridge - For each (physical or virtual) wireless interface, specify to what LAN bridge it will be mapped

> Other relevant notes/hints:
> 
>     * It's wise to cross-check settings in the Basic/Network and Advanced/Virtual Wireless menus.
> 
>     * Be mindful that some FreshTomato releases might use VID: 0 as default others VID: 1.
> 
Be mindful that some FreshTomato releases use VID: 0 as their default LAN VID and others use VID: 1.

>     * One VID must be assigned to the WAN bridge.
> 
Maybe we should specify that it is normally recommended to leave the wan bridge alone. The WAN port can be mapped as pleased, but, for example, VID 2 (vlan2) should be leaved alone and set to WAN0 bridge.

I don't know for sure, but the rumor is that some WAN stuff is hard-coded and depends on the default WAN bridge VLAN.... Or something like that...

>     * One VID must be selected as the default VLAN.
> 
>     * To prevent 802.1Q compatibility issues, avoid using VID: "0". 802.1Q specifies that frames with a "0" tag don't belong to any VLAN. The tag contains only user priority information.
> 
>     * If you trunk with other vendors, note that some discourage from using VID: "1" as it may be specially reserved (for management purposes).

Hope this helps 🙂

whateverusernameisnottaken, 2025-06-10:
I'll get to it, pedro. Just busy at the moment. I think you can trust my changes by now. You're not happy with the work done on the wiki so far? How many complaints do you see about people having difficulty understanding the wiki text?

lancethepants, 2025-07-25:
> * VLAN - the VLAN's unique identifier. Valid settings are 0 to 15. This is locally significant and should be considered as a GUI reference only.

Not 100% true on this change. When creating a new vlan entry and the VID column has been left at '0', the VID value will be set to the VLAN value.
https://github.com/FreshTomato-Project/freshtomato-arm/blob/arm-master/release/src-rt-6.x.4708/router/rc/init.c#L922-L939

This also ensures that if VID is '0' that the VLAN number will be used instead.
https://github.com/FreshTomato-Project/freshtomato-arm/blob/arm-master/release/src-rt-6.x.4708/router/rc/network.c#L1680-L1684

I find the VLAN page Notes hard to follow.  I don't like the existing 'VID' note with it's terms of 'traditional', or 'regular'.  And the set to '0' is wrong because it will be changed from 0 to something else anyways. 

I think people will be looking for terms like 'tagged' and 'untagged'.  I think the terminology needs to be simpler which I understand is hard because VLANs are very technical. Personally I think we could completely throw out the current notes and start from scratch.  In the forums I typically resort to using screenshots of configurations to explain things.   Maybe it could be useful to show screenshot(s) with some common configurations and an explanations of what is happening with settings.  I know there is also the wiki for that kind of stuff, but to have something directly in the Tomato UI for this complicated kind of feature could reduce confusion and questions in the forums?

carakesa, 2026-09-14:
> Maybe we should specify that it is normally recommended to leave the wan bridge alone. The WAN port can be mapped as pleased, but, for example, VID 2 (vlan2) should be leaved alone and set to WAN0 [...]

I'm confused by this wording too.  Should VID 0 not be used for the default VLAN?  Or is it "don't use 802.1Q tags on VID 0"?