FreshTomato-Project/freshtomato-arm #90: DDNS - possible issue with Service: Cloudflare (https)

SpiralBrad, 2025-04-19:
I upgraded to 2025.2, attempted to use the Cloudflare DNS update (for the first time) - 

1. Set hostname to a full hostname (subdomain.example.com) for a domain I have in my Cloudflare panel
2. Unchecked 'proxied' as I don't want it for this
3. Left 'Create record if needed' checked, as I know the subdomain does not exist
4. Set the Zone ID to the zone given by the Cloudflare panel for the domain in question
5. Set 'force next update' so that I'd get an immediate success or failure
6. Created a restricted API key with only the Zone.DNS permission, and set that in FreshTomato

With that restricted Zone.DNS API key, after clicking Save in FT, letting the services restart, and refreshing the DDNS page, I get the "Last Result" of:

`PUT returned HTTP error code 405.`

So then as a test, I switch the API key to my Global API key, rechecked 'force next update', etc etc..., I then get a "Last Result of:

`GET returned HTTP error code 400.`

Thus, I seemingly can't get this to work at all.

Is there something I'm missing?

nigelboid, 2025-05-12:
I am having the same issue, so I simply created a Scheduler task with the following code:

`myAddress=$(curl -s https://api.ipify.org) && {`
`dnsAddress=$(/usr/sbin/hostip host.example.com)`
`if [ "$myAddress" != "$dnsAddress" ]`
`  then`
`    result=$(curl https://api.cloudflare.com/client/v4/zones/$ZONE_ID/dns_records/$DNS_RECORD_ID \`
`      --silent \`
`      -X PATCH \`
`      -H 'Content-Type: application/json' \`
`      -H 'Authorization: Bearer $CLOUDFLARE_API_TOKEN' \`
`      -d '{`
`            "content": "'$myAddress'",`
`            "name": "host",`
`            "proxied": false,`
`            "ttl": 300,`
`             "type": "A"`
`          }')`
`  else`
`    result="Address remains the same: $myAddress ($dnsAddress)"`
`  fi`
`logger -p syslog.info $result`
`}`

More here: https://developers.cloudflare.com/api/resources/dns/subresources/records/methods/edit/

I am using a token in the example above. Change it per Cloudflare's example code to use an API key.

handy-ha, 2025-06-22:
Thanks! I tried many different methods before, but none of them worked.
For some reason, my router doesn't support scheduled tasks—but at least I can now set up a daily schedule on my mini PC to keep the IP address updated for my VPN connection. I'm also using a secondary free DDNS URL as a backup, just in case.

dasm, 2025-10-03:
FreshTomato version: 2025.3

@SpiralBrad @handy-ha I stumbled upon the same issue today. 

I tried to configure DDNS multiple times. I used "Zone ID: 1" which was displayed in the window for the API Token.
Later I noticed 32-chars `Zone ID` field in the main config view for my domain.

When I provided the one, I finally was able to configure my DDNS.