FreshTomato-Project/freshtomato-arm #92: Feature Request: make histname clickable if its port 80 or 443 is reachable

wellloaded, 2025-04-27:
In status-devices (following Fritz devices example) we could have the hostname and/or IP clickable and opened in a new tab if this is responding on port 80 and/or 443

whateverusernameisnottaken, 2025-05-01:
What would be the purpose of this?

HommeOursPorc, 2025-08-16:
> What would be the purpose of this?

I use [upsnap wake-on-lan](https://github.com/seriousm4x/UpSnap) and you can click on the device name to access it which I personally really like. What is very nice is that you can customize the link for each device. I can easily access my different dashboards, router GUI, etc. 

[image]

@wellloaded didn't you make the device discovery script?? I don't know exactly how this could be achieved. In shell we have `pscan` that maybe could be useful here. Maybe name it scan the ports and somehow make the page (JavaScript??) aware of it, maybe like a hidden flag that when the page render make the name clickable is the flag is there, indicating that that particular device can be reached... Just thinking out loud...

wellloaded, 2025-09-02:
> [@wellloaded](https://github.com/wellloaded) you wrote the script so I presume you would know if she works or not you know it better than I do... Or how to fix it.

As I understand, your [commit](https://github.com/FreshTomato-Project/freshtomato-arm/pull/132/commits/12b793dec3065bf3b27702b04c597fbe35ab090a) makes **all** the IP/hostnames clickable.

It is a good starting point, 

The current discovery.sh script can't help much here though. Discovery uses a number of network probes to assess if a host on the network is present or not. Nothing else. This particular feature request is to test the populated list of devices (where the discovery.sh might have helped to spot silent hosts) if any of those hosts has port 80 or 443 opened.

We need to assess how to implement the "secondary" scan on those ports only via JS. It's essentially a modification of your code where you assume all the hosts have port 80,443 opened.

BTW it would be good to think of a way a user might tweak whatever code we come up with to allow different/additional ports like 8080 or whatelse makes sense in his/her network to be configured.

Finally, let's remember: JS code is run from the client machine and this might be accessed remotely. So few things to keep into account, but we can start simple and improve later.

wellloaded, 2025-09-03:
This second version outputs to /tmp/scan_results

```
#!/bin/sh
#portscan v0.2 rs232
ip="$1"
ports="80 443 8080 $2"
open_ports=""

for port in $ports; do
  # Scan single port with 1 second timeout
  result=$(pscan -p $port -P $port -t 1000 $ip | grep "$port")
  if echo "$result" | grep -q "open"; then
    open_ports="$open_ports $port"
  fi
done
open_ports=$(echo "$open_ports" | sed 's/^ *//')
echo "$open_ports"
scan_file="/tmp/scan_results"

# Create file if not exists
touch "$scan_file"

# Remove any existing line for this IP
sed -i "/^$ip /d" "$scan_file"

# If open ports exist, add the line for this IP
if [ -n "$open_ports" ]; then
  echo "$ip $open_ports" >> "$scan_file"
fi
```

wellloaded, 2025-09-03:
This is as far as I got today. 

Instructions:

1) mount --bind the status-devices.asp uploaded here
[status-devices.zip](https://github.com/user-attachments/files/22126933/status-devices.zip)

2) create the scan.sh script from the post here above under e.g. /tmp

3) Either run the script passing the IP (hopefully one where port 80 or 443 are opened) or create a bogus /tmo/scan_results containing e.g. "192.168.1.1 80" or something. Even if the port is not opened! Just as a test...

4) Verify the content of the /tmp/scan_results looks ok

5) refresh the status-devices. After the first refresh, a link icon should appear on the right of the IP. Clicking on it, it opens the relevant link on a new tab. Checking the developer mode via F12 should give additional debugging info if ever needed.

[image]

So where we are is:
- hopefully this is a workable solution, despite being slightly fiddly
- the scan.sh integration needs to be worked out to be either:
a) triggered automatically by the webpage which BTW is the most accurate source of IPs we have, even better than the arptable as an independent timer or hooked to the discovery.sh (easier) to run e.g. once every 5 discoveries
b) triggered manually with a button or clicking on the IP