FreshTomato-Project/freshtomato-arm #93: wireguard setting the wrong peer endpoint

rmeden, 2025-05-05:
FreshTomato Version 2025.2 on Asus RT-AC68R/U ( freshtomato-RT-AC1900U-K26ARM-2025.2-VPN-64K )

Wireguard peer set up with a hostname (DDNS, but it doesn't matter)

If I download (or decode the QR code) the peer config it includes
`Endpoint = 192.168.1.71:51820`

192.168.1.71 is the IP address of my WAN port (behind an AT&T fiber router)

Of course the client won't connect... if I edit the endpoint on the client to the hostname, it works fine.

pedro0311, 2025-05-06:
Depending on the "Peer Parameters" -> "Endpoint" setting, it can only be the WAN FQDN, WAN IP, or a manually entered value. There is no other option. If you have a modem/another router in front of the router, you must enter this value manually - that's what "Custom Endpoint" is for.

Oh, and it was always like this.

closed 2025-05-06 by pedro0311

pedro0311, 2025-05-06:
Not here, on "Config" tab as I said earlier "Peer Parameters" -> "Endpoint".

rmeden, 2025-05-06:
Gotcha, the "Endpoint" parameter appears on both the "config" and "peer" tab "peer parameters" section.

When is the Peer tab peer parameter endpoint ever used?  It doesn't look like it overrides the Config tab endpoint parameter... should it?

(I confirmed setting custom endpoint on the config tab works)

mjraneri, 2025-07-13:
I agree the UI is effective and the QR codes can be generated with the correct information.  However given a bit of wireguard experience (not an expert) it wasn't clear to me that "Peer Parameters" on the "Config" page are primarly for the QR code generation to be passed to the peers you create in the "Peers" tab.  They don't actually impact the wireguard config on the FreshTomato Router, right?

Anyway, it works, but I think the heading could be more clear that those parameters aren't for a peer the router would be connecting to.  They are info for the peers to connect to the router.

(Perhaps I should have realized this because there wasn't a place to paste a public key.)

Also, the router doesn't need the private key to enable the connection between the two endpoints...  Why do I have to provide the peer's private key if I want to create the QR code if it's not required to make the connection?  One of my peers doesn't want to send me his private key. (Which is understandable.)

pedro0311, 2025-07-13:
> However given a bit of wireguard experience (not an expert) it wasn't clear to me that "Peer Parameters" on the "Config" page are primarly for the QR code generation to be passed to the peers you [...]

https://github.com/FreshTomato-Project/freshtomato-arm/commit/dd819a0c7adf129861100008c25bfac6eab7e15c#diff-cbd6b4549210b6c4eee90fa9eddc1c53ac75c0329d694a7d49395d0b71a323f2R1972

> Why do I have to provide the peer's private key if I want to create the QR code if it's not required to make the connection?

https://github.com/FreshTomato-Project/freshtomato-arm/commit/91d0289475b8f4ad894ab181fec12cf7475e5edf