FreshTomato-Project/freshtomato-arm #95: Connection to Linksys EA6700 FreshTomato via SSH either times out or latency slow builds overtime once connected

thecodygriffin, 2025-05-10:
### Description
When I try to connect to FreshTomato via SSH over a wireless network connection, the attempt often times out and/or latency slowly builds overtime.

### Environment Details
* Linksys EA6700 on FreshTomato 2024.1 configured as a router (i.e. gateway, dhcp, etc)
* Linksys EA6700 on FreshTomato 2024.1 (ARM) configured as an access point (note: wired only most of the time)
* Both are configured with IPv6 disabled, intentionally
* Cut Through Forwarding (CTF) and HW Acceleration is enabled "so QoS doesn't work."

### What I Have Tried & Results
* Upon connecting to either the router or access point via a wired ethernet connection, the issue is NOT reproducible
-- Confirmed as NOT reproducible on bare metal and virtual machines through a bridge
* If I am connected to the router via a wireless connection, upon attempting to connect to the access point (which is connected to the router via ethernet), the issue is NOT reproducible
* Upon attempting to connect via 2.4GHz or 5.0GHz wireless network, the issue is reproducible
* Upon attempting to connect from Fedora or Windows 10 (different hardware) on a wireless network, the issue is reproducible
* Upon attempting to connect from the same Fedora or Windows 10 machines on a wired network, the issue is NOT reproducible
* Upon attempting to connect using `ssh -4 username@ipaddress` or `ssh -o IPQoS=none username@ipaddress`, the issue is reproducible

### Ask
Has anyone else experienced this issue or something similar? If so, is there a configuration within FreshTomato that I should change to resolve the issue?

The issue has been reproducible since at least FreshTomato 2023.1 and I haven't seen anything in the release notes for subsequent releases that would have addressed it. Given upgrading FreshTomato requires a complete reconfiguration, I am delaying such an upgrade until either this issue is resolved and/or there is a good reason to do so.

Any thoughts or insight would be greatly appreciated. Thank you.

HommeOursPorc, 2025-05-10:
Hello, did you try to disable CTF?
Are your wireless country and region code correctly configured?
Other than SSH, does the wireless reliability and speed is also affected by this issue? Are you getting slow wifi overall? 
Have you tried with a different channel or channel width and/or different sideband? Which Wireless Network Mode? Which security settings?

pedro0311, 2025-05-10:
Sorry, but this is not a matter for an issue, but for creating a new thread on our forum to find out what you are doing wrong.
https://www.linksysinfo.org/index.php?forums/tomato-firmware.33/

closed 2025-05-10 by pedro0311