FreshTomato-Project/freshtomato-arm #96: Netgear EX7000 switch needs these to work

gabest11, 2025-05-12:
```
et robowr 0x10 0x0 0x1940
et robowr 0x11 0x0 0x1940
et robowr 0x12 0x0 0x1940
et robowr 0x13 0x0 0x1940
et robowr 0x14 0x0 0x1940
et robowr 0x10 0x0 0x1140
et robowr 0x11 0x0 0x1140
et robowr 0x12 0x0 0x1140
et robowr 0x13 0x0 0x1140
et robowr 0x14 0x0 0x1140
```

Already made a post at https://www.linksysinfo.org/index.php?threads/add-support-for-netgear-ex7000.78502/page-8

It would be nice if these could be included in the firmware. Executing them through serial console after each boot is not a solution. Maybe there is a better way, I don't know.

gabest11, 2025-05-12:
This is my quick and dirty solution.

/router/rc/init.c: sysinit(), just after set_jumbo_frame();

```
	switch (get_model()) {
	case MODEL_EX7000:
		eval("et", "robowr", "0x10", "0x0", "0x1940");
		eval("et", "robowr", "0x11", "0x0", "0x1940");
		eval("et", "robowr", "0x12", "0x0", "0x1940");
		eval("et", "robowr", "0x13", "0x0", "0x1940");
		eval("et", "robowr", "0x14", "0x0", "0x1940");
		eval("et", "robowr", "0x10", "0x0", "0x1140");
		eval("et", "robowr", "0x11", "0x0", "0x1140");
		eval("et", "robowr", "0x12", "0x0", "0x1140");
		eval("et", "robowr", "0x13", "0x0", "0x1140");
		eval("et", "robowr", "0x14", "0x0", "0x1140");
		break;
	}
```

gabest11, 2025-05-12:
These two patches cannot compile.
/router/patches/ebtables/104-do-not-build-ipv6-extension.patch
/router/patches/samba3/020-CVE-preparation-v3-6.patch

gabest11, 2025-05-13:
Found the meaning of these in a [datasheet](https://xonstorage.blob.core.windows.net/pdf/broadcom_bcm53128vkqleg_apr22_xonlink.pdf) on page 194. 

Basically it toggles bit 11 for all port. "Power-down", then back to "Normal operation".

pedro0311, 2025-05-13:
Already in PR: https://github.com/FreshTomato-Project/freshtomato-arm/pull/97

closed 2025-05-13 by pedro0311

MarsTomato, 2025-05-14:
> ```
> et robowr 0x10 0x0 0x1940
> et robowr 0x11 0x0 0x1940
> et robowr 0x12 0x0 0x1940
> et robowr 0x13 0x0 0x1940
> et robowr 0x14 0x0 0x1940
> et robowr 0x10 0x0 0x1140
> et robowr 0x11 0x0 0x1140
> et robowr [...]

ahh nice. I would like to move the code/fix in other position - hope thats ok :-)

gabest11, 2025-05-14:
Yea. I put it in sysinit, but there is probably a nicer place. Not sure. Maybe refactor into a function, preferablly with ifdefs that are only valid for EX7000 build

gabest11, 2025-05-14:
Btw, I have not mentioned this yet, but before I found FreshTomato, I was experimenting with dd-wrt. Their Trendnet 818DRU firmware did work the same way, only the switch was not working as well.  Probably the same hardware,

MarsTomato, 2025-05-14:
> Yea. I put it in sysinit, but there is probably a nicer place. Not sure. Maybe refactor into a function, preferably with ifdefs that are only valid for EX7000 build

i will prepare the commit for EX7000.

THX & BR

pedro0311, 2025-05-14:
Thanks, @MarsTomato !