FreshTomato-Project/freshtomato-arm #113: 2025.3 Bricks TendaAC15, 2025.2 doesn't

rugabunda, 2025-07-17:
Client of mine has been updating tomato fine for years without issue, suddenly 2025.3 bricked the device; in 2025.2 device was configured to disable LEDS; After installing 2025.3 leds are enabled. Lan leds stay on if wired. No led light on power, wifi, etc. No DHCP attainable. Device is soft bricked.

Reset NVRAM in CFE recovery mode and restarted; problem persisted.
Downloaded a second time, Re-flashed 2025.3 in CFE; problem persisted.
Flashed last working version (2025.2) in CFE, problem disappeared; loaded previous CFG, no problems.

2025.3 appears to brick TendaAC15.

Anything I can do to help?

MarsTomato, 2025-07-17:
please test https://github.com/MarsTomato/freshtomato-arm/releases/tag/17072025

closed 2025-07-18 by rugabunda

MarsTomato, 2025-07-18:
perfect :-)

pedro will rebuild this 2025.3 release (one/two more fixes detected ... ) so you can just update normally

MarsTomato, 2025-08-29:
@pedro0311 please replace the Tenda AC15 image one more time with release 2025.3

Looks like Tenda did use many differents parts to build the routers... 
Macronix (first source) --> Working
GigaDevice (second source) --> Working (fix 1)
Winbond (third source) --> open (fix 2)

see commit / fix 2
https://github.com/FreshTomato-Project/freshtomato-arm/commit/2ce9045b2ab4c0116421f5172f9de260cca8b5ec

pedro0311, 2025-08-29:
Done!