FreshTomato-Project/freshtomato-arm #116: Roaming Assistant not working anymore

zakk87, 2025-07-20:
Some time ago I was using Roaming Assistant. I have noticed for some time that it is not working completely. No logs regarding disconnection of clients.

[image]

[image]

Router: ASUS AC66U_B1
FT: 2025.03

MarsTomato, 2025-07-26:
Hi zak,

this can be/is normal. I probably need to add that to the Freshtomato Wiki :-)

There are two criteria for the roaming assistant:
1) RSSI
2) idle rate of you client (default rate for checking < **2 Kbps**)

You can test that for example: if your client constantly is downloading something it will not be kicked off your wireless lan, even if the RSSI criteria is crossed.

If the default idle rate (hidden option for ARM branch only) is too low for you, you/we can adjust that

Exmaple - change idle limit from 2 (default) to 20 Kbps:
```
root@R7000:/tmp/home/root# nvram get rast_idlrt
2
root@R7000:/tmp/home/root# nvram set rast_idlrt=20
root@R7000:/tmp/home/root# nvram get rast_idlrt
20
root@R7000:/tmp/home/root# nvram commit
```

pedro0311, 2025-08-26:
I'm sure when @MarsTomato comes back he'll take care of it.

MarsTomato, 2025-10-01:
close

closed 2025-10-01 by MarsTomato