FreshTomato-Project/freshtomato-arm #118: Feature question / request : Able to serve xml page from nginx

captainmac40, 2025-07-25:
Hello and first very good job !!

I want to replace my ISP fiber modem by my EX7000 router under FreshTomato.

For internet only it's ok by using dhcp option 60 on wan port to my ONT (fiber to lan converter).

But for my ISP TV player I need some xml served to my player from router. 

So I want to know if it's possible to tweak nginx to serve this pages to my player

Thanks a lot

Kind regards

pedro0311, 2025-07-27:
You mean this module: https://nginx.org/en/docs/http/ngx_http_xslt_module.html ?
It requires libxml2 (already in FT), but also libxslt, which should be added to FT (I don't know about the size).

captainmac40, 2025-07-30:
No i'm talk about just provide "as it" xml files same as an html page from router. My ISP tv player check on ISP router if some xml file is available to confirm if it's an ISP router or not :)

pedro0311, 2025-07-30:
So this is not an issue or a feature request, you just don't know how to configure nginx.
Here you can ask for help: https://www.linksysinfo.org/index.php?forums/tomato-firmware.33/