FreshTomato-Project/freshtomato-arm #122: admin-cifs.asp -- Page doesn't refresh after clicking the save button

HommeOursPorc, 2025-08-01:
On the Administration > CIFS Client page (admin-cifs.asp), the page doesn't seems to refresh after clicking the save button. I'm using Firefox for Android on my smartphone. FT 2025.3 VPN on RT-AC68U.

The information on the page is not correctly refreshed after clicking save, so it is not possible to know if the CIFS client is properly mounted or unmounted. 
      
[image]

> 

[image]
As you can see in the logs, I thought it wasn't working/mounting correctly but in fact it was mounting properly, it just wasn't refreshed on the page.

>    

[image]
Here the second client is unmounted but the drive size is still shown after clicking the save button. A page refresh correct the problem.

rtrout, 2025-08-02:
> > Note: Total/Free Size never observed reported, only (not mounted).
> 
> Do you mean that the size is **never** reported even after manually refreshing the page?

Correct. /cifs1 is mounted but the Total / Free Size shows like in the screenshot (for me).

HommeOursPorc, 2025-08-02:
I don't know exactly how it works so excuse my ignorance. 

https://github.com/FreshTomato-Project/freshtomato-arm/blob/75e3946a73ac39d43ba93fc277558fbd7894928a/release/src-rt-6.x.4708/router/rc/cifs.c#L104

This line above ^^, can notice_set use pass any message? There are no "mounted" final message in the log to confirm that the system is mounted.

Just an idea. Not sure if it absolutely needs to be an empty string. 

And for the size not being reported, this seems to be done in JavaScript? Here? 

https://github.com/FreshTomato-Project/freshtomato-arm/blob/75e3946a73ac39d43ba93fc277558fbd7894928a/release/src-rt-6.x.4708/router/www/admin-cifs.asp#L135

From the little I understand, it look like it is something like a if else statement. **If** size is reported -> show size "Total / Free Size" **else** "(not mounted)"

There doesn't seem to have anything in between, if the system is mounted but no size is being "reported", then it will show "not mounted".

How come cifs1.size return nothing??? 🤔

HommeOursPorc, 2025-08-03:
> > Just curious, what happen if you use the `df` command for the cifs directory?
> 
> <img alt="Image" width="520" height="43" src="https://private-user-images.githubusercontent.com/6046664/473771843-a6678b1f-10aa-4262-85be-1877704bf710.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NTQyMjk4NzAsIm5iZiI6MTc1NDIyOTU3MCwicGF0aCI6Ii82MDQ2NjY0LzQ3Mzc3MTg0My1hNjY3OGIxZi0xMGFhLTQyNjItODViZS0xODc3NzA0YmY3MTAucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI1MDgwMyUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNTA4MDNUMTM1OTMwWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9NjU4ODRjMGQ3MjA1YjQxZWJjMDgxM2UzNTM0ZjA1N2RmZTBjZmNlOGFkNDg3ZDEzZDM3YTAxYjdkOTZjMWE5ZiZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.GqH0QqJ-OHEwrAAbvQzN1aAfr5q-H5DxnTRQbZFwJcU">
> 
> In hindsight, that may be the problem. I'm not sure tomato is ready for terabytes of storage :-)

It worked fine with my 12TB HDD... How many TB do you have?

rtrout, 2025-08-04:
2TB free

commit e7c379f2 (2025-08-04): GUI: Administration: CIFS Client: fix refreshing 'Total / Free Size' (close #122)
closed 2025-08-04 by pedro0311