FreshTomato-Project/freshtomato-arm #123: Dns functions problem

jkle112, 2025-08-01:
"Lancethepants" features:
- DNSSEC integration and GUI
- DNSCrypt-Proxy selectable/manual resolver
- Tinc Daemon integration and GUI
- Comcast DSCP Fix GUI
Copyright (C) 2014-2022 Lance Fredrickson
[lancethepants@gmail.com](mailto:lancethepants@gmail.com)
 
Netgear R6300V2 with FreshTomato Firmware 2025.3 K26ARM USB AIO_Lite-64K is using in China.
The DNS functions with stubby and dnscrypt-proxy v1 could not to used ,hope upgrade to dnscrypt-proxy v2,add fake SNI.
Thank you!

[syslog.txt](https://github.com/user-attachments/files/21555362/syslog.txt)

pedro0311, 2025-08-02:
I don't understand: what is your problem?

HommeOursPorc, 2025-08-02:
Maybe @lancethepants would know more about this subject?

pedro0311, 2025-08-02:
@HommeOursPorc : dnscrypt-proxy v2? Nope, read my previous posts on LI - it's in go, so it's no-go-go :)

HommeOursPorc, 2025-08-02:
Oh ok! I didn't know, thanks for the information @pedro0311 😊

Sorry @jkle112, it look like dnscrypt-proxy v2 is not possible on FreshTomato.

jkle112, 2025-08-02:
Most time,could not to used stubby and dnscrypt-proxy v1,if could not upgrade to dnscrypt-proxy v2,could add smartdns,I could used it to replace stubby/dnscrypt-proxy v1/adblock.
https://github.com/PikuZheng/smartdns
Thank you!

tomkel, 2026-02-18:
I've been running dnscrypt-proxy v2 from entware, managed with rc.unslung, using WAN up scripts if necessary. But it would be really nice if v2 were integrated with freshtomato instead.  

I've also installed a go dev toolchain on the router for compiling programs. Only catch is that I have to make sure I'm compiling for softfloat and uClibc.