FreshTomato-Project/freshtomato-arm #146: OpenVPN with disabled IPv6

FabianHassler, 2025-09-13:
On my system, IPv6 is disabled (under Basic/IPv6). If I connect to my OpenVPN provider, the server tries to push an IPv6 address. I get the following error in my log:

Sep 13 10:47:40	daemon	notice	openvpn-client1[5945]	/usr/sbin/ip -6 addr add <IP>/64 dev tun11
Sep 13 10:47:40	daemon	err	openvpn-client1[5945]	Linux ip -6 addr add failed: external program exited with error status: 2

The VPN connection does not start.

I also cannot ignore the push from the server. Both command "push-remove" and "push-filter" do not seem to be recognized.

pedro0311, 2025-09-17:
`pull-filter ignore "whatever-you-want-to ignore"` doesn't work??

FabianHassler, 2025-10-01:
> `pull-filter ignore "whatever-you-want-to-ignore"` doesn't work??

That resolves the issue. It seems that I have mixed up pull-filter and push-filter.